olikraus/u8g2 #357: SSD1309 128x64 SPI - Only text working

metropt, 2017-09-13:
Hi :)

I was trying your library with the following oled https://pt.aliexpress.com/item/1-54-inch-White-OLED-Display-Module-128x64-SPI-IIC-I2C-Interface-OLED-Screen-Board-3/32821083083.html and a Arduino M0 however I can get texto from it:

https://photos.app.goo.gl/qPPmk9AL1XkPZBUB3

With U8x8 library I got text also:

https://photos.app.goo.gl/1r6iEDflORlZU3872

I tried I2C but only with SPI it worked. I tried all the SSD1309 and SSD1309 HW SPI variations :( With adafruit SSD1306 library I also got text on it but scroll doesn't working per example.

What should I do?

olikraus, 2017-09-14:
> What should I do?

I do not know, because I do not understand your problem. What exactly is your problem?

metropt, 2017-09-14:
The video looks ok to you?

olikraus, 2017-09-14:
Which video? I pressed on the links, but no video appers (at least not on my linux laptop)

And: If you ask "What should I do?", then you need to tell me what you want to achieve.

One more thing: This is a spare time project. I do not get ANY money here. I am willing to answer questions from users, but please do not let me guess what you want. State your problem and ask a clear and precise question.

metropt, 2017-09-14:
If you press the image it doesn’t start a video?

> One more thing: This is a spare time project. I do not get ANY money here. I am willing to answer questions from users, but please do not let me guess what you want. State your problem and ask a clear and precise question.

I know that. Thats why I made the videos I tell you all the things I tried. With what should I do next I want to know what path to take to evaluate where the problem comes from.

Thank you

olikraus, 2017-09-14:
> If you press the image it doesn’t start a video?

No, instead I saw the same picture twice. Maybe i was to  impatient.
Meanwhile i saw the video. But, you wouldn't believe: I waited for 42 seconds, but still I do not what your problem is. 

> With what should I do next I want to know what path to take to evaluate where the problem comes from.

Maybe you should just tell me the problem.

ok, let me recall you first post.

> I tried I2C but only with SPI it worked. 

So what is the problem? You want to switch to I2C?
So the problem statement could be: I2C does not work, what can I do.

> I tried all the SSD1309 and SSD1309 HW SPI variations

What does this mean? Great you tried everything! So was it working? What was not working?
How is the fact, that you tried HW SPI related to the fact that I2C does not work (note: SPI vs I2C).
Should this give me more information? 

> With adafruit SSD1306 library...

What does this mean? What do you want to tell me?

>  ...but scroll doesn't working per example.

I have absolutly no idea what you want to tell me. What is the problem? Scrolling does not work with Adafruit examples? If so, you are completly wrong here.

Again, and final request: What is your problem?

metropt, 2017-09-14:
I'm sorry, my mistake. For some reason I wasn't expect some random graphics on the draw clip test and assume it was an error. It is working as expected. Thank you and sorry for my confusion.

olikraus, 2017-09-14:
Ok...

closed 2017-09-14 by olikraus