olikraus/u8g2 #359: Full Graphic Smart Controller (128x64) for RAMPS 1.4, how to rotate the screen 180deg

eduardoelujanf, 2017-09-16:
Hello, sorry to bother but I need some help.

I designed a 3d printer like the prusa i3... 
It has a Full Graphic Smart Controller on the front. 
I have some design limitations that forced me to mount it upside down. 

Basically I need to rotate it 180deg (things look upside down).
The encoder rotation is another issue, but that I know how to solve.

I'm using a Ramps 1.4 board with Marlin Version 1.1.5 – August 2017 (newest)
The Full Graphic Smart Controller works fine. I only need to rotate the screen.

Also, should I use "u8glib" or "u8g2" as my library?

This is my display: http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller

olikraus, 2017-09-16:
I think this is something you have to ask the reprap people. I am not relateded to 3d printing project. 
However, to my knowledge, they still use u8glib.

olikraus, 2017-09-23:
no further feedback/information --> closing

closed 2017-09-23 by olikraus