olikraus/u8g2 #360: Connecting display ST7920 to U8g2 library

mba1973, 2017-09-17:
Hi,
I am trying to use the U8g2 library with a ST7920 display and an arduino.
I would like not to use the arduino IDE so I am writing the code in Atmel studio.
I have read the guide on how to setup the display but I can't get it to work.
The display works fine. I have used some other simple code for showing text and graphic.
Do you have an example that I could have?

Regards Morten

olikraus, 2017-09-17:
DId you read the porting instructions here?

https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

Oliver

mba1973, 2017-09-17:
Yes. Here are some of my code:

#include <u8g2.h>

u8g2_Setup_st7920_s_128x64_f(&disp, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_samd20); //Configure the display object, this will also initialize a 128 byte frame buffer
u8g2_InitDisplay(&disp);		//Send initialization code to the display
u8g2_SetPowerSave(&disp, 0);

uint8_t u8g2_gpio_and_delay_samd20(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr){
	
	switch(msg){
		//Initialize SPI peripheral
		case U8X8_MSG_GPIO_AND_DELAY_INIT:
		initializeIO();
		break;
		
		//Function which implements a delay, arg_int contains the amount of ms
		case U8X8_MSG_DELAY_MILLI:
//		_delay_ms(arg_int);
		
		break;
		//Function which delays 10us
		case U8X8_MSG_DELAY_10MICRO:
//		_delay_us(10);
		
		break;
		//Function which delays 100ns
		case U8X8_MSG_DELAY_100NANO:
//		_delay_us(0.1);
		
		break;
		//Function to define the logic level of the clockline
		case U8X8_MSG_GPIO_SPI_CLOCK:
//		port_pin_set_output_level(DISPLAY_SCLK, arg_int);
		
		break;
		//Function to define the logic level of the data line to the display
		case U8X8_MSG_GPIO_SPI_DATA:
//		port_pin_set_output_level(DISPLAY_MOSI, arg_int);
		
		break;
		// Function to define the logic level of the CS line
		case U8X8_MSG_GPIO_CS:
//		port_pin_set_output_level(DISPLAY_CS, arg_int);
		
		break;
		//Function to define the logic level of the Data/ Command line
		case U8X8_MSG_GPIO_DC:
//		port_pin_set_output_level(DISPLAY_DC, arg_int);
		
		break;
		//Function to define the logic level of the RESET line
		case U8X8_MSG_GPIO_RESET:
//		port_pin_set_output_level(DISPLAY_RESET, arg_int);
		
		break;
		default:
		return 0; //A message was received which is not implemented, return 0 to indicate an error
	}
	
	return 1; // command processed successfully.
}

But I just get some error message like undefined reference to u8x8_SetPowerSave, u8x8_InitDisplay, u8x8_byte_4wire_sw_spi and so on.

olikraus, 2017-09-17:
hmm.. sounds like an Atmel Studio problem. Did you include all u8g2 files?

mba1973, 2017-09-17:
I have included u8g2-master\csrc

olikraus, 2017-09-17:
This directory?

https://github.com/olikraus/u8g2/tree/master/csrc

mba1973, 2017-09-17:
Yes

olikraus, 2017-09-17:
Well, all function are included there, so it seems to be a linker problem with Atmel Studio.

mba1973, 2017-09-17:
Ok thanks. I have to look into it then.
Regards Morten

olikraus, 2017-09-30:
will close this for now... feel free to reopen/comment.

closed 2017-09-30 by olikraus