olikraus/u8g2 #361: Alternative OLED 64x32 Device

olikraus, 2017-09-19:
Change 2f back to 1f for the 64x32 device
U8X8_CA(0x0a8, 0x1f), /* multiplex ratio: changed from 0x1f to 0x2f */

create new device for this, update examples.

see #254

k1hop, 2017-09-22:
If anyone is curious to see this display operating with this change, here's a video I did of some preliminary experimentation.  Sorry, the camera is a bit out of focus. Using U8G2 library.
https://youtu.be/ecOLQpK0U3s

Russ - K1HOP

k1hop, 2017-09-23:
hi... thanks for creating the new constructor(s) to support this change.
I had a chance to test it out this morning.  Before downloading and installing your new zip file to Arduino, I first removed the change I had done to the C file and confirmed that the old problem persisted.  Here's a photo of the 64x32 screen with the old C file and constructor showing the problem.  As you can see, the image appears shifted up by a couple of rows (half a screen) and the lower half is full of random RAM junk which never gets cleared or addressed with new data.

[image]

k1hop, 2017-09-23:
I then downloaded and installed your new version of the library with the new constructor(s).  I changed my Arduino code to use the new constructor:  U8G2_SSD1306_64X32_1F_F_HW_I2C u8x8(U8G2_R0, 19, 18);
This worked just fine, as attested by the following screen images.

[image]
[image]

k1hop, 2017-09-23:
If you care to view a portion of the full demo using both the old and new constructors, you can see short videos at my YouTube channel: https://www.youtube.com/channel/UCkJYUHvNJll_MdXTmdzZ2gg

olikraus, 2017-09-23:
Nice Videos and Pictures. So all is done and fixed, right? We can close this issue?

Thanks a lot for your contribution!

k1hop, 2017-09-23:
Looks great to me!  Would be nice/comforting if other users of same display chimed in.  But I'm A-OK now! :)

olikraus, 2017-09-23:
there will be an official release soon :-)

closed 2017-09-23 by olikraus