olikraus/u8g2 #362: Help with I2C detection

jgrossen, 2017-09-20:
I need to detect that a display is connected to the I2C bus before using the userInterface routines.  Is there some routine I can use get the displays I2C address? I am using the SW I2C option.  I am trying to develop a small module with a display and 3 buttons that will plugged into an Arduino interrupting the main code so that some parameters can be changed.  Then when unplugged the main code in the Arduino will startup running again

olikraus, 2017-09-20:
At least in u8g2 there is no such procedure, because u8g2 only writes to the display.
In general you could use wire lib to test whether a i2c device is available on the bus, but I do not exactly know what command to use here. 
Also note that there are some OLED displays out there with incomplete I2C support. Those OLEDs can not be detected at all.

olikraus, 2017-09-30:
no further comment, closing...

closed 2017-09-30 by olikraus