olikraus/u8g2 #363: SSD1305 OLED - graphics incorrectly offset

max-bishop, 2017-09-20:
First of all thank you for a great library.

Using u8g2 with 2.23" 128x32 OLED screen and SSD1305 Driver with sample sketch:

> GraphicsTest.ino

Using the following constructor:

`U8G2_SSD1305_128X32_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 6, /* dc=*/ 7, /* reset=*/ 9);`

The OLED powers up and displays graphics. 

The issue is the displayed graphics appear to be offset in X and scaled incorrectly. Also rogue pixels are being displayed on screen right.

Image showing the issue is attached. 

Help or suggestions are welcome.

[image]

olikraus, 2017-09-20:
Try to change the x offset here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1305.c#L152
Maybe use 4 instead of 2.

max-bishop, 2017-09-20:
You sir, are awesome.

That fixed it.

Thank you very much!

closed 2017-09-20 by max-bishop

olikraus, 2017-09-21:
:)

chicodefe, 2020-02-17:
you have any update  for no modify the drive?

olikraus, 2020-02-17:
Not at the moment. Du you own the same display?

edstott, 2022-02-20:
Hi, just a note that I've had the same problem with this chip and I've successfully applied the same fix in the source.
Display type is Winstar WEO012832ABPP3N00000
Thanks!