olikraus/u8g2 #364: Feature Request/Question WRT Ucglib::setClipRange

cwalv2, 2017-09-21:
Ucglib seems to have some more useful drawing features (primarily the clipRange features), but...:
a) Ucglib seems to not be getting updates
b) Ucglib doesn't support monochrome screens (unless I'm wrong)

Any chance of getting some of these features in u8g2?
Any chance of renewed interest in Ucglib and/or allowing Ucglib to push to monochrome screens?

Thanks,
-Michael

olikraus, 2017-09-21:
> Ucglib seems to not be getting updates

well yes, but it is still planned to do an update once

> Ucglib doesn't support monochrome screens (unless I'm wrong)

Yes, there once was a split of u8glib. RGB parts went to ucglib, monochrome parts are now called u8g2.

> Any chance of getting some of these features in u8g2?

The "setClipRange" feature is a essentional part of ucglib. The extension to the user api function was available almost for free. This is not the case for u8g2, where the clipping is more spezialized and also required for the paged output mode. Adding this freature will increase size of the lib and lower the speed.

Implementation ideas:

full buffer mode:
The existing clipping could be replaced (same case as ucglib): Introduction of a callback function which either calls the existing windows clipping or the more flexible sub-window clipping

page buffer mode:
The existing page clipping procedure can not be replaced: Introduce a callback function which is called before the page clipping procedure. This callback is either empty or does the sub-window clipping.

olikraus, 2017-09-30:
moved because of urgent bug fixes

olikraus, 2017-10-13:
moved to 2.20

olikraus, 2017-11-18:
the clip function should be added to
void u8g2_DrawHVLine(u8g2_t *u8g2, u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t len, uint8_t dir)
in
u8g2_hvline.c

yangyud-cn, 2017-12-01:
To allow more flexibility, especially with clipping enabled, the coordinates should also accept negative numbers, for example,  drawCircle(-10, 5,  50),   drawStr(-20, 20, "Hello").

olikraus, 2018-02-13:
hmmm. a long pending topic. A nice and usefull feature.... But I didn't found time :-p

olikraus, 2018-02-17:
move to next milestone...

olikraus, 2018-10-27:
more ideas on this:

Current situation:
Sequence is this:
hline/vline calls the display rotation procedure
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_hvline.c#L212

the display rotation calls the 4dir procedure:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_setup.c#L214

the 4dir procedure calls the 2dir procedure:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_hvline.c#L196

the clipping happens in the 2dir procedure.
Involved calls are:

u8g2_DrawHVLine()
u8g2->cb->draw_l90()
u8g2_draw_hv_line_4dir()
u8g2_draw_hv_line_2dir()   <<< clipping happens here
u8g2->ll_hvline()

Maybe the clipping could be reused, if it is done more earlier on the user coordinates.
clipping than has to happen against the current page coordinates
  u8g2_uint_t user_x0;	/* left corner of the buffer */
  u8g2_uint_t user_x1;	/* right corner of the buffer (excluded) */
  u8g2_uint_t user_y0;	/* upper edge of the buffer */
  u8g2_uint_t user_y1;	/* lower edge of the buffer (excluded) */
to ensure, that all buffer updates are correct.

if we add a user defined clip window, then we have to calculate the intersection between the above page window first and use the result as a clipping window.
 
Added asserts for further testing, test file utf8/clip_test_picture_loop.

  U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 12, /* dc=*/ 4, /* reset=*/ 6);	// Arduboy (Production, Kickstarter Edition)
  make -f Makefile.184.uno
     text	   data	    bss	    dec	    hex	filename
   8440	    152	    420	   9012	   2334	HelloWorld.elf

olikraus, 2018-10-27:
```
void u8g2_SetMaxClipWindow(u8g2_t *u8g2);
void u8g2_SetClipWindow(u8g2_t *u8g2, u8g2_uint_t clip_x0, u8g2_uint_t clip_y0, u8g2_uint_t clip_x1, u8g2_uint_t clip_y1 );
void setMaxClipWindow(void)
void setClipWindow(u8g2_uint_t clip_x0, u8g2_uint_t clip_y0, u8g2_uint_t clip_x1, u8g2_uint_t clip_y1)
```

olikraus, 2018-10-28:
@yangyud-cn It may not look like this, but in general negative number are accepted (the negative value will be converted to an unsigned number). It it still doesn't work, you need to put u8g2 into 16 bit mode (see FAQ).

olikraus, 2018-10-28:
added clipping to the new feature selection wiki page
https://github.com/olikraus/u8g2/wiki/u8g2optimization

olikraus, 2018-10-28:
todo: documentation

olikraus, 2018-10-28:
I think this is done....

closed 2018-10-28 by olikraus