olikraus/u8g2 #365: ST7567 LCD (JLX12864G-1509) Contrast

hardl0ck, 2017-09-21:
Hello! 
I`m try to use JLX12864G-1509 LCD based on ST7567 COG with STM32F103CB arduino.

in contructor line:
U8G2_ST7567_PI_132X64_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 6, /* data=*/ 4, /* cs=*/ 7, /* dc=*/ 5, /* reset=*/ 3);

LCD 128*64, but constructor do not understand 128X64...

SW interface is because I could not correctly connect the HW interface. Tell me how to correctly connect HW interface of SPI ?  SPI: SS, SCK, MISO, MOSI;  LCD: RS, RES, CS, SDA, SCK...

I`m try ContrastTest example and have result on video:
https://www.youtube.com/watch?v=Vz37DZ8w95c

hardl0ck, 2017-09-21:
I asked, and I will answer.

In file **u8x8_d_st7567.c** line 122
`  U8X8_C(0x026),		                /* v0 voltage resistor ratio */`
change to
`  U8X8_C(0x023),		                /* v0 voltage resistor ratio */`

Voltage ratio (Regulation Ratio) changed from 6.0 to 4.5 (see page 26 in ST7567 datasheet).

Now with 
`u8g2.setContrast(42);`
i have result:
[image]

olikraus, 2017-09-21:
Thanks for your input. 

I have created beta release 2.18.1, which includes a new constructor with your modifications:

```
U8G2_ST7567_JLX128X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);  
U8G2_ST7567_JLX128X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);  
```

You can download the beta release here: 
https://github.com/olikraus/U8g2_Arduino

Install via "add zip library" menu.

Can you test u8x8/flipmode example? Will it work with your display?

Thanks!

hardl0ck, 2017-09-21:
Thanks for new version!

I`m tryin FlipMode sample with new version library. 
Just add 1 line in "setup"
`u8x8.setContrast(42);`

Result on video:
https://youtu.be/gfj2lF2N0Og

olikraus, 2017-09-21:
Great. Thanks for the video. I fixed the contrast issue also. It should be correct now by default.
Beta 2.18.2 is available for download as zip file at the same location.

hardl0ck, 2017-09-21:
OOOPS.
hw spi not worked...

worked, but wrong contrast:
```
//U8G2_ST7567_PI_132X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 6, /* data=*/ 4, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);  // Pax Instruments Shield, LCD_BL=6
//U8G2_ST7567_PI_132X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);  // Pax Instruments Shield, LCD_BL=6
```

worked fine, 
`//U8G2_ST7567_JLX12864_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 6, /* data=*/ 4, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);
`
not worked, clear screen and led on STM32 board blinked....  
`U8G2_ST7567_JLX12864_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);`

olikraus, 2017-09-21:
> worked, but wrong contrast:

Contrast is still wrong???

> hw spi not worked...

Did it work before?

hardl0ck, 2017-09-21:
> Contrast is still wrong???
Contrast wrong in 132X64, but HW SPI works. 
In 128X64 (v2.18.2) Contrast is correct.

> Did it work before?
Yes. In version before add 128X64 HW SPI works fine. Now HW SPI works in 132X64, but no work in 128X64...

olikraus, 2017-09-21:
> Contrast wrong in 132X64, but HW SPI works.

The correct contrast depends on the display. For the 132x64 display, the contrast is correct.

> In 128X64 (v2.18.2) Contrast is correct.

good.

> Yes. In version before add 128X64 HW SPI works fine. Now HW SPI works in 132X64, but no work in 128X64...

You mean, it worked in 2.18.1, but did not work in 2.18.2? Strange, I have not changed anything here. 
Also from 2.17.x there is no change. Can you crosscheck wiring?

hardl0ck, 2017-09-21:
HW SPI for 128x64 did not work in version 2.18.1 and is still not working in 2.18.2
HW SPI works fine for 132X64 in 2.17.x, 2.18.x.
in sample "Hello world" i`m trying this contructor lines. The first three lines work, but last line does not work.

`U8G2_ST7567_PI_132X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 6, /* data=*/ 4, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);  // Pax Instruments Shield, LCD_BL=6`
`U8G2_ST7567_PI_132X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);  // Pax Instruments Shield, LCD_BL=6`

`U8G2_ST7567_JLX12864_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 6, /* data=*/ 4, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);  `
`U8G2_ST7567_JLX12864_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);`

All wirings are correct (work with two first lines).`

BUT! HW SPI works in FlipMode sample (U8X8)
`U8X8_ST7567_JLX12864_4W_HW_SPI u8x8(/* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);`

hardl0ck, 2017-09-21:
In first part this video FlipMode sample with constructor line
`U8X8_ST7567_JLX12864_4W_HW_SPI u8x8(/* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);`
In second part video - Hello World with:
`U8G2_ST7567_JLX12864_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 13, /* reset=*/ 14);`

Led connected to PB1. [Shematic STM32F103C](https://github.com/leaflabs/maplemini/raw/master/maplemini.pdf)

https://youtu.be/S5Vy7TbY-YU

olikraus, 2017-09-21:
Thanks a lot for all your testing activities. I think i found the bug. I have created beta release 2.18.3. Now everything should be fixed (hopefully).

hardl0ck, 2017-09-22:
Thanks! 
All work fine. FPS Test:
STM32F103 ST7567_JLX12864_F_4W_SW_SPI		SW SPI	        FPS: Clip=11.9 Box=12.0  @=9.8 Pix=10.6		
STM32F103 ST7567_JLX12864_F_4W_HW_SPI		HW SPI	        FPS: Clip=134.1 Box=162.8  @=40.1 Pix=58.6

olikraus, 2017-09-22:
Wow, excellent values.

olikraus, 2017-09-23:
issue list cleanup --> closing this

closed 2017-09-23 by olikraus