olikraus/u8g2 #366: Generating custom font from C byte arrays

h4jang, 2017-09-21:
Hi Oliver,

I created a custom font in a form of byte array in C for my arduino project, such as:

const uint8_t A[] PROGMEM = {
  0x00,         // 00000000 
  0x00,         // 00000000
  0x00,         // 00000000
  0x38,         // 00011100
  0x44,         // 00100010 
  0x82,         // 01000001
  0x82,         // 01000001 
  0xFE,         // 01111111
  0x82,         // 01000001 
  0x82,         // 01000001
  0x82,         // 01000001
  0x82          // 01000001
};

Now I am trying to convert this into an actual font that's supported by U82G library so that I can take full advantage of your library instead of drawing each character one by one. However, as far as I know I can only generate a custom font using a bdf file. Is there any way that I can do that?

Thanks in advance!

closed 2017-09-21 by h4jang

akamensky, 2020-01-03:
so the issue is closed without any info? I am looking to use https://github.com/greiman/SSD1306Ascii/blob/master/src/fonts/font5x7.h with this library as I like to look of this font on my 1.3" screen.

olikraus, 2020-01-03:
There are many fonts available for the ASCII only version of u8g2 (u8x8). Have a look at the u8x8 font page.

akamensky, 2020-01-03:
I’ve looked. Those fonts are 8x8 sized and look very unpleasant to me if I am being honest. Monospaced font doesn’t need to be square boxed. That font above is 5x7 font that looks pretty clear on the small screen.

olikraus, 2020-01-03:
Hmm u8x8 only supports glyphs on 8x8 pixel boundaries. Then either use u8g2 or the mentioned ascii lib...