olikraus/u8g2 #370: Test dogxl240

olikraus, 2017-09-27:
... for next release because of modifications #339

olikraus, 2017-09-27:
- FPS seems to be partly corrupted, other examples do not work at all
- Changing the optimization flags, does not change this (tested in Arduino IDE)
- Tested with DOGXL240 and Arduino Due. Other displays affected?

olikraus, 2017-09-30:
/tmp/arduino_build_608416/sketch/HelloWorld.ino.cpp.o: In function `U8G2::setPageCurrTileRow(unsigned char)':
/home/kraus/Arduino/libraries/U8g2/src/U8g2lib.h:163: undefined reference to `u8g2_SetPageCurrTileRow'

olikraus, 2017-09-30:
beta release 2.18.12 should fix this bug.

closed 2017-09-30 by olikraus

arnoldvillasanta, 2017-09-30:
The fix works with UC1611 LCD (240x160) for all the full_buffer samples:
Great support and timely fix as always.

[image]

olikraus, 2017-09-30:
Thanks a lot. I wonder that noone complained about this before. The full buffer problem must exist since several versions. 

I also just released 2.18.13. It will be available for the Arduino IDE soon.

arnoldvillasanta, 2017-09-30:
Maybe most users have limited MCU RAM that they opted for the page_buffer.  You're such a good project owner to keep up with all these changes (sometimes with too demanding users).

olikraus, 2017-09-30:
:)