olikraus/u8g2 #371: UC1638

jchifos, 2017-09-27:
I modified UC1608  file to use UC1638.  It seems to work except rows of pixels intermittently do not display correctly.  If i set boxes at height of 8 or 16 then no problems.

Below is arduino code:

`#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

U8G2_UC1608_240X128_1_4W_HW_SPI u8g2(/*rotation*/U8G2_R0, /* clk13 , MOSI11,*/ /*cs*/ 2, /*dc*/3, /*reset*/4);   /*WAS: rotation/U8G2_R2 */

void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.firstPage();
  do {
    u8g2.drawBox(40, 32, 160, 5);
    u8g2.drawBox(40, 64, 160, 5);
    u8g2.drawBox(40, 96, 160, 5);
    u8g2.drawBox(40, 128, 160, 5);
  } while ( u8g2.nextPage() );  
  delay(2000);
}`

1638 init code:

static const uint8_t u8x8_d_uc1608_240x128_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_C(0x024),            		/*	 set temp comp*/
  U8X8_C(0x0c0),            		/*	mirror y and mirror x */  /* WAS: c2 */
  U8X8_C(0x0a2),            		/*	line rate */
  U8X8_C(0x0d6),            		/*	gray scale 2 */
  U8X8_C(0x0eb),            		/*	 set bias*/
  U8X8_C(0x095),            		/*	 set 1 bit per pixel, pattern 0*/
  U8X8_C(0x089),            		/*	 set auto increment, low bits are AC2 AC1 AC0 */  
  
  U8X8_C(0x081),		/* set contrast, EA default: 0x054 */    /* UC1638*/
  U8X8_D1(0x0a0),		/* set contrast, EA default: 0x054 */    /* UC1638*/

jchifos, 2017-09-27:
[image]

olikraus, 2017-09-27:
hmmm... sounds good, would it be helpful if I add a new device for this controller?

jchifos, 2017-09-27:
I am having problems. Sometimes it works than other times a partial row of pixels will be offset.  any thoughts on how to fix this?

olikraus, 2017-09-30:
Shell I start working on this? I mean, I will create a u8g2 device for this and add the commands from above. But i need your support to fix and test the new device. While doing so, we can also try to fix your problems from above. 

So... shell we both start working on this?

jchifos, 2017-09-30:
Yes.  Lets do it.  I can send the files I modified.  I will send on Monday (2 Oct).

olikraus, 2017-10-02:
what is the dimension of your display? 240x128? any datasheet of the display available?

jchifos, 2017-10-02:
[COG-VLGS7291_COMSEG mapping_2017-9-19.zip](https://github.com/olikraus/u8g2/files/1349237/COG-VLGS7291_COMSEG.mapping_2017-9-19.zip)
[u8x8_d_uc1608.zip](https://github.com/olikraus/u8g2/files/1349238/u8x8_d_uc1608.zip)

jchifos, 2017-10-02:
I am using 1 bit per pixel.  I have tried both 1 bit per pixel and 2 bit per pixel with similar results.  I prefer 1 bit per pixel.

olikraus, 2017-10-02:
Notes on the UC1638
 * B/W mode includes two commands: Display enable (0x0c9, 0x0ad) and display mode (0x095)
 * B/W mode seems to cover two page, only even page values required?
 * UC1638 differs from UC1608 by using DC=1 for cmd args: How will I2C work? Analysis need, I2C deactivated for now.

olikraus, 2017-10-02:
I have created a beta release:

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Please download and install via the add zip library menu of the Arduino IDE.
What will you see on the screen?

jchifos, 2017-10-02:
I tried the beta release.  Display was blank.
1) The U8X8_CA command  does not appear to force DC=1 on argument byte.
2) CS is active low
3) reset pulse width >= 5msec    
4) post reset wait >= 150msec
5) Sclk < 4MHz.  I set to 1Mhz
6) spi mode 3
7) Controller automatically increments 2 pages in B/W mode.  (no need to *2)

olikraus, 2017-10-03:
> Display was blank.

oh

> The U8X8_CA command does not appear to force DC=1 on argument byte.

Can you crosscheck this? I have changed the CAD (command arg data) procedure from cad011 for the uc1638 here: https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c#L448 
So DC line should be 1 now for arguments.

> CS is active low

Looks also good:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1638.c#L155

> reset pulse width >= 5msec
post reset wait >= 150msec
Sclk < 4MHz. I set to 1Mhz
spi mode 3

ok, I have fixed this

> Controller automatically increments 2 pages in B/W mode. (no need to *2)

But i still need to reassign the current write position. Do i need to use *2 then? I think this can be figured out once something is visible.

I have created a new beta release 2.19.2:

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

olikraus, 2017-10-07:
Any feedback here?

CrNMGuy, 2018-08-28:
Hey olikraus, 
would like to go further on the UC1638
how can i help ?
Got full documentation here 

Got this device : 192x96 dots, I2C connection, 3,3V, LG192962-DW-v33 with UC1638 controler

[lg192962-ffdwh6v-v33 rev 0.1.pdf](https://github.com/olikraus/u8g2/files/2328359/lg192962-ffdwh6v-v33.rev.0.1.pdf)
[uc1638c_a1.1.pdf](https://github.com/olikraus/u8g2/files/2328360/uc1638c_a1.1.pdf)

[LG19296-12-I2C.txt](https://github.com/olikraus/u8g2/files/2328363/LG19296-12-I2C.txt)

olikraus, 2018-08-28:
Oh, a really old issue.

Does the existing `U8G2_UC1638_160X128` device work (at least a little bit)?

CrNMGuy, 2018-08-29:
At the moment, there is no definition available for U1638 on I2C
i Checked here : https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h

Can i create one ?

olikraus, 2018-08-30:
It is here for example:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L6529

The U8X8 is in U8x8lib.h and the U8g2 constructor is in U8g2lib.h

olikraus, 2021-06-15:
Currently I have this in my lab:
https://de.aliexpress.com/item/32900322454.html
[image]

olikraus, 2021-06-19:
@CrNMGuy It looks like, that I have exactly the same LCD, right?
It comes preconfigured in 8080 mode. So you did change this accordingly to I2C?

olikraus, 2021-06-20:
SPI mode works as expected now, but there are still some issues with I2C. 
One of the issues is this: What is the I2C address of the device? Is the address base value 0x78 or 0x70? I didn't find a proper hint in the display documentation.

olikraus, 2021-06-20:
I2C is fixed. The address of the LG192962 can be selected between 0x78 (CS=0) and 0x7c (CS=1). In U8g2 the default is now 0x78 for the UC1638. Background: There are actually four I2C addresses selectable, 0x70, 0x74, 0x78 and 0x7c
 * U8g2 intentionally breaks with the rule to use the lowest possible I2C address
 * Many other displays have the same I2C address 0x78
 * It can be assumed, that the active high CS is not used and internally connected to high, this will allow only 0x78 and 0x7c
 * Anyhow: The user of this display must ensure to connect the CS signal to GND to use the u8g2 default address. Otherwise, if CS is connected to 3.3V, then use 0x7c as I2C address.

olikraus, 2021-06-20:
I have created beta 2.29.8:
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

closed 2021-06-26 by olikraus

srcastor, 2023-01-09:
Hello, Same issue here, uc1638C 256x128 , until today I have been using this file but slightly modified on tile_hight, tile_hight and mirror

> [COG-VLGS7291_COMSEG mapping_2017-9-19.zip](https://github.com/olikraus/u8g2/files/1349237/COG-VLGS7291_COMSEG.mapping_2017-9-19.zip) [u8x8_d_uc1608.zip](https://github.com/olikraus/u8g2/files/1349238/u8x8_d_uc1608.zip)

Works perfect and would be great to add it officially, with every update I have to patch manually, I attach the datasheet aswell in case it can be improved

[JHD240128-G02PFWD-B1.pdf](https://github.com/olikraus/u8g2/files/10375280/JHD240128-G02PFWD-B1.pdf)
[u8x8_d_uc1608_modified.zip](https://github.com/olikraus/u8g2/files/10375293/u8x8_d_uc1608_modified.zip)

olikraus, 2023-01-09:
The current file uc1608 contains more than 500 lines:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1608.c

However your code only contains 355 lines. I have no I idea what you modified. So I can't add this to the repo.

srcastor, 2023-01-09:
Sorry, I must not have been clear enough,  The original file it's from this (year 2017) comment https://github.com/olikraus/u8g2/issues/371#issuecomment-333556809

to make it clear:
jchifos pick 2017 u8x8_d_uc1608.c and modified to match UC1638 (I don't know how much modified)
Me modified jchifos's file to adapt to  240x128 JHD240128 screen (Line 187,191  and 199, just 3 values)

I don't know if you can just adapt my file as U8G2_UC1638C_240X128 or something, the current U8G2_UC1638_192X96 is not working  with my model (empty  screen) besides also the different resolution.

PS: I just checked the current version against jchifos version and seems the major difference starts on line 311 to the end with the uc1608 erc240x120 and uc1608 dem240064 new models , if you compare both files without thoose models you can see jchifos changes

olikraus, 2023-01-09:
ok, i am still confused. Let me first create a new issue