olikraus/u8g2 #372: support for SSD1309 128x32 OLED HW SPI

max-bishop, 2017-09-29:
I'm putting in a request for support for SSD1309 128x32 OLED HW SPI

The current library supports the following constructors and I was hoping you would be able to support a slightly different resolution.

```
U8G2_SSD1309_128X64_NONAME0_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
U8G2_SSD1309_128X64_NONAME2_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
```

Fingers crossed :)

olikraus, 2017-09-29:
so, what happend?

max-bishop, 2017-09-29:
Nothing :)

The screen was completely blank.

The OLED that I'm using has the same pinout as an identical OLED with a SSD1305 chip that you helped me with a week ago. I used the same ZIF connection to the Arudino for that display and it works so I don't think it's my wiring.

olikraus, 2017-09-30:
Currently there is an issue with the F constructor. Can you try a page buffer example?
Also: Try the SSD1306 examples for 128x32.

olikraus, 2017-09-30:
Unfortunately full buffer mode is broken at the moment (#370) please use page buffer examples for your tests.

olikraus, 2017-09-30:
The full buffer mode is now fixed with 2.18.13 (should be available soon in Arduino Library Manager).

olikraus, 2017-10-02:
Is something working now, how shell we continue?

max-bishop, 2017-10-02:
Thanks for your speedy reply! I just updated the library to 2.18.13 this morning.

The following constructors are still not working:

```
U8G2_SSD1309_128X64_NONAME0_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
U8G2_SSD1309_128X64_NONAME2_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
```

I also tried the following SSD1306 constructors and they also did not work:

```
U8G2_SSD1306_128X32_UNIVISION_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 6, /* dc=*/ 7, /* reset=*/ 9);			
U8G2_SSD1306_128X32_UNIVISION_F_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 6, /* dc=*/ 7, /* reset=*/ 9);			
U8G2_SSD1306_128X32_UNIVISION_F_3W_SW_SPI u8g2(U8G2_R0, /* cs=*/ 6, /* dc=*/ 7, /* reset=*/ 9);	

```

The good news is I did try the SSD1305 constructor and it worked!

`
U8G2_SSD1305_128X32_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 6, /* dc=*/ 7, /* reset=*/ 9);`

olikraus, 2017-10-02:
ah, ok... cool. So we can close this issue?

max-bishop, 2017-10-02:
Yes let's close it.

Other users can see the notes re: using the SSD1305 constructor.

closed 2017-10-02 by max-bishop

olikraus, 2017-10-02:
:-)