olikraus/u8g2 #374: support 256x128 LCD

aradarbel10, 2017-09-30:
for example: 256x128 LCD HYUNDAI HG25604-B from All Electronics

olikraus, 2017-09-30:
Reference: https://forum.arduino.cc/index.php?topic=503084

Let me know whether you want to use u8g2 for this display. I can just offer to implement support for this dimension (and remember, this is a open source project and I do this for free in my spare time), but I need you for testing.

olikraus, 2017-10-07:
Any update on this topic?

aradarbel10, 2017-10-07:
Not yet, probably tommorw Ill go to few shops and buy everything I need. In
israel all the stores are close on saturday.

olikraus, 2017-10-07:
Great, I just was about to clean up this issue list. I will wait for your input and keep this open.

aradarbel10, 2017-10-11:
Hello again,
as you said, I got a power supply of -20v.
just using a normal DC power supply (MCH-303A)...
but in chapter 7 there is a potentiometer, should I ignore that?
anyways,  I tried to connect it to  -21v and its working! all of the screen is turned on! Now the hard part... 
which is the code. I uploaded the u8g2 hello world sample with the right constructor and it sends a weird error:

"vrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt X of 10: not in sync: resp=0x00"

And while trying to upload (unsuccessfully...) the display is kinda blinking...
also, finally its sending a message says that it can't upload the sketch to the board.
I am on the Arduino uno board and COM3 so this is not the problem. is it a problem with the hardware?

Arad.

olikraus, 2017-10-11:
Well, the potentiometer is there for contrast adjustment. So usually it is required, but for testing, a fixed source might be ok.

> avrdude: stk500_getsync() attempt X of 10: not in sync: resp=0x00"

This is not u8g2 specific. You should be able to program the Arduino. Maybe you have selected the wrong port.

olikraus, 2017-10-29:
no feedback... can this be closed?

aradarbel10, 2017-10-29:
Im sorry, but still no... as I mentiond before, I had a problem with my arduino, it will take more time till I recive it from ebay... I dont even know when I will recive it and cant do anything about it.

olikraus, 2017-10-29:
ok, no problem

olikraus, 2017-12-30:
can i close this issue? You can reopen or open a new issue, if required...

closed 2018-03-24 by olikraus