olikraus/u8g2 #375: how kerning works for arabic ?

smarty551, 2017-10-02:
i found a kerning table for this Arabic font here so how can i use it ?
http://www.adambaker.org/kerning_font.php
thank you

olikraus, 2017-10-02:
I once started to implement kerning, but finally failed. Kerning tables get too big and will not fit into RAM/ROM of small embedded controllers. So finally, I removed kerning procedures. However some might still be there.

There is a small workaround, described in one of the issues.

olikraus, 2017-10-02:
some details are here: #42

smarty551, 2017-10-02:
ok i will try to get around kerning in arabic and pass it to you

smarty551, 2017-10-02:
but the main problem that it has to be font specific even that i was thinking about a loop with all the conditions of kerning (the same one you were talking about in the #42 )

olikraus, 2017-10-03:
Yes, the current solution is very specific. But anything else will generate very huge tables. The will not fit into the uC ROM.

smarty551, 2017-10-03:
sorry but have you tried to work it with the same idea of bdfconv that the user will have to put the text in file then generate array that will be fed into drawstr

olikraus, 2017-10-03:
no
during my own experiments, i just generated the kerning between all glyphs of the font (however, number of glyphs can be selected/restricted). The problem still is, that the table gets too big.
I finally stopped working on this.

olikraus, 2017-10-07:
I know this is not a good answer, but I do not see how this can be implemented on small uC systems.

closed 2017-10-07 by olikraus