olikraus/u8g2 #376: Seeking help setting up u8g2 for dsPIC33E

soundscu, 2017-10-03:
u8g2 is a phenomenal piece of work, many thanks for the effort.

I am seeking assistance in getting u8g2 up and running on a Microchip dsPIC33EP256MU806, hardware I2C interface, and Winstar OLED 128x64 display with SSD1309 controller.

Prior to finding u8g2, I created a simple putc() function that I'm able to call with printf() for 8-line x 21-character "terminal emulation."  Unfortunately, my experience with c is functional but rudimentary, and I am struggling to understand how to create the required u8g2 HAL for the dsPIC.  

Any assistance would be very much appreciated, and I would be happy to pay something for it.

Many thanks in advance for help.

rbresca, 2017-10-03:
I'm trying to port the library to dsPic33EP256MC502 right now.
I'm using MplabX and XC16 (free version).
At the moment I'm able to compile (I had to remove almost all the fonts and the setup functions).
I want to use a SSD1306 oled display.
I'm also able to send data to the display via I2C.

The only problem is that the screen is somewhat corrupted. In simulation everything seems ok, but, with the real thing, I send wrong data.
I've also forced the content of the buffer to exclude the drawing procedures from the problem.
[image]

In the attached image, the line should be continuos, from (0,0) to (127,63).
The same is true for everything I try to draw on the screen. 

Programming micro is only a hobby for me, so I don't know how to proceed. x16 bug? a paging error in the memory? The interrupt function which manages the I2C state machine (autogenerated with MCC)?

soundscu, 2017-10-03:
This could be a problem at the hardware level or low level spi configuration.  I have a lot of experience in this area, and an excellent scope for confirming data integrity in the i2c bus.

Would you be willing to share your HAL with me?  We may be able to solve each other's problems.

rbresca, 2017-10-03:
No problem. Tomorrow morning I'll try to reduce the zip dimension and send the code.
By the way, I have a Picoscope. It decodes I2C data and the flow seems to be correct (I've also compared the output with an Arduino with the same display, which works without problems).

soundscu, 2017-10-03:
There must be something that is not identical between your pic33 and the
arduino. Things to look at:
- Pull-up resistors of appropriate value are required on both i2c lines.
- Is the clock speed the same, or similar, for both? What is it?
- There might not be adequate time between bytes. Are you checking status
bits to see completion of send, or waiting a predetermined time period for
each byte, or a combination of these, to ensure each byte is sent in it’s
entirety and the display has time to process it?
- You mentioned interrupt handling. Many people have difficult with this, I
can certainly help with it once I see the code.

olikraus, 2017-10-03:
Just a note from my side: Microchip C Compiler does not support removal of unused functions and data structures in the free version of their tool chain. This makes it a little bit difficult to handle u8g2, because you have to remove unused functions and data structures manually.

soundscu, 2017-10-03:
I have a paid full license for xc16.  If I can contribute anything helpful
by using my licensed compiler, let me know.
I’ve created a u8g2 project in MPLAB X, added i2c support files, and
successfully compiled. All I need is a working HAL between u8g2 and i2c to
make it actually work.

rbresca, 2017-10-04:
[Encoder16.X.zip](https://github.com/olikraus/u8g2/files/1354755/Encoder16.X.zip)
 As olikraus noted, I had to remove all unused setup function to avoid allocating memory.

The communication hardware should be ok. The oscilloscope signal is very good.
I think I'm using the correct code to manage the bus.
Using a bigger delay I can see that, for each page, I draw the first 23 bytes correctly, then I miss a block (the rightmost part of the page is shiftted to the left). Then I repeat the last part of the page.
The process repeats for each page.

So, something, corrupts the index of the pointers of the tiles. Maybe in the function u8x8_cad_ssd13xx_i2c() where the page is sent in 24 bytes packets.

rbresca, 2017-10-04:
Right to the point. I've changed the u8x8_cad_ssd13xx_i2c() code to send the complete page and now the display is shown ok.

p = arg_ptr;
       while( arg_int > 130 )
      {
	u8x8_i2c_data_transfer(u8x8, 130, p);
	arg_int-=130;
	p+=130;
      }

(130 before was 24).
I've allocated a big buffer for I2C communication so I can send the whole page in one shot

olikraus, 2017-10-04:
Standard Arduino Core libraries only allocate a very small buffer, so I was forced to work with much smaller buffers.

soundscu, 2017-10-05:
I have it working for dsPIC33EP256MU806 on I2C2 with a SSD-1309 controller/OLED.  Changing the I2C port is trivial enough that I haven't bothered to upload -- it's easy enough for anyone to find and change the few things that need adjustment for their particularly hardware.  ** Many thanks to both of you for getting this to work! **

Now I'm trying to turn all of it into a library file for use within a larger project.  I'm stumbling a bit with this, but hope to have it figured out over the next few days.  This is not something I've done much of, so the learning curve is slowing me down.

Thank you again for getting u8g2 to work with pic33.

Jim (Soundscu)

rbresca, 2017-10-08:
I've found some time to play with my project.
I've found two issues. 
First, the stat variable, used by the I2C functions must be static. Since the I2C routines are interrupt based, the variabile could be updated even if the caller function returns. If the stat variable is in the stack, the same memory area will be used by other functions, corrupting them.
Second, I left a default timeout in the I2C send data routine. It is too short for the 128 packet. For this reason, the function u8x8_byte_pic_hw_i2() returns 0 and, as a consequence, the transmission is not ok.

soundscu, 2017-10-13:
I apologize, I think this is a "dumb" question, but I'm stuck.

When I attempt to use any font other than u8g2_font_logisoso16_tf, my compiler throws an undefined reference error.  The one font that works is the one used by rbresca in his simple test code. 

I see all the font references in the header file.  Am I missing the actual font data?  What do I need to do to access the font library in the u8g2 documentation? 

Many thanks.

soundscu, 2017-10-13:
I think I have answered my own question.  Font files are here:
https://github.com/olikraus/u8g2/tree/master/tools/font/build/single_font_files

rbresca, 2017-10-14:
yes. I had to remove all fonts, otherwise the compiled file is too big for the PIC.
The free version of Pic compiled do not automatically remove unused objects.
You have to add manually the  fonts you want to use

olikraus, 2017-11-24:
clean up for the issue list

closed 2017-11-24 by olikraus