olikraus/u8g2 #378: UC1601

thom-ek, 2017-10-04:
Hi,
I started to implement support for UC1601 and my results are like on attached screenshot. I started by duplicating code from UC1604 which looks similar. I draw two lines crossing from corners, but only bottom 8 lines are visible and from left I see 4 pixels wide random dots - you can see that line starts from 5th column instead of 1st). When I make some experiments I was able to put random pixels on whole display, so the display isn't broken I guess (there is probably a hardware problem with 9th column - it's always black). I also tried to make hardware flip - content has been flipped, but still same bottom part of display is only working. I'm using esp8266 and SPI 3W.
 
[image]

Can you help me adding support for this display. I can send you a datasheet for it. I saw that u8glib (first version) had support for this display but unfortunately this old lib crashes on esp8266 (I'll try to make some testing on atmel chips to see how it's working).

olikraus, 2017-10-04:
Can you also test with 4W SPI? 3W SPI is not so much tested. I assume 4W SPI is more reliable. I just want to avoid extra bug due to the interface.

I will have a look at the u8glib code and create a new device for this, once I am back in my lab.

thom-ek, 2017-10-04:
I'll try 4W this night and give you a feedback.

olikraus, 2017-10-04:
Can you also give me more details on the display? What is the size? Any datasheet?

thom-ek, 2017-10-04:
Datasheet: http://www.artronic.com.pl/pdf/pl/UC1601.pdf
Display size: 128x32

thom-ek, 2017-10-05:
Hint - I found that UC1601 should be compatible with ST7565 which you already support (KS0108 too), but I've tried all of ST7565's and they don't work (no display). I've checked 4W HW & SW on my implementation - no display, on 3W it's working. 
While making some experiments I sent direct commands to set position and write some bytes to it - I was able to fill whole display. I think something must be wrong with U8X8_MSG_DISPLAY_DRAW_TILE and memory mapping. I've copied draw_tile from ST7565 and still, only 8 bottom rows are displayed.

My working simple test (fills left half of screen with y bits, right half of x bits and 4 banks of 8 pixels height):

	void setPixel(u8x8_t *u8x8, int x, int y, uint8_t a)
	{
	    u8x8_cad_SendCmd(u8x8, 0x000 | ((x&0x0f)));
	    u8x8_cad_SendCmd(u8x8, 0x010 | ((x&0xf0)>>4) );
	    u8x8_cad_SendCmd(u8x8, 0x0b0 | (y));
	    for(int i=0;i<128;i++) {
	      uint8_t d = i < 64 ? a : i;
	      u8x8_cad_SendData(u8x8, 1, &d);
	    }
	}
	
	uint8_t u8x8_d_uc1601_common(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
	{
	  uint8_t x, y, c;
	  uint8_t *ptr;
	  switch(msg)
	  {
	    case U8X8_MSG_DISPLAY_DRAW_TILE:
	      u8x8_cad_StartTransfer(u8x8);
	        for(int y=0;y<4;y++)
	            setPixel(u8x8, 0, y, y+1);
	
	      u8x8_cad_EndTransfer(u8x8);
	      break;
	  }
	}

olikraus, 2017-10-05:
Unfortunately I was not able to work on this yesterday. Hope to spent some more time today on this.

thom-ek, 2017-10-05:
no problem :)

olikraus, 2017-10-05:
I checked the UC1601 controller. Well, there are some big differences to the ST7565..

https://github.com/olikraus/u8g2/blob/master/doc/controller_cmds.txt

Next step: Create u8g2 device for the UC1601s (your pdf refers to UC1601s)

thom-ek, 2017-10-05:
Init procedure works quite well for - even when sending unsupported commands like bias. Memory mapping should be same or very similar on all of them, but something must be a little different. One of differences might be that internally UC1601 has 132x65 pixels, not 128x64 like LCD itself?

olikraus, 2017-10-05:
Actually ST7565 also has 132x65 pixel.

olikraus, 2017-10-05:
Somehow the command sequence from u8glib does not match your datasheet:
```
static const uint8_t u8g_dev_uc1601_c128032_init_seq[] PROGMEM = {
  U8G_ESC_CS(0),            	/* disable chip */
  U8G_ESC_ADR(0),           	/* instruction mode */
  U8G_ESC_CS(1),             	/* enable chip */
  U8G_ESC_RST(15),           	/* do reset low pulse with (15*16)+2 milliseconds (=maximum delay)*/

  0x0a3,		        /* 0x0a3: LCD bias 1/7 ,  0x0a2: LCD bias 1/9  --> invalid for UC1601s*/
  0x0a0,		        /* 0x0a0: ADC set to normal,  0x0a1 ADC set to inverted --> Framerate on the UC1601s */
  0x0c8,                        /* common output mode: set scan direction normal operation/SHL Select, 0x0c0 --> SHL = 0, normal, 0x0c8 --> SHL = 1 --> not supported on the UC1601*/
...  
```
Do you have any link or datasheet for the display itself?

olikraus, 2017-10-05:
I found a display with the same controller which includes the init sequence (section 5.3): http://www.midas-components.com/pdf/BTHQ128064AVC2.pdf
I will takeover this sequence

olikraus, 2017-10-05:
https://github.com/olikraus/u8g2/commit/effacecf6936f2099a817c6be7fc0ce36fe06a26

thom-ek, 2017-10-05:
I've tried your commit and:
1. contrast should be in range up to 255 (at 128 it's almost nothing visible).
2. on display I see random dots - looks like you are writing to different memory (at range 32-63) than it is displayed. I added 4 to y_pos ad draw_tile and it drawn first 8 pixels from top (when I add 5 it renders at second 8px height row, at 6 it renders 3rd part and at 7 4th part, so I can render on while display). Similary I can change init from 0xc6 to 0xc2 and don't change y_pos and I've got same effect but bottom 8 pixels are drawn. I saw when chip memory is not written it displays random pixels.
3. for testing - I draw on first 32 rows different image than on second 32 rows. When I change display size at display_info structure it renders on display different part of buffer, but still only one row (8 pixels).

olikraus, 2017-10-06:
wow, you are fast. I have not even checked the code by myself. Which is your constructor call?

thom-ek, 2017-10-06:
As I wrote at the beginning I started own implementation, so I only replaced your file as you didn't modify other files yet :) Constructors I have copied from uc1604 implementation.

olikraus, 2017-10-06:
I have created a beta release:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Install this zip file via the Arduino IDE add zip file menu of the library manager.

The example will include the constructor for the UC1601 and 3W SPI.
I have increased contrast and shifted display by 32 lines (0x060). How will it look?
Maybe you can send me a picture of the hello world example.

thom-ek, 2017-10-07:
You are brilliant! Problem with displaying only part of screen was because I've been using clearBuffer() and sendBuffer() pair, instead of firstPage() and nextPage(). But I think it should work at both modes? After change it's working perfectly! There is still problem with contrast. I have to remove shifting by 2. Display uses full byte range, but only from 180 to 255 range is visible. Also I had to change ram mapping from 0xc6 to 0xc4 because it was flipped horizontally.

I've checked 4W SW and 4W HW - both are working (there was cabling problem), but... when using HW SPI display is mirrored vertically - changing ram mapping has no effect - very strange. 

This display also has I2C mode, so You can add constructor for it.

[image]

olikraus, 2017-10-07:
Looks good. I have create a new release 2.19.5:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Can you test the u8x8/flipmode example? The text should start in the upper left corner in both modes.

thom-ek, 2017-10-08:
1. ESP8266 has no SPI.begin() with pin parameters, it's same like in AVR. Change line 382 at U8x8lib.cpp to: #if defined(ESP_PLATFORM) || defined(ARDUINO_ARCH_ESP32).
2. FlipMode(1) SW SPI renders random pixels: https://www.dropbox.com/s/45tvkjdx38lmiey/Wideo%2008.10.2017%2C%2002%2015%2003.mov?dl=0
On HW SPI only always random pixels:
https://www.dropbox.com/s/m2mdhs04ijvjw1u/Wideo%2008.10.2017%2C%2002%2023%2053.mov?dl=0
3. I2C - could not get it to work. I don't know how addressing works and what value I have to set.

olikraus, 2017-10-08:
1. Oh, i think it got changed. It was like this in the history.

new doc: http://arduino-esp8266.readthedocs.io/en/latest/libraries.html
https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.h#L56

2. 
SW SPI: ok, guess the offset is still wrong for the flip mode, i will change this.
HW SPI: Strange, maybe the SPI mode is wrong. There are actually four modes (0 .. 3). Can you modify them and test, which one works?
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1601.c#L84

3.
Usually it is setI2CAddress (if i remember correctly, but it is there in the u8g2 docs also). Yet I should ensure, that the default address for this device is set correctly. I will check this also.

thom-ek, 2017-10-08:
1. SPI.pins() is available from framework 2.4.0, but stable is 2.3.0 (it's used by platformio), so make a #ifdef for it's use.
2. spi_mode does not change anything. It looks like displaying invisible part of screen (next 32 pixels) like at flipmode(1).
3. But what value should I set? At display side I can only change one bit of address by CS0 pin.

olikraus, 2017-10-08:
ok, I guess i need to fix your issue mentioned in point 1.
Nevertheless I had created a change request for this:
https://github.com/esp8266/Arduino/issues/3691

olikraus, 2017-10-08:
I2C address:
Bit pattern for the address is:
0 1 1 1 cs1 cs0 x y
x y are set by the u8g2 device, but cs0 and cs1 have to be known. cs0 and cs1 are used as chip select for SPI interface. cs0 is low active, cs1 is high active. I guess only cs0 is available to the outside interface, so cs1 should be assigned to 1 on the display module itself:

0 1 1 1 1 cs0 x y

For proper functionality you must connect cs0 to a valid logical level. Let me assume, you connect cs0 to ground (either directly or by digitalWrite( <pin-nr>, 0), then the address is:

0 1 1 1 1 0 x y

U8g2 will take care on x and y so we have:

0 1 1 1 1 0 0 0

which is 0x78

So there might be two problems: 
1. The default address is not set to 0x078
2. You need to connect cs0 to ground, it must not float when using I2C interface.

olikraus, 2017-10-08:
I have updated the code and created a new beta release 2.19.6:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

thom-ek, 2017-10-08:
Almost good :) When flip(1) screen is moved 4px right. See: https://www.dropbox.com/s/a34blmrgdcwhsmo/Wideo%2008.10.2017%2C%2018%2027%2038.mov?dl=0

thom-ek, 2017-10-08:
I2C - after setting address to 0x78 it's working at SW mode, but with very high error rate. HW mode not working.

olikraus, 2017-10-08:
looks good. I have fixed the x offset and lowered timing for SPI and I2C. New beta 2.19.7 created:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

thom-ek, 2017-10-08:
1. SW SPI - works perfect! You are great! 🥇 
2. HW SPI - I've got it to work! My fault was to use MISO pin for DC... what a shame ;)
3. SW I2C - no change, tried to call Wire.setClock to lower speed, but this has no change.
4. HW I2C - got it to work (again, cabling problem) but works like SW mode. BTW: ESP8266 in fact doesn't have hardware I2C. Only ESP32 has hardware for it. 

I've checked same code also on ATmega and SW 4W is working ok, HW SPI can't check due to pins used by programmer, I2C SW also has high error and I2C HW renders top line and CPU stops responding (watchdog led not flashing). I think there must be cabling/grounding/etc problem with my prototype board and we can give up. Also, I've found disclaimer in datasheet that they have no license to use I2C, so the hardware maker must obtain such license itself - this was at 2007, so it may already be not true.

So, thank you very much for spending time to make this display working for me. I'm very thankful :)

olikraus, 2017-10-09:
Great. Thanks a lot for all the testing effort. I am happy, that we finally have support UC1601 within u8g2.
Thanks!

thom-ek, 2017-10-09:
If you will want some more testing, I'm ready to go :) Now, we may close this issue.

closed 2017-10-09 by thom-ek

olikraus, 2017-10-09:
It it works for you, then I think testing is good enough so far. :-)