olikraus/u8g2 #381: ESP-IDF (ESP 8266 & ESP32) Component Support

neilpanchal, 2017-10-09:
I've come across an interesting roadblock.

[Arduino-esp32](https://github.com/espressif/arduino-esp32) dictates 4 ways of using the Arduino API with ESP-IDF in the instructions/readme.

## Method 1
One of those ways is to use Arduino API as an ESP-IDF component as described [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md). I managed to add Arduino API to ESP-IDF using this method. Next, I've added a component.mk file to @olikraus's U8g2_Arduino [fork](https://github.com/neilpanchal/U8g2_Arduino). 

Component.mk has the following options to add source directors and include files:
```
COMPONENT_SRCDIRS:=src src/clib
COMPONENT_ADD_INCLUDEDIRS:=src src/clib
```
Now, U8g2 is a component of ESP-IDF and we can use the wonderful `make menuconfig` utility that ships with ESP-IDF to add configuration parameters such as SPI ports or selection between VSPI & HSPI.

Everything works except **There is one problem**. Usually, I am able to edit the SPI frequency in the [u8x2_d_ssd1306_128x64_noname.c](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x64_noname.c#L235), however, no matter what frequency I set and upload to ESP32, I always get 8MHz as the clock freq. I've validated it using a scope.

## Method 2
Another way to use ESP-IDF + Arduino API is to leverage the [Platform IO ESP32](http://docs.platformio.org/en/latest/platforms/espressif32.html)  framework. So, I tried the same test as above and I changed the `u8x8_d_ssd1306_128x64_noname.c LINE 235` to use 10 MHz as the clock frequency. This time, using Platform IO toolchain, it works! I checked it with a scope and I am getting 10 MHz as the clock frequency.

I am pretty sure I am missing something obvious here. 

Any ideas on what must be going on? Any chances I make to the `u8x8_d_ssd1306_128x64_noname.c` using Method 1 gets ignored when I compile the source. Method 2 works flawlessly and I can get a much better performance by tweaking  the `u8x8_d_ssd1306_128x64_noname.c`

olikraus, 2017-10-09:
ok, i think it is good to have this documented here, but is there anything I can do within u8g2?

neilpanchal, 2017-10-09:
@olikraus 

I am scratching my head and it looks like no matter what I set here, it makes no difference to the SCLK freq:
```
void SPIClass::beginTransaction(SPISettings settings)
{
    //check if last freq changed
    uint32_t cdiv = spiGetClockDiv(_spi);
    if(_freq != settings._clock || _div != cdiv) {
        _freq = settings._clock;
        _div = spiFrequencyToClockDiv(10000000UL); // Directly passing 10MHz instead of relying on U8g2 to set it and it still doesn't work!
    }
    spiTransaction(_spi, _div, settings._dataMode, settings._bitOrder);
    _inTransaction = true;
}
```

neilpanchal, 2017-10-09:
@olikraus Yes, I need to file this issue with ESP32-Arduino guys. Thanks.

closed 2017-10-09 by neilpanchal