olikraus/u8g2 #382: u8g2lib porting stm32f103(arm cortex-m3)

leesohaeng, 2017-10-10:
I started a new project today. 
The processor is stm32f103c8t6(Cortex-m3). 
The device i will use is SH1106(1.3 OLED). 
So I want to use u8g2lib, but I don't know which file to add to the project. 
I'm looking for an example, but I can't find it. 
So I ask you to help me with this.

please, help.

thanks.

olikraus, 2017-10-10:
There is a porting guide on the U8g2 wiki.
You need only files from the csrc directory.

leesohaeng, 2017-10-10:
thanks, olikraus.

Copy all files in the directory. (\u8glib-master\u8g2\csrc )

my source code is here....

..........
  // USER CODE BEGIN 2
  u8g2_Setup_sh1106_128x64_noname_f(u8g2, U8G2_R0, 8x8_byte_4wire_sw_spi,u8x8_gpio_and_delay_lpc11u3x);

  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
  u8g2_SetPowerSave(&u8g2, 0); // wake up display

  HAL_Delay(100);
..........

but, Four errors and three warning occur when compiled.

-errors-
'U8G2_R0' undeclared (first use in this function)
'u8x8_byte_4wire_sw_spi' undeclared (first use in this function)
'u8x8_gpio_and_delay_lpc11u3x' undeclared (first use in this function)

-warnings-

implicit declaration of function 'u8g2_InitDisplay' [-Wimplicit-function-declaration]
implicit declaration of function 'u8g2_SetPowerSave' [-Wimplicit-function-declaration] 
implicit declaration of function 'u8g2_Setup_sh1106_128x64_noname_f' [-Wimplicit-function-declaration]

please, some more information..

thanks.

olikraus, 2017-10-11:
looks like you forgot to include u8g2.h

leesohaeng, 2017-10-11:
'u8g2.h' is included precisely.

olikraus, 2017-10-11:
If you use 
#include "u8g2.h"
at the beginning of all your files, then this error will not appear.

closed 2017-10-12 by olikraus