olikraus/u8g2 #383: UC1701X is shifted left ...

miro1360, 2017-10-12:
I bought this one:
http://www.ebay.com/itm/12864-128X64-Serial-SPI-Graphic-COG-LCD-Display-Screen-Build-in-LCM-Module-/272581767692
they mentioned this IC: UC1701X

In HW mode display is blank, nothing here.

In SW mode right side of screen is displayed wrong: https://ctrlv.cz/shots/2017/10/12/dpqW.png
I connected it with arduino mega 2560 with constructor:
U8G2_UC1701_EA_DOGS102_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 51, / data=/ 50, / cs=/ 52, / dc=/ 9, / reset=*/ 8);

What am I doing wrong? Or IC UC1701X is not supported? If so, can you add it into list?

Thanks for reply.

olikraus, 2017-10-12:
Ebay descriptions sometimes contains wrong information. The name of the display is mentioned as JLX12864, so maybe you could try
U8G2_ST7567_JLX12864_F_4W_SW_SPI
or
U8G2_ST7588_JLX12864_F_4W_SW_SPI

You could also try other UC1701 compatible devices, like the
U8G2_UC1701_MINI12864_F_4W_SW_SPI
or even any other ST7565 based device with the same resolution (UC1701 is compatible to ST7565)

miro1360, 2017-10-12:
Thank you Oli, I finally found it:
U8G2_UC1701_MINI12864_F_4W_SW_SPI ... this one is working perfect, actually my problem was in bad contrast, after setting contrase at about .setContrast(150) it is working perfect, thanks for it

olikraus, 2017-10-12:
Greatl. 

Regarding HW SPI: Did you connect the LCD to the correct SPI port? Note that mega2560 has two different SPI subsystems. There is a different u8g2 constructor for the second SPI port.

miro1360, 2017-10-12:
Yes I did it, mega2560 have separated SDA/SCL pins https://ctrlv.cz/shots/2017/10/12/uJcI.png
and also I changed u8g2 constructor for HW SPI, I tried them all, one after another with contrast change, but without success ...

olikraus, 2017-10-12:
HW SPI should work without problem, nevertheless, if SW SPI is good enough, then we can close this issue, right?

miro1360, 2017-10-12:
things like baro sensor and rtc timer are working in this SPI, maybe it is specific problem with arduino mega2560, in future I try it with bigger LCD and give you reply about result ... yes, I am fully satisfied with SW SPI and you can close this issue, thank you for your library

closed 2017-10-12 by miro1360