olikraus/u8g2 #386: German Umlauts: getStrWidth() returns differnt values for characters with same width

luebbe, 2017-10-13:
Hi folks,
I stumbled across this with one font. Haven't tried with other fonts, so I can't tell whether it's a problem with this specific font.
I'm using u8g2_font_helvR18_tf to draw centered text. When the text contained umlauts, it wasn't centered anymore.
* getStrWidth('AOU') returns 51 Pixel
* getStrWidth('ÄÖÜ') returns 67 Pixel

Only tested with this one font
Does getStrWidth() count hidden UTF-8 bytes as well?

Cheers & thanks

olikraus, 2017-10-13:
Which values will you get with getUTF8Width?
 *   getUTF8Width('AOU') 
 *   getUTF8Width('ÄÖÜ') 

Applying unicode string to getStrWidth will probably lead to unexpected results.

luebbe, 2017-10-13:
It obviously  does :)
getUTFWidth did the trick. I'm usin drawUTF already. Should have thought of this ...

Thanks for your help and sorry for the fuzz

closed 2017-10-13 by luebbe

olikraus, 2017-10-14:
:)