olikraus/u8g2 #388: Using CS Pin to use two SPI Displays

neilpanchal, 2017-10-15:
Hello,

I am trying to use 2 displays using the chip select (CS/SS) pin. I am sharing the same MOSI, RST, DC, SCLK pins. 

Here is my code:
```
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI display1(U8G2_R0, /* CS = */ 16, /* DC = */  17, /* RS = */  4);
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI display2(U8G2_R0, /* CS = */ 2, /* DC = */  17, /* RS = */  4);

void setup(void) {
    display1.begin();
    display2.begin();
}
void loop(void) {
	displayText(&display1);
	displayText(&display2);
}

void displayText(U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI *display) {
	display->clearBuffer();
	display->setCursor(64,32);
	display->print("HELLO WORLD");
	display->sendBuffer();
}
```

Only `display1` works. The `begin()` call doesn't work for both displays. If I switch the order of the `begin()` call like this:

```
void setup(void) {
    display2.begin();
    display1.begin();
}
```
Now only `display2` works. 

Using `begin()` inside the `void loop(void)` function results in blank screen on both `display1` and `display2` as follows:

```
void loop(void) {
	display1.begin();
	displayText(&display1);
	display2.begin();
	displayText(&display2);
}
```

I am using ESP32-arduino framework. Reading on the internet, CS/SS pin on Arduino is fixed to 10. I don't know why! Therefore, there is no CS/SS pin that can be passed to the `begin()` function. 

Even though my example code displays the same thing on both displays, goal is to be able to display different things by alternating frame rendering on both displays.

Any ideas?

neilpanchal, 2017-10-15:
Perhaps the problem is here?

```C++
      /* issue #377 */
      /* issue #378: removed ESP8266 support, which is implemented differently */
#if defined(ESP_PLATFORM) || defined(ARDUINO_ARCH_ESP32)
      /* ESP32 has the following begin: SPI.begin(int8_t sck=SCK, int8_t miso=MISO, int8_t mosi=MOSI, int8_t ss=-1); */
      /* not sure about ESP8266 */
      if ( u8x8->pins[U8X8_PIN_I2C_CLOCK] != U8X8_PIN_NONE && u8x8->pins[U8X8_PIN_I2C_DATA] != U8X8_PIN_NONE )
      {
	/* SPI.begin(int8_t sck=SCK, int8_t miso=MISO, int8_t mosi=MOSI, int8_t ss=-1); */
	/* actually MISO is not used, but what else could be used here??? */
			SPI.begin(u8x8->pins[U8X8_PIN_I2C_CLOCK], MISO, u8x8->pins[U8X8_PIN_I2C_DATA]);
      }
      else
      {
			SPI.begin();
      }
#else
      SPI.begin();
#endif
```
Does this block of code in U8x8lib.cpp run separately for each display? Pardon my C++ ignorance, but once #ifdef executes, it won't set the SPI.begin() pins again. Is that true?

olikraus, 2017-10-15:
Can you test the same with software SPI constructor? I mean will it work with SW_SPI?

neilpanchal, 2017-10-15:
I have an error in my first post. Correction:
```C++
void setup(void) {
	display2.begin();
	display1.begin();
}
```
Result: `display1` is on. `display2` is off.

```C++
void setup(void) {
	display1.begin();
	display2.begin();
}
```
Result: `display2` is on. `display1` is off.

@olikraus 
Just tired using SW_SPI constructor. Here is my code:

```C++
// u8x8_SetPin_4Wire_SW_SPI(getU8x8(), clock, data, cs, dc, reset);
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI display1(U8G2_R0, /*CLK*/ 18, /*MOSI*/ 23, /*CS*/ 16,
                                                /*DC*/ 17, /*RS*/ 4);
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI display2(U8G2_R0, /*CLK*/ 18, /*MOSI*/ 23, /*CS*/ 15,
                                                /*DC*/ 17, /*RS*/ 4);

void displayFrame(U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI *pDisplay, int i, String *displayName);

extern "C" void app_main() {
    initArduino();
    display1.begin();
    display2.begin();
    while (1) {
        String display1Name = "DISPLAY 1";
        String display2Name = "DISPLAY 2";
        for (int i = 0; i < 60; i++) {
            displayFrame(&display2, i, &display1Name); // This function is same as my original post, but now displays "DISPLAY X" text on the oled itself to identify.
            // delay(10);
            displayFrame(&display2, i, &display2Name);
            // delay(1000);
        }
    }
}
```

Now, I see `display2` on but it is flipping between the two frames. 
Video here: https://www.youtube.com/watch?v=XvWrbEYkOUI

This makes zero sense! I am passing a pointer to `display1` to `displayFrame()` but `display2` shows both - "DISPLAY 1" and "DISPLAY 2" text.

neilpanchal, 2017-10-15:
I switched back to HW_SPI and hooked up a logic analyzer to see what's going on. I don't have a way to debug the ESP32 at the moment. I slowed down the SPI frequency to get more samples - frequency set to 100 kHz.

### DISPLAY 1
`display2` is commented out. This is a single shot event in `void setup()` that renders some text.

```C++
    display1.begin();
    // display2.begin();

    display1.clearBuffer();
    display1.setCursor(64, 20);
    display1.print("DISPLAY 1");
    display1.sendBuffer();

    // display2.clearBuffer();
    // display2.setCursor(64, 20);
    // display2.print("DISPLAY 2");
    // display2.sendBuffer();

```
[image]

### DISPLAY 2
`display1` rendering code is commented out. Results make sense as the `CS - D2` line goes low during data transfer.
```C++
    // display1.begin();
    display2.begin();

    // display1.clearBuffer();
    // display1.setCursor(64, 20);
    // display1.print("DISPLAY 1");
    // display1.sendBuffer();

    display2.clearBuffer();
    display2.setCursor(64, 20);
    display2.print("DISPLAY 2");
    display2.sendBuffer();
```
[image]

### DISPLAY1 + DISPLAY2
Now, if I have both displays rendering as follows:
```C++
    display1.begin();
    display2.begin();

    display1.clearBuffer();
    display1.setCursor(64, 20);
    display1.print("DISPLAY 1");
    display1.sendBuffer();

    display2.clearBuffer();
    display2.setCursor(64, 20);
    display2.print("DISPLAY 2");
    display2.sendBuffer();
```

[image]

I get nothing on `display1`. Hmm.... `CS - D1` line should always remain high when rendering `display2`. 😕

olikraus, 2017-10-15:
Last Pic looks correct to me. Begin() will send a small init sequence together with a full clear of the window.
D1 clear. D2 clear. D1 send buffer. D2 send buffers. CS is set correctly.

neilpanchal, 2017-10-15:
@olikraus You're correct. I was able to verify this by adding delays between `begin()` and `sendBuffer()`. I think I have found the issue - it is with the second `display2` begin sequence. It doesn't register as a valid SPI code. I'll upload the data shortly.

olikraus, 2017-10-15:
What is a valid SPI code?

neilpanchal, 2017-10-15:
@olikraus The logic analyzer software has a SPI decoder. I can set two cursors and it will decode HEX/ASCII/BYTE message. I don't know why D2 `begin()` doesn't decode.

neilpanchal, 2017-10-15:
```C++
    display1.begin();
    display2.begin();

    display1.setCursor(64, 20); // Printing on D1
    display1.print("DISPLAY");

    display2.sendBuffer(); // Send buffer on D2
```
Is the buffer shared? I never printed anything with `display2` object but it shows up on the display 2.

olikraus, 2017-10-15:
One point, that just came into my mind: You probably require two reset lines also, because begin() will also execute a reset of the display. I think the delay in your logic analyzer between first and second clear is caused by that reset pulse.
Because D1 and D2 share the reset line, render sequence should be this.
- reset D1 & D2 (of course only D1 should be reset here)
- send Init sequence to D1
- clear D1
- reset D2 & D1 (here, only D2 should be reset). Now D1 forgets about how to display things and will stay blank
- send init sequence to D2
- clear D2
- send buffer to D1, But nothing will happen, because the init sequence for D1 got lost
- send buffer to D2... D2 should show content correctly

olikraus, 2017-10-15:
> Is the buffer shared? I never printed anything with display2 object but it shows up on the display 2.

yes exactly, but not because of the shared buffer, but because of the shared reset line

BTW: Buffers are indeed shared.

neilpanchal, 2017-10-15:
> One point, that just came into my mind: You probably require two reset lines also, because begin() will also execute a reset of the display.

👍 Good one. Let me try separating reset lines.

neilpanchal, 2017-10-15:
>  @olikraus The logic analyzer software has a SPI decoder. I can set two cursors and it will decode HEX/ASCII/BYTE message between the markers. I don't know why D2 `begin()` doesn't decode.

I found  the problem with the logic analyzer. Now it decodes it. There were two CS lines and I had set CS-D1 line for SPI decoder settings. It can only use one CS line.

olikraus, 2017-10-15:
Ok, if separation of the reset line is the answer, then i guess i should create an FAQ entry on this.

Q: With multiple SPI displays, which lines should be separate and which could be common?
A: Each display requires individual CS and reset lines. Clock and data lines can be shared.

neilpanchal, 2017-10-15:
@olikraus Confirmed working. 🍻

For anyone else having a go, here is the final code for reference:

```C++
    display1.begin();
    display2.begin();

    display1.setCursor(64, 20);
    display1.print("DISPLAY 1");
    display1.sendBuffer();

    display2.setCursor(64, 20);
    display2.print("DISPLAY 2");
    display2.sendBuffer();
```

neilpanchal, 2017-10-15:
@olikraus Thanks for the help!

closed 2017-10-15 by neilpanchal

olikraus, 2017-10-16:
No problem. I learned something new.

neilpanchal, 2017-10-16:
@olikraus I did a PR for FAQ. 

If anyone is interested in using a common reset pin for all displays, you can comment out the following in `U8g2_Arduino/src/u8x8_display.c`:

```C++
void u8x8_d_helper_display_init(u8x8_t *u8x8) {
    /* 2) apply port directions to the GPIO lines and apply default values for the IO lines*/
    u8x8_gpio_Init(u8x8);
    u8x8_cad_Init(u8x8);

    // /* 3) do reset */ 
// Comment the following 6 lines
    // u8x8_gpio_SetReset(u8x8, 1);
    // u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_MILLI, u8x8->display_info->reset_pulse_width_ms);
    // u8x8_gpio_SetReset(u8x8, 0);
    // u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_MILLI, u8x8->display_info->reset_pulse_width_ms);
    // u8x8_gpio_SetReset(u8x8, 1);
    // u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_MILLI, u8x8->display_info->post_reset_wait_ms);
}
```
and add it to your main.cpp just before `display1.begin()` like so:

```C++
    u8x8_gpio_SetReset(display1.getU8x8(), 1);
    u8x8_gpio_Delay(display1.getU8x8(), U8X8_MSG_DELAY_MILLI,
                    display1.getU8x8()->display_info->reset_pulse_width_ms);
    u8x8_gpio_SetReset(display1.getU8x8(), 0);
    u8x8_gpio_Delay(display1.getU8x8(), U8X8_MSG_DELAY_MILLI,
                    display1.getU8x8()->display_info->reset_pulse_width_ms);
    u8x8_gpio_SetReset(display1.getU8x8(), 1);
    u8x8_gpio_Delay(display1.getU8x8(), U8X8_MSG_DELAY_MILLI,
                    display1.getU8x8()->display_info->post_reset_wait_ms);

    display1.begin();
    display2.begin();

    display1.setCursor(64, 20);
    display1.print("DISPLAY 1");
    display1.sendBuffer();

    display2.setCursor(64, 20);
    display2.print("DISPLAY 2");
    display2.sendBuffer();
```

This will reset `display1` + any other displays sharing the RST pin.

olikraus, 2017-10-16:
Another option would have been to add the reset pin only to display1 and assign U8X8_PIN_NONE to display2. Of course the sequence 
```
    display1.begin();
    display2.begin();
```
must not be changed in this case.