olikraus/u8g2 #389: Use of "SetContrast"

Tulsa2008, 2017-10-15:
Olikraus:
I hate to bother you as I know you have a lot going on.

I am using your lib with constructor "U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2 (U8G2_R0, 3, 2, 4);"
and have had great success with it.   I am using two displays running on a Arduino Mega 2560 for the most part without any problems.  

I have found that it would be advantages to be able to change the contrast on the displays due to large temperature swings in ambient temperature.  I would like to use the function "u8g2_SetContrast(u8g2_t *u8g2, Value);"  I know that you can control the contrast of the display as I saw a reference to setting the contrast value in "U8g2lin.h" and in u8x8_d_st7565.c.  

I have tried to use the "SetContrast" function and cannot get it to compile.  If I use "u8g2.SetContrast(Value);" the compiler comes back and complains about "No such member named SetContrast".  If I use "u8g2_SetContrast(u8g2_t *u8g2, Value);" the compiler complains about wanting something before the "*".  I am using Arduino's IDE compiler.

Does the constructor I am using have the "SetContrast" function??  Is it possible to change contrast after the program is compiled??

Thank you in advance for any light you can shed on this issue

Ken

 

.

olikraus, 2017-10-15:
First, please note that C++ and C are case sensitive. so setContrast is different from SetContrast.
Second, I do follow two different naming conventions for C++ and C.
In C++ I use the Arduino style guide, where the verb after the dot starts with a lower case. So it is u8g2.setContrast()

For the C variant
```
u8g2_SetContrast(u8g2_t *u8g2, Value);
```
you have to specify a pointer variable or an address to the u8g2 object.
You can get this also from the Arduino class

```
u8g2_t *u8g2_ptr;
...
u8g2_ptr = u8g2.getU8g2()
...
u8g2_SetContrast(u8g2_ptr, 123);
```

Tulsa2008, 2017-10-16:
Olikraus:
First, Thanks you for your quick response to me on this issue.
I am aware that C is case sensitive.  I missed the difference in "u8g2_SetContrast" and "u8g2.setContrast".  I used "u8g2.setContrast" and it compiled fine.  Problem solved.

Thanks again for pointing that out to me.

Ken

olikraus, 2017-10-16:
great :+1:

closed 2017-10-16 by olikraus