olikraus/u8g2 #391: Marlin 32 bit port to U*G2 - need some help

Bob-the-Kuhn, 2017-10-16:
I've been playing with Marlin 2.0.x in an effort to move over to U8G2.  I've got it partially working with the LPC1768 on a couple of SPI displays so I know that the custom callback & GPIO routines are working..

Currently I have two issues I need some help/advice with.

----

Marlin has some custom fonts that aren't in the U8G2 format.  Is there any newer advice than "use bdfconv" from issue #170?

----

My setup lines are currently variations of:
`u8g2_Setup_st7565_64128n_1(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);`

All is well when they're in UGg2lib.h but that's not a long term solution. 

I have not been able to get it compile when I move them out of the U8G2 library. 

I'm trying to keep the code base changes to a minimum so I'm trying to move them into the current DOGM file.  The top level layout is:
```
MAIN
 Setup() {
  ....
  #include "DOGM.h"

file DOGM.h:
  ...
  #include <U8g2lib.h>
  //extern u8g2_t u8g2;  
  //u8g2_t u8g2;
  //u8g2.begin();  // u8g2 now requires this
  uint8_t u8x8_LPC1768_4wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
  uint8_t u8x8_gpio_and_delay_LPC1768(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

  u8g2_Setup_st7565_64128n_1(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
  u8g2_Setup_st7565_64128n_2(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
  u8g2_Setup_st7565_64128n_f(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
  ...
  #if defined(VIKI2_deisplay)
    U8G2_ST7565_64128N_2_4W_SW_SPI u8g2(U8G2_R0, DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);
  #endif
  ...
 void init_display){
    u8g2.begin()
  }
```
With this setup I get the following compile error.  I'm using PlatformIO.
```
error: expected constructor, destructor, or type conversion before '(' token
u8g2_Setup_st7565_64128n_1(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
^
```

The commented lines are just a few of the variations I've tried.

I've looked through the Wiki, the examples/issues it points to and RIOT but have not found a detailed example of using U8G2_setup outside the library.

olikraus, 2017-10-16:
> Marlin has some custom fonts that aren't in the U8G2 format. Is there any newer advice than "use bdfconv" from issue #170?

How did you create the fonts for u8glib? I have just replaced bdf2u8g by bdfconv (which is the new name for the same tool on u8g2 side).  I mean, the original font data should be available as bdf file, isn't it? 

> I have not been able to get it compile when I move them out of the U8G2 library.

I need MUCH more information on this. 

Are you using a Arduino IDE? 
If yes, then why do you take the burden of writing your own gpio_and_delay procedure? Just use the existing u8g2 procedures, which build on SPI.h and Wire.h
If no, then you should not include "#include <U8g2lib.h>" but instead use "u8g2.h" only.

The problem is: U8g2lib.h is a C++ header AND assumes a lot of Arduino things. So if your environment is not compatible with Arduino, then U8g2lib.h will not work.

AnHardt, 2017-10-16:
> Marlin has some custom fonts that aren't in the U8G2 format. Is there any newer advice than "use bdfconv" from issue #170?

@Bob-the-Kuhn 
The recommended workflow for u8glib is:
Load the *.fon-files in `Fony.exe`. Export to *.bdf-files. Use `bdf2u8g` to convert to *.h-files .

When I made the fonts for Marlin Fony was the simplest pixel-font-tool i could find. It had just enough functions, but not that many - i could not understand. `Fontforge` is a nice, capable program - but total overkill if you have to produce pixel-fonts with only up to 255 symbols. Sadly Fonys .bdf import was unreliable (for me) then, while the export was no problem. So I decided for .fon to be the primary format. 
Providing both formats was no option to avoid discrepancies.

olikraus, 2017-10-16:
Fully agree, I also use Fony for any special u8g/u8g2 fonts. So your workflow will be the same, except, that you need to use bdfconv. 
U8g2 bdfconv is more powerful and has a new set of commandline options (mainly to handle unicode and output format). 

U8g2 FAQ includes a small description on this:
https://github.com/olikraus/u8g2/blob/ae207103010074aeae6d6a3b9691b5cbaa27f672/doc/faq.txt#L183

Let me know if you need further instructions on bdfconv.

Bob-the-Kuhn, 2017-10-16:
I'm using PlatformIO within Atom to compile for the Re-ARM board which has a LPC1768 mcu.  

Here's a dropbox link to my [software](https://www.dropbox.com/sh/iovm5oh6cl9j9mt/AAAZNF2niT0WoTp7gPncEko7a?dl=0) .  

Here's the [repository](https://github.com/Bob-the-Kuhn/RCBugfix-M43-and-spindle/tree/U8G2-port) .  The branch name is U8G2-port.

I tried including u8g2.h but that didn't help.

To see the compile error 
1. Use the dropbox link to download the software.
2. Within Platformio/Atom set the project folder to **RCBugfix-M43-and-spindle**
3. Select **PIO Build (Re-ARM)** to start the build process.  After about 60 seconds the "expected constructor, destructor, or type conversion" errors will pop up.

To enable passing the pins into U8G2 I've enabled  **U8X8_USE_PINS** and **ARDUINO_AVR_DIGISPARK** via build flags in the **platformio.ini** file.

The compiler is complaining about lines 51-53 s in the file **RCBugfix-M43-and-spindle\Marlin\src\lcd\ultralcd_impl_DOGM.h**

----

All I know about the Marlin fonts is what's in the current Marlin repository.  I have no idea where the BDF files are at.

I tried importing the Marlin fonts into Fony 1.4.6 but got nowhere. Most likely it's my lack of knowledge about font formats that's the problem.

It could be that I need 1.4.7.  I can't find that on the web.  Could one of you please post Fony 1.4.7 or point me to it?

The fonts I need to convert are all in the Marlin\src\lcd\dogm directory.

olikraus, 2017-10-16:
I think the Marlin original fonts in the fony format (.fon) are here:
https://github.com/MarlinFirmware/Marlin/tree/1.1.x/buildroot/share/fonts
Fony.exe itself is available here:
http://hukka.ncn.fi/?fony

for bdfconv you need to load the .fon file into fony and export the same as .bdf. Then feed the .bdf file into bdfconv (with the correct cmdline options) and generate the .c code.

===

I can not say much on the error message. It would be easier to have the specific message together with the code on github.

Bob-the-Kuhn, 2017-10-16:
FYI - I've done some performance testing of U8G vs. U8G2.  Here's the [raw data](https://www.dropbox.com/s/y1rgkeu1d8v8mj2/U8G2%20vs%20U8G%20rev%201.xlsx?dl=0).

In general U8G2 is slower but I don't think it's enough to be a major concern.  I think the bigger discussion will be about RAM usage on the mega2560.

I'll be posting this in Marlin's issues later today.

Bob-the-Kuhn, 2017-10-16:
Thanks for the pointers to the fonts.

I've been to that Fony page.  The 1.4.7 link is broken.

----

The GitHub link to the repository is in the previous post.

Here's the error messages I get.
```
In file included from Marlin\src\lcd\ultralcd.cpp:92:0:
Marlin\src\lcd\ultralcd_impl_DOGM.h:51:27: error: expected constructor, destructor, or type conversion before '(' token
u8g2_Setup_st7565_64128n_1(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
^
Marlin\src\lcd\ultralcd_impl_DOGM.h:52:27: error: expected constructor, destructor, or type conversion before '(' token
u8g2_Setup_st7565_64128n_2(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
^
Marlin\src\lcd\ultralcd_impl_DOGM.h:53:27: error: expected constructor, destructor, or type conversion before '(' token
u8g2_Setup_st7565_64128n_f(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
^
```

AnHardt, 2017-10-16:
> The 1.4.7 link is broken.
1.4.6 is ok for our purpose.

Interesting numbers. :-)

olikraus, 2017-10-16:
```
In file included from Marlin\src\lcd\ultralcd.cpp:92:0:
Marlin\src\lcd\ultralcd_impl_DOGM.h:51:27: error: expected constructor, destructor, or type conversion before '(' token
u8g2_Setup_st7565_64128n_1(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
^
Marlin\src\lcd\ultralcd_impl_DOGM.h:52:27: error: expected constructor, destructor, or type conversion before '(' token
u8g2_Setup_st7565_64128n_2(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
^
Marlin\src\lcd\ultralcd_impl_DOGM.h:53:27: error: expected constructor, destructor, or type conversion before '(' token
u8g2_Setup_st7565_64128n_f(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
^
```

ok, but what is the purpose of these statements. They look like function calls without being placed inside a function. Just remove lines 51 to 53.

Bob-the-Kuhn, 2017-10-16:
I thought that **u8g2_Setup_st7565_64128n_f(...)** is what linked **u8x8_LPC1768_4wire_sw_spi** and **u8x8_gpio_and_delay_LPC1768** to **U8G2_ST7565_64128N_2_4W_SW_SPI u8g2( U8G2_R0, 52, 51, 16, 59, 255);**.

If I just delete them then won't the Arduino versions be used instead of the LPC1768 ones?

olikraus, 2017-10-16:
> If I just delete them then won't the Arduino versions be used instead of the LPC1768 ones?

The mentioned constructors will always use the Arduino variants. I see three options:
1. Implement SPI.h/Wire.h for LPC1768
2. Rewrite the constructors
3. Replace the constructors

My knowledge on the Marlin code is too less to decide something here.

Bob-the-Kuhn, 2017-10-16:
Apparently I've been out in left field on this.

I just re-read the port to new MCU page and it only mentions "u8x8_Setup(...)".  I haven't the faintest idea how I got from u8x8_Setup to u8g2_Setup.

Marlin is expanding to the 32 bit processors.  The question came up about moving from u8g to u8g2  No one had a good idea of what that meant.  When I saw the "port to new MCU" section I thought that it was time to play with u8g2.  

I'm going to get a good night's sleep & re-read the u8g2 docs and the examples.

Bob-the-Kuhn, 2017-10-17:
SUCCESS!!!

I changed to the following sequence and it worked.  Just declared a new class.
```
uint8_t u8x8_LPC1768_4wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8x8_gpio_and_delay_LPC1768(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

class U8G2_ST7565_64128N_2_4W_SW_SPI_Bob : public U8G2 {
  public: U8G2_ST7565_64128N_2_4W_SW_SPI_Bob(const u8g2_cb_t *rotation, uint8_t clock, uint8_t data, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
    u8g2_Setup_st7565_64128n_2(&u8g2, rotation, u8x8_LPC1768_4wire_sw_spi, u8x8_gpio_and_delay_LPC1768);
    u8x8_SetPin_4Wire_SW_SPI(getU8x8(), clock, data, cs, dc, reset);
  }
};

```

U8G2_ST7565_64128N_2_4W_SW_SPI_Bob u8g2( U8G2_R0, 52, 51, 16, 59, 255);

----

To do list:
1. One bit map mostly works, others don't.
2. Inverting the display (R0 vs. R2) results in garbaged characters & changes to how bit map displays.
3. Find Polish 5x7 FON or BDF
4. Re-ARM U8G apparently transmits just the changed pages but mega2560 and U8G2 always transmit the entire display.

olikraus, 2017-10-17:
> Fony 1.4.6 has a check box for "don't trim glyph bitmaps" when exporting as a BDF. Should that be checked or not? The default is not checked.

bdfconv also trims the glyph bitmaps, so it probably does not matter whether this happens twice.

> Are the "-f 1 -m " options correct in the following command line?

The "-m" options selectes the glyphs which should be copied from the bdf file to the u8g2 font. Syntax is correct. You may additionaly select multiple ranged, separated with a comma. Whether it makes sense from a functional perspective, i can not say.

> I changed to the following sequence and it worked. Just declared a new class.

looks good :-)

> To do list:

Except point 3, other entries are not so clear to me. Please clarify if you want me to support here.

AnHardt, 2017-10-17:
Maybe the original author (@android444) of pl has the .fon or .bdf around.

AnHardt, 2017-10-17:
Font remade from scratch:
https://github.com/AnHardt/Marlin/pull/75

Bob-the-Kuhn, 2017-10-17:
@AnHardt - thanks for the Polish font.  I was sad to see that you had to re-create it.

@olikraus - thanks for your help.  I'll get back to this forum if I need more help.

Bob-the-Kuhn, 2017-10-17:
Just incase they're of interest to you .. here's the [Marlin U8G2 fonts](https://www.dropbox.com/s/fq7uz1xzq7ed31j/Marlin%20U8G2%20fonts.zip?dl=0).

The 6x9 might be of general interest.  It's the -Misc-Fixed-Medium-R-Normal--9-90-75-75-C-60-ISO10646-1 font.

olikraus, 2017-10-17:
Looks good. So i can close this issue for now?

Bob-the-Kuhn, 2017-10-18:
I need your help on the garbage character display.  Hopefully you can point me in the right direction.

This is on a VIKI2 display.

As best I can tell 
- U8G2 does all the rotation work.
- Marlin's only involvement in rotation is setting up the rotation within  `U8G2_ST7565_64128N_2_4W_SW_SPI u8g2( (&u8g2_cb_r0), ....) `properly.
- Marlin's drawing & text routines know nothing about rotation.  Marlin sends the same info when R0 is in effect as when R2 is in effect.

Do I need to create the buffer (`u8g2_t u8g2;`) or initialize it?

Maybe I've got the pointers to the buffer wrong and Marlin is sending in some garbage.
```
uint8_t* page  = u8g2.getBufferPtr();
uint8_t* first_page = page;

if (first_page == u8g2.getBufferPtr())

#define PAGE_UNDER(yb) (u8g2.getU8g2()->user_y0 <= (yb))  
#define PAGE_CONTAINS(ya, yb) (PAGE_UNDER(yb) && u8g2.getU8g2()->user_y1 >= (ya))  
```

Here's pictures of the screen when R0 is in effect and when R2 is in effect.  
- R0 splash - graphics OK, text is garbage
- R0 info - text is OK (ignore the upper 1/3 - I think that's a Marlin issue)
- R2 splash - graphics not right, text is OK
- R2 info - text is garbage (ignore the upper 1/3)

[image]

[image]

[image]

[image]

olikraus, 2017-10-18:
> U8G2 does all the rotation work.

yes, correct

> Marlin's only involvement in rotation is setting up the rotation

right

>  Marlin's drawing & text routines know nothing about rotation.

This is how it should be.

> Do I need to create the buffer (u8g2_t u8g2;) or initialize it?

With the C++ constructor, this is not required. "u8g2_t" is part of the U8g2 class.

> if (first_page == u8g2.getBufferPtr())

This will not work. "u8g2.getBufferPtr()" will always return a pointer to the buffer. This buffer will never change throughout the lifetime of your program. So, this condition will always be true according to your code. What do you want to achieve here?

Probably you what to use https://github.com/olikraus/u8g2/wiki/u8g2reference#getpagecurrtilerow here:
u8g2.getPageCurrTileRow() will return 0 if this is the first page to be rendered.

```
#define PAGE_UNDER(yb) (u8g2.getU8g2()->user_y0 <= (yb))  
#define PAGE_CONTAINS(ya, yb) (PAGE_UNDER(yb) && u8g2.getU8g2()->user_y1 >= (ya))  
```

There is a semi-official function to test whether there is an intersection with the current page:
 u8g2_IsIntersection(u8g2.getU8g2(), x, y, w, y)

Not sure whether this is the same...
```
#define PAGE_CONTAINS(ya, yb) u8g2_IsIntersection(u8g2.getU8g2(), 0, ya, 10, yb-ya+1) 
```
You could also go deeper and call the intersection test for y only:

Bob-the-Kuhn, 2017-10-19:
I finally figured out that the "garbage" text and bitmaps were actually inverted.

Playing with `u8g2.setBitmapMode `and `u8g2.setDrawColor `seems to have gotten 95% or 100% of each screen to display properly.  

The last issue I'm aware of looks like the top 2 or 3 pixels in a box have been erased.  I know they disappeared part way through playing with `u8g2.setDrawColor`.  Something to investigate tomorrow.

I'm pretty sure that either `u8g2.setBitmapMode` or `u8g2.setFont` is messing with `u8g2.setDrawColor`.  Seems like I need to issue  `u8g2.setDrawColor(white)` after each of them.  Too many changes to be sure.  Another thing for tomorrow.

olikraus, 2017-10-19:
Hmm.. I think I need to look at this problem more closely. Actually the bitmap code was partly contributed and i failed to do complete tests on that.
ToDo: Create bitmap example with various color options and rotations.

Bob-the-Kuhn, 2017-10-19:
I'm fully functional now.  Just needed one more `u8g2.setDrawColor` after `u8g2.setFont`.

Here's the things I needed to do in addition to what was in the transition guide:
•	u8g.  to u8g2.
•	U8g_ to u8g2_ except for u8g_fntpgm_uint8_t and U8G_FONT_SECTION
•	U8G_SECTION to U8G_FONT_SECTION
•	getU8g to getU8g2
•	setPrintPos  to  setCursor

Thanks to both of you for the invaluable help.

olikraus, 2017-10-19:
> I'm pretty sure that either u8g2.setBitmapMode or u8g2.setFont is messing with u8g2.setDrawColor. Seems like I need to issue u8g2.setDrawColor(white) after each of them. Too many changes to be sure. Another thing for tomorrow.

Thanks for this. At least I have to check the documentation for these functions.

olikraus, 2017-10-19:
Neither setBitmapMode nor setFont will modify the draw color, but drawBitmap leaves the drawcolor in an undefined state. Did you use drawBitmap()?

closed 2017-10-19 by olikraus

Bob-the-Kuhn, 2017-10-21:
Yes, there are three of them.

Would you believe that I'm having trouble with displays on a mega2560/RAMPs controller?  The text & boxes are all correct but the bitmaps look like random noise.  Off to read the manual & see what options I have to play with.

olikraus, 2017-10-21:
I fixes the Bug in drawBitmap.

Bob-the-Kuhn, 2017-10-21:
The bitmap problem was solved by reversing the bits within each byte.

olikraus, 2017-10-21:
ok, yet the difference between drawBitmap and drawXBM is the reversed bits within the bytes.

Bob-the-Kuhn, 2017-10-26:
I must have done something wrong when I did my original coding for U8G2.  When I went to manually rebase it, the code penalty is now about 1400 bytes for U8G2 vs. U8G when targeting the RepRap Discount Full Graphic controller.

Unfortunately we have a sizeable group that is still trying to fit Marlin into the 128K FLASH AVR chips so space is at a premium.  Looks like we'll be staying with the old U8G for now.

pillopaolo, 2025-05-31:
Hi Bob, are you willing to share the full code of your successful test mentioned above?
I am really struggling to use U8G2 with LPC1768, no idea how to code u8x8_LPC1768_4wire_sw_spi and 8x8_gpio_and_delay_lpc1768.
Thanks in advance

pillopaolo, 2025-06-10:
> I once ported u8g2 to a LPC11xx device (software emulated I2C). Maybe this
> helps:
> https://github.com/olikraus/lpc11u3x-gps-logger/blob/master/u8g2_logo/src/u8x8_lpc11u3x.c
>
> LPC11 should be somehow similar to LPC17 and i just did use GPIOs.

Thanks a lot.

However I am using SW_SPI for standard 3D printer ST7920 display, using
arduino-LPC1768 platform
For GPIOs I use Marlin cpp code.
Is this approach too bad?

Thanks
Paolo

olikraus, 2025-06-12:
SW SPI would not differ much from my approach with the LPC11xx, you would just need to define some more GPIOs and also some more of the timing messages, see the porting manual here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback