olikraus/u8g2 #392: SetContrast #2

Tulsa2008, 2017-10-16:
Olikraus:
Now I really feel like a fool.

During the last issue I had with the "SetContrast" function, I could not get it to compile.  Using "u8G2.setContrast(Value);" I now can get it to compile but I can't get it to work.  

I have found that I  can control the contrast of the display by changing the setting the contrast value in  u8x8_d_st7565.c.  By doing so, I have found with my display, if I set the value to 0x10, the screen is basically white.  If I set the value to 0x2B the screen is basically black.  So that tells me that the range of the contrast "Value" for my display must be between the two values delineated above.  

From the u8G2 reference manual, I get the impression that the "setContrast" function should work pretty much like most of the other functions.  That is to say, call the function, give the value a number and use the display as normal.  

Below is the simple program I wrote to test the function:
<
#include <Arduino.h>
#include <SPI.h>
#include <U8x8lib.h>
#include <U8g2lib.h>

  U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2 (U8G2_R0, 3, 2, 4); // Proto board display

void setup() {
  Serial.begin(74880);
  u8g2.begin();

   u8g2.clearBuffer(); u8g2.sendBuffer();              // Clear Screen
   u8g2.setFont(u8g2_font_pxplusibmvga9_tr);   //Large Bold Font
   u8g2.drawStr(8,28,"HelloWorld 1");                 // Load buffer
   u8g2.sendBuffer();                                            //Send buffer to display
   delay(5000);                                                      //wait for 5 seconds

   u8g2.clearBuffer(); u8g2.sendBuffer();             // Clear Screen
   u8g2.setContrast(0x2B);                                   //Set Contrast
   u8g2.setFont(u8g2_font_pxplusibmvga9_tr);   // Large Bold Font
   u8g2.drawStr(8,28,"HelloWorld 2");                 // Load buffer
   u8g2.sendBuffer();                                            //Send buffer to display
}

void loop() {
//Nothing here
}
> 
In the test program presented above, the program compiles and loads without any problems.
The first line "HelloWorld1"  shows up on the screen as expected.
There is a five second delay and the screen goes blank.  HelloWorld2 does not display.
Using the Value of "0x2B" the screen should be very dark, nearly completely black.  I have also used the value of "0x18" which is in the "u8x8_d_st7565.c." and the screen still shows up as blank. 

Again, I hate to be pain, but I am stuck.

Thanks in advance for your help.

Ken

olikraus, 2017-10-16:
I once made the decision, that the argument for setContrast should be in the range from 0..255 for ALL displays. Now the st7565 has a range of 6 bit (0..63) so had to scale down the range of setContrast:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7565.c#L171

So, if you have a correct value for your display as per datasheet, you need to multiply this value by 4:
u8g2.setContrast(0x02b * 4);

Tulsa2008, 2017-10-16:
That Worked!!!

Thanks a bunch

Ken

olikraus, 2017-10-16:
:-)

closed 2017-11-24 by olikraus