olikraus/u8g2 #394: will it work with waveshare 4.2inch_e-Paper_Module? with lua

shimondoodkin, 2017-10-18:
https://www.waveshare.com/wiki/4.2inch_e-Paper_Module
seems there is no matching display driver
how to make it work?
can somebody develop this, to help me?

shimondoodkin, 2017-10-18:
found this, esp32 brahttps://nodemcu.readthedocs.io/en/dev-esp32/en/modules/u8g2/
supports u8g2.

olikraus, 2017-10-18:
indeed this 4.2" module is not supported. Adding support is thinkable, but almost impossible without having the device in my lab.

shimondoodkin, 2017-10-18:
there is similar display  implemented already
maybe  it is possible to look in the code of the example code.zip and do the small modifications required.

shimondoodkin, 2017-10-18:
https://github.com/olikraus/u8g2/issues/318 similar implemenations

shimondoodkin, 2017-10-18:
i had compared the code of 2.9 and 4.2 , the files are different, but with similar structure

olikraus, 2017-10-18:
The structure will be the same, even for u8g2. The most biggest question is: What init code is required for the display. Or let me start with a more simple question: What is the name of the build-in graphics controller for that display. Will we get the datasheet for this controller?

MrJazzmax, 2017-10-23:
Hi Oliver,, datasheet for this display is here: https://www.waveshare.com/wiki/File:4.2inch-e-paper-specification.pdf
I got the same problem as shimon,, I woud like to use this display with your graphic library.
Im not sure if this help you .. (probably you have seen this datasheed already).
PS: I can sent you my own Waveshare 4.2 inch display (any problem)

olikraus, 2017-10-23:
The problem is, I simple don't have time at the moment to work on such a complicated device. Have a look at the issue regarding the implementation of the 2.9" display. It requires so much investigation, which i can not do at the moment.

shimondoodkin, 2017-10-23:
I'm sorry. My project canceled.

olikraus, 2017-10-23:
Reference to 2.9" epaper discussion: #318

MrJazzmax, 2017-10-23:
I understand,, but maybe one day, when you get a time and taste to get back to this display ..
Controller: IL0398 : http://www.good-display.com/download_detail/downloadsId=537.html

thanks for your fast respond !

olikraus, 2017-10-24:
I reopen this... let's see... 
From where do you know that this is the correct controller?
At least it seems to fit.

MrJazzmax, 2017-10-24:
because I found another library for this display,, exactly this device (waveshare 4.2) and there was description .. Also I compared my display  with good display, it seems to be the same manufacturer at first glance. very similar like in case of waveshare 2.9 (IL3820).

MrJazzmax, 2017-11-01:
I forgot to write the name of device,, Its GxGDEW042T2

bobybobybob, 2018-02-11:
Waveshare has Arduino Libary too. Maybe it is OK for 4.2" Paperdisplay. https://github.com/soonuse/epd-library-arduino

ZinggJM, 2018-02-14:
It works with GxEPD: https://github.com/ZinggJM/GxEPD
And there is new GxEPD2_AVR and GxEPD2_32 "under construction" with interface similar to U8G2.

olikraus, 2018-02-14:
This issue started long back. Until now I did not found any time to implement support for another e-paper display. Also the question is, is it really worth to do so, as there are other great libs, supported by @ZinggJM  .

At the moment I do not think, that I will be able to provide support for the 4.2" e-paper requested here.

baoyingcheng, 2018-02-23:
the driver chip is il0398 and I can offer you the datasheet and demo driver app in stm32
[4.2inch_epaper_stm32.zip](https://github.com/olikraus/u8g2/files/1750136/4.2inch_epaper_stm32.zip)

shimondoodkin, 2018-02-23:
What was wanted is to use semi-large epaper display with node mcu with elua. Which is not possible with any lib availible. u8g2 is integrated into node mcu.

ZinggJM, 2018-02-23:
I don't know elua. But if it can be extended with C++ classes, it should not be too difficult to adapt and add the class GxEPD2_32_BW. You would need to change the few places using the Arduino SPIClass, and setMode() and digitalWrite() IO calls.

olikraus, 2020-01-19:
I decided to concentrate on oleds and lcds. There are other good libs for e-paper displays (see above)

closed 2020-01-19 by olikraus