olikraus/u8g2 #395: u8g 'menu'example port to u8g2 need help

snapedumbledore, 2017-10-18:
u8g2 haven't 'U8GLIB_ST7920_128X64_4X(sck, mosi, cs [, reset])'  ;     I use 
‘U8G2_ST7920_128X64_1_SW_SPI(rotation, clock, data, cs [, reset]) | page buffer, size = 128 bytes ；
U8G2_ST7920_128X64_2_SW_SPI(rotation, clock, data, cs [, reset]) | page buffer, size = 256 bytes ；
U8G2_ST7920_128X64_F_SW_SPI(rotation, clock, data, cs [, reset]) ‘
instead 。but no one display correctly. What should i do ,thanks.

olikraus, 2017-10-18:
> What should i do 

send code, send pictures, send more information

snapedumbledore, 2017-10-18:
when i use u8glib,display correct:
[image]
when use u8g2lib ,like that:
[image]
///////////////////////////////////u8g/////////////////////////////////
#include "U8glib.h"

U8GLIB_ST7920_128X64_4X u8g(8, 9, 10);	// SPI Com: SCK = en = 18, MOSI = rw = 16, CS = di = 17

#define KEY_NONE 0
#define KEY_PREV 1
#define KEY_NEXT 2
#define KEY_SELECT 3
#define KEY_BACK 4

// DOGS102 shield configuration values
//uint8_t uiKeyPrev = 2;
//uint8_t uiKeyNext = 4;
//uint8_t uiKeySelect = 5;
//uint8_t uiKeyBack = 3;

// DOGM128-Shield	 configuration values
// DOGXL60-Shield configuration values
uint8_t uiKeyPrev = 7;
uint8_t uiKeyNext = 3;
uint8_t uiKeySelect = 2;
uint8_t uiKeyBack = 8;

uint8_t uiKeyCodeFirst = KEY_NONE;
uint8_t uiKeyCodeSecond = KEY_NONE;
uint8_t uiKeyCode = KEY_NONE;

void uiSetup(void) {
  // configure input keys 
  
  pinMode(uiKeyPrev, INPUT_PULLUP);           // set pin to input with pullup
  pinMode(uiKeyNext, INPUT_PULLUP);           // set pin to input with pullup
  pinMode(uiKeySelect, INPUT_PULLUP);           // set pin to input with pullup
  pinMode(uiKeyBack, INPUT_PULLUP);           // set pin to input with pullup
}

void uiStep(void) {
  uiKeyCodeSecond = uiKeyCodeFirst;
  if ( digitalRead(uiKeyPrev) == LOW )
    uiKeyCodeFirst = KEY_PREV;
  else if ( digitalRead(uiKeyNext) == LOW )
    uiKeyCodeFirst = KEY_NEXT;
  else if ( digitalRead(uiKeySelect) == LOW )
    uiKeyCodeFirst = KEY_SELECT;
  else if ( digitalRead(uiKeyBack) == LOW )
    uiKeyCodeFirst = KEY_BACK;
  else 
    uiKeyCodeFirst = KEY_NONE;
  
  if ( uiKeyCodeSecond == uiKeyCodeFirst )
    uiKeyCode = uiKeyCodeFirst;
  else
    uiKeyCode = KEY_NONE;
}

#define MENU_ITEMS 4
const char *menu_strings[MENU_ITEMS] = { "First Line", "Second Item", "3333333", "abcdefg" };

uint8_t menu_current = 0;
uint8_t menu_redraw_required = 0;
uint8_t last_key_code = KEY_NONE;

void drawMenu(void) {
  uint8_t i, h;
  u8g_uint_t w, d;
[... 170 lines cut ...]
}

void setup() {
  // rotate screen, if required
  //u8g2.setRot180();
  u8g2.begin();
  uiSetup();                                // setup key detection and debounce algorithm
  menu_redraw_required = 1;     // force initial redraw
}

void loop() {  
  //u8g2.clearBuffer();
  uiStep();                                     // check for key press
    
  if (  menu_redraw_required != 0 ) {
    u8g2.firstPage();
    do  {
      drawMenu();
    } while( u8g2.nextPage() );
    menu_redraw_required = 0;
  }

  updateMenu();                            // update menu bar
  
}

olikraus, 2017-10-18:
hmm... ok, looks like a wiring problem. I pushed the ST7920 very much to its limits with u8g2.
Try this:
locate u8x8_cad.c
Remove (or comment) lines 357 (https:https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c#L357//github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c#L357)
to
371

snapedumbledore, 2017-10-19:
I comment the lines  u8x8_cad.c(357-371),but the problem still exist. 
only the 'u8glib' works well.@olikraus

snapedumbledore, 2017-10-19:
[image]
still like this.

olikraus, 2017-10-19:
Hmmm, You could also try to increase the delay which surrounds the code there.

snapedumbledore, 2017-10-20:
I think there ara something wrong with my LCD .I use "u8glib",when I test "menu"example,it's OK.
but when I try others like"u8glogo"example ,failed.
[image]

olikraus, 2017-10-20:
It will probably not help if I say, that the ST7920 is the most unreliable display controller which is out there. Well, my suggestion is to make wires much shorter. Also consider soldering your connections to the display.

snapedumbledore, 2017-10-20:
emmmm.I will try it ,thanks

closed 2017-11-24 by olikraus