olikraus/u8g2 #396: Custom font question

Nakonzaid, 2017-10-18:
Hi there!
I'm trying to use a custom font but I can´t convert the .ttf file to the format u8g2 can use

https://fontstruct.com/fontstructions/show/1014500/falconded

This font was released  by uriba107 in his DEDUINO project, where he uses the old u8glib library:

https://github.com/uriba107/deduino

I read de FAQ and unsuccessfully tried some ways to convert the .ttf file to .bmf and then to .c  (otf2bdf first and bdfconv after)

Please, can you help me?

On the other hand, i can see at uriba107 project that he created a .h file that maybe is some kind usable for this.
That .h is like this

```
/*
  Fontname: -FontForge-FalconDED wide-Book-R-Normal--16-120-100-100-P-97-ISO10646-1
  Copyright: Copyright alancoolhand 2014
  Capital A Height: 10, '1' Height: 10
  Calculated Max Values w=11 h=12 x= 4 y= 7 dx=10 dy= 0 ascent=11 len=24
  Font Bounding box     w=11 h=12 x= 0 y=-1
  Calculated Min Values           x= 0 y=-1 dx= 0 dy= 0
  Pure Font   ascent =10 descent=-1
  X Font      ascent =10 descent=-1
  Max Font    ascent =11 descent=-1
*/

const u8g_fntpgm_uint8_t FalconDED_wide[3140] U8G_FONT_SECTION("FalconDED_wide") = {
  0,11,12,0,255,10,1,229,3,198,32,255,255,11,255,10,
  255,0,0,0,10,0,1,2,10,10,10,4,0,192,192,192,
  192,192,192,192,0,192,192,6,3,3,10,2,7,204,204,204,
  8,10,10,10,1,0,102,102,255,255,102,102,255,255,102,102,
  6,10,10,10,2,0,48,120,180,176,120,52,52,180,120,48,
  8,10,10,10,1,0,67,166,70,12,24,24,48,98,101,194,..........................
```

Here:

https://github.com/uriba107/deduino/blob/master/arduino_code/DEDunio_arduino/falconded_wide_u8g.h

As you can see, i'm pretty lost.

Regards!

olikraus, 2017-10-18:
The FontStruction “FalconDED” (https://fontstruct.com/fontstructions/show/1014500) by “uri_ba” is licensed under a Creative Commons Attribution Non-commercial Share Alike license (http://creativecommons.org/licenses/by-nc-sa/3.0/).

otf2bdf -p 16 -n FalconDED.ttf -o FalconDED.bdf
(attached FalconDED.bdf as FalconDED.txt)

[FalconDED.txt](https://github.com/olikraus/u8g2/files/1395508/FalconDED.txt)

bdfconv -f 1 -b 0 -m "32-255" -n FalconDED -o falconded.c FalconDED.bdf

[falconded.c.txt](https://github.com/olikraus/u8g2/files/1395541/falconded.c.txt)

Nakonzaid, 2017-10-19:
That helps a lot!

Thank you very very very very very much!!!!

That's exactly what I need. 
I guess I was not correctly entering any parameters

I've done some tests on my 12864b screen and it works really nice!
Well, now I must try to make the fonts smaller since the text I want to display does not fit on the screen (i must show 25 characters per line but now i only can show 16).
Maybe 6x6 pixels fonts or something like that (screen is 128 x 64, so I have calculated 128/25 = 6.72 ).
I guess I have to figure out how to convert that fonts to 6x6
Or is there any way to show fonts on screen smaller than they really are? 

I have seen that in the u8glib library there was a sentence to display a text twice the size of its font (setScale2x2), so I thought that, hopefully, there might be a sentence or any easy way to reduce the scale of a font in u8g2.
Maybe it just does not exist and I have to reduce the size of the fonts from the beginning (creating a new smaller font)...

olikraus, 2017-10-19:
> I guess I have to figure out how to convert that fonts to 6x6

Difficult with the font you provided because the font from fontstruct.com is a bitmap font.

Maybe you can choose a font from u8g2. U8g2 comes with many small fonts for this purpose
https://github.com/olikraus/u8g2/wiki/fntlistall

I have also removed scale 2x2 because there are also many big fonts available.

Nakonzaid, 2017-10-19:
Thank you very much again!

I need that particular font because the purpose of the project. 
I'm trying to show on the screen some specifical messages that need some special characters

What I try to do is to show on a physical screen the contents of an instrument in an F-16 simulator (Falcon BMS 4.33)

In the cockpit (up to the right):
https://i.ytimg.com/vi/sKodsmXL0eE/maxresdefault.jpg

And the result of the project should be:
https://i.ytimg.com/vi/bg-qZTIPsCQ/maxresdefault.jpg
https://i.ytimg.com/vi/ASe59UmDSTg/maxresdefault.jpg

So apparently I will have to create a source from the beginning.

olikraus, 2017-10-19:
I just wanted to say, that any downscale the original falcon font from above will not work. You can simulate this with the tool "ftview" from the truetype demo collection. You can use + and - keys to skale the ttf font up and down. It looks horrible when scaled down.

Nakonzaid, 2017-10-19:
Then I think the better way to go is create a totally new font with the appropiate size.
I mean:
bitmap font (with appropiate size of each character) --> transform .ttf to .bdf (with otf2bdf  like you did) --> transform .bdf to .c (with bdfconv)

olikraus, 2017-10-19:
You could also use fony.exe (http://hukka.ncn.fi/?fony) which can export to bdf directly (which will avoid the otf2bdf step). Note however, that I heard that the latest version of fony.exe is corrupt. Instead version 1.4.6 seems to be fine.

Nakonzaid, 2017-10-19:
Thanks!
Good idea.
I will try.

Nakonzaid, 2017-10-24:
Hi again!

I'm trying to create my own fonts, but i'm having trouble.
When I try

otf2bdf -p 16 -n FalconDED.ttf -o FalconDED.bdf

I get an error; freetype6.dll missing

And i can't instalol freetype because I get antother error (my pc us x64 but freetype is x32 and it can't run)
What can I do?

Nakonzaid, 2017-10-24:
I've registered manually those .dll's but now i get

otf2bdf: unable to open temporary file 'tmp/otf2bdf4948'

Nakonzaid, 2017-10-24:
Wait a minute!
I got it!
It works!

closed 2017-11-24 by olikraus