olikraus/u8g2 #397: support for SSD1327 128x128

raghunathgr8, 2017-10-19:
Hi Oliver,
Thanks for your brilliant U8g2lib. It works like a charm for me for my SSD1306 128x64 OLED.
I am upgrading my product to SSD1327 based 128x128 OLED and noticed that currently supported device list shows only SSD1327 96x96 for seeed studio OLED. 
Request you to consider supporting the 128x128 resolution as well.

warm regards
Raghunath

olikraus, 2017-10-19:
> cloned static const uint8_t u8x8_d_ssd1327_128x128_init_seq[] with following changes
  U8X8_CA(0x0a8, 0x07f),        /* multiplex ratio: 0x05f * 1/128duty */
  U8X8_CA(0x0a2, 0x000),        /* display offset, shift mapping ram counter */

olikraus, 2017-10-19:
I have created a beta release 2.20.0:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Download and install via the add zip library menu in the Arduino IDE.

Take the helloworld example, constructor
`U8G2_SSD1327_128X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

What will you see? Will it work?

raghunathgr8, 2017-10-20:
Thanks for the quick turnaround!
I have downloaded the zip. My hardware is setup for 3.3V SAMD21 with HW SPI
I hope I can test with that.
My display controller interface (VMCI) operates around 3.3V and the UNO
GPIO lines are driven at 5V.

I will let you know by end of the day.

Once again, thanks very much for your energetic  support.

Best regards
Raghu

olikraus, 2017-10-20:
The HW SPI constructor ist there, but mit mentioned in the examples

raghunathgr8, 2017-10-20:
Good Morning!
I have good news.
I used your beta release and tested the helloworld sketch. And it worked
fine.
I could test only with 4-wire HW SPI and not SW SPI due to my HW setup
constraint as explained in my previous email.

I will shortly send a screen shot of the display for your convenience.
I will be sending it to your email directly since github mail size will be
exceeded by the JPEG attachement.

Best Regards
Raghu

Best regards
Raghu

olikraus, 2017-10-20:
Great. Can you also give me feedback whether the u8x8/FlipMode example works correctly?
In both cases the text should start in the upper left corner.

raghunathgr8, 2017-10-20:
Flipmode seems to have some issue. The texts in normal and reverse video do not align to upper left corner. It appears that the display size is taken to be 128x96. I have sent you the screen shot by email for your kind reference.

olikraus, 2017-10-20:
I think I have fixed the flip mode. I also renamed the constructor, it is now:
```
U8G2_SSD1327_MIDAS_128X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

olikraus, 2017-12-09:
no further feedback... closing this...

closed 2017-12-09 by olikraus

raghunathgr8, 2017-12-11:
Thanks for the releasing the  library version 2.20.1 with support for SSD1327 128x128.

ghmac, 2018-08-20:
Hi olikraus,
I just purchased the I2C version of this SSD1327 128x128 1.5" 16-bit grey level oled by MakerHawk-US.
Would you please consider updating your constructor list for these?
Thanks

olikraus, 2018-08-20:
Not all constructors are mentioned in the examples. The I2C constructor is already there, see the Wiki.

Andrzej1000, 2018-09-07:
Hi.
I have SSD1327 form waveshare.
The library working on SPI.
Unfortunately I2C does not work.
The display does not light anything.
I checked example code from waveshare and it works.
Olikraus could you check what's going on?
Thanks...

bookerhsu4026, 2019-11-10:
I use u8g2 helloword demo code to build SSD1327  oled
[中景园高清1.5寸oled 128x128液晶屏ssd1327带灰度全程技术支持](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.585f2e8dQtzd0p&id=568527842788&_u=a1htbi28a946).
It's not work.
SPI OR I2C .
I try any device in u8g2 , that include ssd1327.
not work.
I run demo code form provider, that's work.

NikilRagav, 2020-07-13:
hi @olikraus 

I'm having some issue with the Rotate parameter causing an offset. Instead of 0,0 starting at the top left of the rotated display, it is starting at something to the effect of x=-10 or something like that

NikilRagav, 2020-07-13:
Actually it looks like the last page of the OLED is not printing anything

Only 128 x 96 pixels are controllable in Full Buffer mode

`U8G2_SSD1327_EA_W128128_F_HW_I2C screen(U8G2_R0, /* reset=*/U8X8_PIN_NONE);`