olikraus/u8g2 #398: support for ssd1326

MPereira-lp, 2017-10-19:
Hi!

I've tried wiring the screen like this:

[image]

I wasn't able to use the _C3_ on pin 22. Is that bad?

I've tried to change the width and height on several other screens, and I've "activated" the 16bit on the "u8g2.h" file. The screen still dead...

Any tips?

Thanks

JamesGKent, 2017-10-19:
what arduino are you driving this with? i note that the datasheets for this controller state a nominal voltage of 3.3v, if you've used a 5v arduino without level shifting it's likely fried...

MPereira-lp, 2017-10-19:
I'm using an arduino uno, but I'm using the 3.3v.

VDDIO, VDD, and VCC are supposed to go into the 3.3v right?

JamesGKent, 2017-10-19:
according to [this page](http://www.solomon-systech.com/en/product/advanced-display/oled-display-driver-ic/ssd1326/) it should be:
- VDD=2.4V - 3.5V                     
- VDDIO = 1.7V - VDD                     
- VCC = 9.0V - 15.0V

where VCC is used for actually driving the display, so putting 3.3v in here would explain why nothing appears on screen. you need to check what voltage your display module requires and provide that.

olikraus, 2017-10-19:
Additionally, all five datalines must not exceed a voltage of 3.3V Although you might have used Arduino Uno 3.3V for driving the power lines, the GPIOs of the Arduino still have 5V output for logic high.
For the data lines you should use a level shifter like the 74HC4050. You display will not work without levelshifter (let us hope that it wasn't destroyed by the GPIO voltage).

olikraus, 2017-10-19:
For the generation of the 9-15V I had used a LM2704 step up converter...

olikraus, 2017-10-19:
> I wasn't able to use the C3 on pin 22. Is that bad?

I once had a display, which did not work because of a missing cap.

olikraus, 2017-10-19:
After reading your PM: What is your exact display type? Any link to Datasheet? Your schematic is a starting point, but adding a 74HC4050 and a proper LM2704 is mandatory from my perspective.

MPereira-lp, 2017-10-20:
Hi!

Thanks for the answers.

[This is the datasheet](http://www.buydisplay.com/download/manual/ER-OLED018-1_Series_Datasheet.pdf)

[and this is where I bought it from](http://www.buydisplay.com/default/blue-1-8-inch-256x32-oled-display-module-serial-i2c-and-ssd1326)

They sell boards ready to use. But I'd rather use an arduino or a raspberry, for matters of space. I don't even know if I can communicate with this board they sell and read input variables on the display.

The 74HC4050 is used to regulate the low voltage right?

olikraus, 2017-10-20:
Is it ok, to continue on the github page regarding technical topics?
I will reply there.

Oliver

MPereira-lp, 2017-10-20:
Perfect, so it should be pretty simple to wire... (?)
I can't find the LM2704 on the eletronics stores around here... any alternative?

about the "VDDIO = 1.7V - VDD" , it means it should have a voltage above 1.7V and below VDD voltage (3.3V)... should I use a resistor and connect to the same line?

Thanks

olikraus, 2017-10-20:
There are many alternatives to the LM2704. I just found the LM2704 very reliable. It is available from e.g. mouser.com.

> about the "VDDIO = 1.7V - VDD" , it means it should have a voltage above 1.7V and below VDD voltage (3.3V)... 

yes

> should I use a resistor and connect to the same line?

i think this is not required

MPereira-lp, 2017-10-20:
My question is, if I connect VDDIO and VDD to the same line (the same V), should it work?

olikraus, 2017-10-20:
Yes, I think so.

olikraus, 2017-10-23:
> How should I wire the 74HC4050? I connect the VCC to the arduino 5V, GND to GND. 

No! You must connect VCC of the 74HC4050 to 3.3V of the Arduino Board.

> Then I connect the arduino pins on the 1A and the screen on 1Y (e.g.)?

Yes, direction is from A to Y, from Arduino to Display.

olikraus, 2017-10-23:
> So, I've been testing the 74HC4050, and If I connect the PSU with 5V to the A, the output on Y is 3V. Which is what is supposed to happen.

yes

> But If I connect the PSU (5V) to Y, the output on A is 10V (which would solve the missing LM2704). Would that damage anything??

A is an input and not an output. The only function of the 4050 is to get 5V and output 3.3V. It is not at all able to generate any output voltage. If you have measured any other voltages (10v), then your measurement is faulty (mostly because A is "floating". Try this: Add a 10K resistor between A and GND. Repeat your measurement: You will observer 0V at A. You will need a proper step up converter to get a suitable and reliable voltage. If you would to the same experiment with the 10K resistor with a step up converter, you will still see the 9V to 12V.

> So, which one would be the closest to the SSD1326?

Closest what? I have not even started to analyse the SSD1326. Probably none of them. Usually a new code has to be developed.

> I'm using SPI right now. Would it be better to use I2C?

SPI is fine. Level shifting for I2C is much more complicated, because I2C requires bidirectional communication. So,... let's continue with SPI.

MPereira-lp, 2017-10-24:
Well, right now I've connect everything as it is supposed to, and there is no light on the screen.

It makes no difference on using the GND on the arduino or the GND on the PSU right?

MPereira-lp, 2017-10-24:
[Is this helpful?](http://www.buydisplay.com/download/democode/ER-OLED018-1_DemoCode.txt)

olikraus, 2017-10-24:
> It makes no difference on using the GND on the arduino or the GND on the PSU right?

should be the same

> Is this helpful?

Everything is helpful. I will have a look on how to continue with the code.
Can you post a picture of your setup?

MPereira-lp, 2017-10-24:
https://i.imgur.com/mKsySYt.jpg
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Should I make a sketch of the wiring?

olikraus, 2017-10-24:
Nice pics. Did you see whether lines are toggled while adding any other u8g2 SSD132x constructors?
I mean, did you do any hardware check?

Meanwhile I did a crosscheck of the SSD132x controller:

```
		SSD1325		SSD1326		SSD1327		SSD1329
		128 x 80 x 4	256 x 32 x 4	128 x 128 x 4	128 x 128 x 4
	
0x15	set col		set col		set col		set col	
0x23	gr acc		-			-			-
0x24	draw rect		-			-			-
0x25	copy			-			-			-
0x26	scroll		scroll		scroll		-
0x27	-			-			scroll		-
0x2e	end move		end move		end move		-
0x2f		start move	start move	start move	-
0x75	set row		set row		set row		set row
0x81	contrast		contrast		contrast		contrast
0x82	-			-			-			2nd charge
0x84	current		current		nop			-
0x85	current		current		nop			-
0x86	current		current		nop			-
0x87	current		current		-

0x90	-			-			-			Icon ctl
0x91	-			-			-			Icon current
0x92	-			-			-			Icon current
0x93	-			-			-			Icon on/off
0x94	-			-			-			Icon all
0x95	-			-			-			Icon blink
0x96	-			-			-			Icon duty

0xa0	remap		remap		remap		remap
0xa1	start line		start line		start line		start line
0xa2	offset		offset		offset		offset
0xa4	mode		mode		mode		mode
..
0xa7	mode		mode		mode		mode
0xa8	multiplex ratio	multiplex ratio	multiplex ratio	multiplex ratio
0xab	-			-			regulator		-
0xad	master config	-			-			-
0xae		disp. on		disp. on		disp. on		disp. on
0xaf		disp. off		disp. off		disp. on		disp. on
0xb0	comp enable	-			-			-	
0xb1	phase len		phase len		phase len		phase len
0xb2	row period	frame freq	nop			frame freq
0xb3	div ratio [3]	clock div [2]	clock div 		clock div [2]	
0xb4	comp level	-			-			-
0xb5	-			-			GPIO			-
0xb6	-			-			2nd phase	-
0xb7	-			default gray	-			default gray
0xb8	gray level		gray level		gray level		gray öeveö
0xb9	-			-			linear LUT		-
0xbb	-			2nd charge	nop			2nd charge
0xbc	precharge	precharge	precharge	precharge
0xbe	comh volt		comh volt		comh volt		comh volt
0xbf		VSL			-			-			-
0xd5	-			-			fn select B	-
0xe3	nop			-			-			nop
0xfd		-			cmd lock		cmd lock		cmd lock

```

I assume all of them should turn on something. Most similar to the SSD1326 seems to be the SSD1329.
I think I will create a new constructor based on that controller for your SSD1326 display.

MPereira-lp, 2017-10-26:
I've been testing the voltage on the arduino lines with the SSD1329, and I get something around 0,6V on each line.

Since I dont have any other arduino one at the moment I've tested also on a micro, and the voltage on line 8 and 10 is 4.6V and the other ones are around 0.2V.

Is this normal?

olikraus, 2017-10-26:
Depends. A scope will tell you more precisly what happens on the line. You should see some activity according to the clock and data lines. CS and D/C line will change less often.

MPereira-lp, 2017-10-27:
Sorry for asking. But what do you mean by "A scope"? I've been using the multimeter to check the lines, is that what you mean?

Thanks

MPereira-lp, 2017-10-27:
[image]

Might this be the reason the screen is not lighting up?
Do you have any idea of what is the use of this?

Sorry for making so many questions

olikraus, 2017-10-27:
no problem. Your questions are very good. 
For your power supply: I personally would remove the initial power enable circuit. Unless you want to create a real low power app, it is not required. The display itself can be put into power save command by using a software command. Moreover, the R1200 has its own enable input, why not using this input, so somehow the two transistors seem to be fully redundant.

From reading the datasheet, the R1200 seems to be a nice setup converter. So from my understanding you do not need R4, R5, U2 and U3. Just connect Vin directly to the Vin input of the R1200.

Sorry for "scope". I was thinking of an oscilloscope.

olikraus, 2017-10-27:
I have to updated my discussion on the R1200. As mentioned if the datasheet it is mainly for LED backlights. So it is not really applicable for your OLED. Especially the rectifier is missing (which is indeed not required for the backlight application).

A better choice would be this device.

[image]

olikraus, 2017-10-28:
ok, I just saw, that you took the step up converter from the datasheet. Well, then it should be fine. Still R4, R5, U2 and U3 are not required.

olikraus, 2017-10-28:
For further testing: I have created 2.20.2 beta of u8g2. 
Download location: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Please use u8x8 interface for testing. I have included experimental constructor for ssd1326, but maybe ssd1329 may also work (a little bit).
U8g2 interface requires 16 bit mode, which is disabled by default (so better start testing with u8x8).
Of course I do not know whether the new constructor works...

MPereira-lp, 2017-10-30:
> ok, I just saw, that you took the step up converter from the datasheet. Well, then it should be fine. Still R4, R5, U2 and U3 are not required.

Yes, I took it from the screen datasheet, and I was not aware it is a powersupply. From what I understand it is a 5V to 12V power supply right? Might that be the reason my screen is not lighting up? or it should it work fine with a regular 12V DC transformer (psu) as I'm using?

MPereira-lp, 2017-10-30:
At the moment I'm out of UNO's. Arduino nano should do just fine right?

olikraus, 2017-10-30:
Any Power supply ist fine, nano is also ok

MPereira-lp, 2017-10-30:
I've been just checking with line are connecting to ground. And line 11(SDIN) and 12(SCLK) are connected to GND, it is not supposed right? If I disconnect 11 and 12 from the 74hc4050 it stops the connection to GND...

MPereira-lp, 2017-10-30:
[image]

**C1,C2 - 4.7uF/25V
C3,C4 - 01uF

R1- 680Kohms**

This is the scheme of my wiring right now. Everything as it is supposed to be?

MPereira-lp, 2017-10-30:
Good news, it already light up!
[This is what is showing on the display.](https://youtu.be/NDNPGxaJEEQ) What do you think?

dnsseguin, 2017-10-30:
Hi,

Check you reset connection,

regards

MPereira-lp, 2017-10-30:
Hi!

I've checked with multimeter and seems to be wired right...

olikraus, 2017-10-30:
> I've been just checking which lines are connecting to ground. And line 11(SDIN) and 12(SCLK) are connected to GND, it is not supposed right? If I disconnect 11 and 12 from the 74hc4050 it stops the connection to GND...

You probably mean pin 13. 
Maybe you have soldering issue on your board with the GND connection. And no, there should be no GND connection.

olikraus, 2017-10-30:
> Good news, it already light up!
This is what is showing on the display. What do you think?
After a reset it looks like this.

Can you post a signle picture with a better resolution?
Of course the new SSD1326 device is not tested at all. Can you also check with the other devices? The SSD1329 seems to be very similar.

MPereira-lp, 2017-10-31:
I've tested with SSD1329 SSD1327 SSD1325 and SSD1322. All of them give a very close screen output. The SSD1322 is the only one that does not work at all.

I'll re-do all the wiring and soldering and try again.

olikraus, 2017-11-02:
ok, which constructor did you use?

MPereira-lp, 2017-11-03:
Screen Constructor?
[This is the exact screen.](http://www.buydisplay.com/default/blue-1-8-inch-256x32-oled-display-module-serial-i2c-and-ssd1326)

All the datasheets I provided you are from this exact screen.

MPereira-lp, 2017-11-03:
Today as I turned on the arduino the screen looked like [this.](https://youtu.be/ECWHWbDFkOY)

I don't think I've changed anything. Looks like hardware or software issues?

olikraus, 2017-11-03:
What about the newly added 
U8G2_SSD1326_ER_256X32_1_4W_SW_SPI
constructor?
What will be seen with this device. Please also enable 16 bit for this.

MPereira-lp, 2017-11-03:
The previous pics and videos are also with the 16 bit enabled and the U8G2_SSD1326_ER_256X32_1_4W_SW_SPI.
After I've uncommented the line (  _U8X8_CA(0x0ad, 0x002),		/* master configuration: disable embedded DC-DC, enable internal VCOMH */_  ) it is looking pretty close to the ssd1329...

olikraus, 2017-11-04:
ups, i made a mistake in the ssd1326 code. Please download and test version 2.20.4:
https://github.com/olikraus/U8g2_Arduino

MPereira-lp, 2017-11-04:
It's looking much better now! Not done yet, but it's getting much closer 😃

[Here is the video.](https://youtu.be/cvzO1XMbqSA) (I've replaced the word "_Big_" on the _GraphicTest_)

One thing I've noticed, is that the GraphicTest does not loop anymore. Is that normal?

olikraus, 2017-11-04:
Graphics Test should work ad usual. I have not changed this.

olikraus, 2017-11-04:
Did you enable 16 Bit Mode again?

olikraus, 2017-11-04:
Yes, i think we are getting closer. I have created version 2.20.5:
https://github.com/olikraus/U8g2_Arduino

Note that 16 bit mode is again disabled. But you can keep it disabled as long as you use u8x8 examples.

MPereira-lp, 2017-11-04:
I've enabled 16bit before trying. But it is not needed if I use the **u8x8** instead of **FullBuffer**?

MPereira-lp, 2017-11-04:
The version available to download is 2.20.4

MPereira-lp, 2017-11-04:
Much better. Still, the small text is not readable. Any thoughts?

https://youtu.be/6nslNUYw0WY

olikraus, 2017-11-04:
ok, pls try again, version is 2.20.6

MPereira-lp, 2017-11-04:
It is looking PERFECT!

[Check it here!](https://youtu.be/GpRMoH02OpU) What do you think?

olikraus, 2017-11-04:
Looks good. Although I think that there are some artefacts. Can you test also the u8x8/FlipMode example? 
Another test would be the u8g2 graphics test (with activated 16 bit mode).

One more question: I saw, that the SSD1326 has a black/white mode. Currently the gray-scale mode is activated and the graphics content is upscaled inside u8x8. The drawback is, that currently 4 times more data needs to be transfered. The question is, shell we also work on a more faster version for this device? This faster version will not support u8x8, only u8g2 will be possible. The current version support both u8g2 and u8g8. I will keep the current (working) version, but shell we implement also a high speed u8g2 version? I mean finally the question is, are you willing to do some more testing?

MPereira-lp, 2017-11-06:
Yes, I am willing and interested in further testing. I will test the FlipMode and post the video.
I have both screens, the blue and the white versions. Do you want me to test the white ones?

You'r doing a great work and I look forward to compensate you. AT LEAST I will send you a couple of screens, boards, flat cables... wtv you need to have samples of this OLED screens. If you see fit another type of compensation that won't be a problem!

There are a couple o Issues I see at the moment. The screen sometimes goes OFF after a while (should that be a voltage issue? Maybe just the arduino malfunctioning?

The other Issue is the "brightness". I'm quite sure these screens have a lot more brightness, I'm quite sure where to change it, but not sure how. Can you help with that?
Sometimes after a while the screen looses brightness, like saving mode(?).

Will I be able to connect 2 screens per arduino?

MPereira-lp, 2017-11-06:
Ok, so [here is the video for the White version, and with the _Full_buffer/Graphic_test_ (16 bit mode)](https://youtu.be/eNjKchTHLFc)

olikraus, 2017-11-06:
> Yes, I am willing and interested in further testing. I will test the FlipMode and post the video.
I have both screens, the blue and the white versions. Do you want me to test the white ones?

I do not think that the color matters ;-)

> The screen sometimes goes OFF after a while

Hmm... could this be a power supply issue? How does your power supply look like at the moment?
 
> The other Issue is the "brightness".

Same topic: Power supply. But maybe we need to look at the commands to check whether there is any other option for this.

> Will I be able to connect 2 screens per arduino?

Yes of course.

olikraus, 2017-11-06:
> Ok, so here is the video for the White version, and with the Full_buffer/Graphic_test (16 bit mode)

hmm the link shows the u8x8/Graphics_test.

MPereira-lp, 2017-11-06:
>hmm the link shows the u8x8/GraphicsTest.

I'm quite sure it is the **Full_buffer**. I've also tried the **page_buffer** and they all look the same to me 😕 how can you tell the difference?

Anyway, [here](https://youtu.be/mTeQ30lpcZg) is the **page_buffer/FlipMode**

[21 more replies]