olikraus/u8g2 #401: using u8g2lib in custom arduino library

OH8KVA, 2017-10-23:
I'm trying to organize my software more easier to manage. I made a custom arduino library for display management. Alas it does not compile. Here is my code:

`
#ifndef _Display_h
#define _Display_h

#include <Arduino.h>
#include <Wire.h>
#include <U8g2lib.h>
#include <Model.h>

class Display {
private:
  Model const *_model;
  U8G2_SSD1306_128X64_NONAME_1_HW_I2C d(U8G2_R0);
  char _buf[20];
public:
  Display(Model const *model) {
    _model = model;
  };
  void init() {
    d.begin();
  };
  void display() {
    d.firstPage();
    do {
      d.setFont(u8g2_font_ncenB14_tr);
      d.drawString(0, 0, ultoa(_model->time, _buf, 10));
      d.drawString(0, 15, ultoa(_model->loopTime, _buf, 10));
    } while (d.nextPage());
  };
};

#endif`

I got a compile error:
`U8g2lib.h:52: In file included from
Display.h:13: from
Display.cpp:8: from
 
u8g2.h: 400:17: error: expected identifier before '(' token
   #define U8G2_R0 (&u8g2_cb_r0)
Display.h:19: note  in expansion of macro U8G2_R0
   U8G2_SSD1306_128X64_NONAME_1_HW_I2C d(U8G2_R0)
`

It compiles in .ino file, should it comple in cpp file too? I'm bit confused.

olikraus, 2017-10-23:
This is a C++ mistake and not a U8g2 problem. You should initialize U8G2_SSD1306_128X64_NONAME_1_HW_I2C class in your constructor:
http://en.cppreference.com/w/cpp/language/initializer_list

OH8KVA, 2017-10-23:
Yes, of course.
This works:

class Display {
private:
  Model const *_model; 
  U8G2_SSD1306_128X64_NONAME_1_HW_I2C d;
  char _buf[20];
public:
  Display(Model const *model): d(U8G2_R0) {
    _model = model;
  };
....

closed 2017-10-25 by olikraus