olikraus/u8g2 #402: sytax highlighting

m9365428, 2017-10-23:
Love the library but noticed a lack of a Keywords.txt
This makes keeping track of functions in the code a bit tedious.
Any chance of getting one added to the library. (If i get time I can work on it but I don't know how to look at the source so it wont be comprehensive.)

olikraus, 2017-10-24:
Yes, I know, it is missing for a long time. In fact I do not know and where to start. Any contribution would be great.

m9365428, 2017-10-24:
Ok just added a test keywords.txt with a good bit of info in the note.
Honestly its no where near complete but it works.
Load it up and checkout the code examples to see effect.

olikraus, 2017-10-24:
where can i find your file?

m9365428, 2017-10-24:
Sorry new to github thought my file would be visible.
Established new Pull request

olikraus, 2017-10-25:
Thanks for the pull request.

Added your comment here also for reference:
=======

Test run for keywords.txt
quick test using commands listed in https://github.com/olikraus/u8g2/wiki/u8g2reference
included one font in literals for testing.

Thought: If font codes where rewriten as u8g2.font.ncenB08.tr then fewer entries would be needed and it might simplify backend code.

method:

    add new line for keyword in proper section
    after keyword intert literal TAB not spaces. I edited in note pad and copied over.
    after TAB insert word type (KEYWORD 1, KEYWORD 2, KEYWORD3, LITERAL 1, LITERAL2)

more documentation on keyword.txt
https://gist.github.com/gschandler/8704468 (template)
https://www.arduino.cc/en/Hacking/LibraryTutorial (near the bottom)
https://spencer.bliven.us/index.php/2012/01/18/arduino-ide-keywords/ (what are the types)

olikraus, 2017-10-25:
todo: release script must copy keywords.txt to destination lib.

m9365428, 2017-10-25:
Ok awesome. Glad I could help.
Like I said it's not a completed file as it doesn't include all commands and arguments but it does establish the framework to build everything on.

olikraus, 2017-10-28:
keywords.txt is now included in the current beta release 2.20.2

olikraus, 2017-10-29:
works as expected... Thanks!

closed 2017-10-29 by olikraus