olikraus/u8g2 #404: MASSIVE SIZE

scargill, 2017-10-25:
Hi

I grabbed the u8G library to use an SSD1306 I2c display... and the demo program took an entirely reasonable 11k out of my 32K 328 - and a reasonable amount of RAM.

When I noticed this was no longer supported... I went for the u8g2 library and ran the similar demo... this took up an entirely unreasonable 22k and a lot more..

Is that the library or something in the demo?

Pete.

olikraus, 2017-10-25:
I need more information here. Which example/code did you test? Which constructor did you use?

olikraus, 2017-10-26:
any feedback?

scargill, 2017-10-26:
I used the I2c driver for the SSD1306 - and in both cases the graphic demo.  Do you by any chance still allocate a 1k RAM buffer when using the 32 pixel high display as against the 64 pixel high display??

olikraus, 2017-10-27:
The graphics demo occupies a lot of flash and can not be compared with the u8glib graphics demo.
For a 128x32 display vs 128x64 display only half of RAM would be required in full buffer mode. Page buffer is the same in both cases. 
I guess it is better to compare HelloWorld examples.
Nevertheless it is true that u8g2 requires more Flash (and RAM) because of the fact that u8g2 now uses standard Arduino libraries for bus communication (I2C, SPI).

olikraus, 2017-10-29:
> Do you by any chance still allocate a 1k RAM buffer when using the 32 pixel high display as against the 64 pixel high display??

No

closed 2017-11-24 by olikraus