olikraus/u8g2 #405: SSD1306 and MPU6050

danieldeesousa, 2017-10-25:
Hello,

I am working on a project which uses both a MPU6050 (https://www.dfrobot.com/wiki/index.php/6_DOF_Sensor-MPU6050_(SKU:SEN0142)) and a SSD1306 display (https://www.amazon.ca/Moudle-Serial-Display-Arduino-Raspberry/dp/B01N1IIX5J/ref=sr_1_1?ie=UTF8&qid=1508965115&sr=8-1&keywords=oled+arduino).

I am using the I2CDevlib and U8g2 libraries to control both of these modules.

I have tried using the modules at different times and they are working. Once I try to implement them together, they do not work. Could one of the libraries be recognising the wrong device.

Help?

Thank you.

Daniel De Sousa

olikraus, 2017-10-25:
Did you use the SW I2C constructor from u8g2? If so, you must change to HW I2C, see first question of the u8g2 FAQ:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

danieldeesousa, 2017-10-25:
Nevermind. The accelerometer is still not retrieving data. When I unplug it, it works

olikraus, 2017-10-26:
How can I support?

danieldeesousa, 2017-10-26:
Sorry for the confusion. Like I said, I am using the two libraries but it looks like there is confusion between the two devices. It looks like they are interpretting each others data and the Arduino is mixing up the signals

olikraus, 2017-10-26:
U8g2 follows the standard Arduino specification to allow cooperation of multiple devices. The only mistake from u8g2 side could be, that u8g2 changes transfer speed: You could try to comment the line here: https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L588

The other device could be confused because of that.

One more point: I2Cdev must be configured to use standard Wire.h lib. I mean, both lib can only cooperate if they use the same I2C lib. This is something you have to ensure.

danieldeesousa, 2017-10-26:
Thanks for your help. I will try that out. How would I ensure that this is happeninig?

olikraus, 2017-10-26:
> How would I ensure that this is happeninig?

you mean?

danieldeesousa, 2017-10-26:
How can I check if I2CDev is configured properly. They are not well documented like your library is and I am kind of stuck.

olikraus, 2017-10-26:
> How can I check if I2CDev is configured properly.

It has severl #ifdef and allows the selection different I2C libraries. If you have not modified anything compared to the repo it should default to Wire.h which should be ok.

danieldeesousa, 2017-10-26:
Hmm that is strange as I did not change anything in the library besides the frequency at which it sends information to the FIFO buffer

olikraus, 2017-10-26:
A complete other reason for the malfunction might be a overflow of the RAM area.

danieldeesousa, 2017-10-26:
Great idea. I do not think that is the problem as I have taken every measure to not allow this ie. using two page buffer ,etc. My RAM is also at 55% so I dont think there is a high chance of this being the problem.

closed 2017-11-24 by olikraus

jbrosowski, 2017-12-31:
Hi, I have the same problem here. @danieldeesousa , have you been able to solve it?

danieldeesousa, 2017-12-31:
Yes. Try using the hardware i2c configuration. As I did here:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0);

jbrosowski, 2018-01-01:
Thank you! And a happy new year!
I'm also using hardware i2c:
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
In other words: You are using the full frame variant, while I'm using the page buffer variant. I will test if I can get rid of some global variants, as switching to full frame rises memory allocation to 95%...

danieldeesousa, 2018-01-02:
Correct. Did it work for you?

jbrosowski, 2018-01-02:
I was not able to reduce the memory footprint of my program so far,  so I did a little test with a less complex example program: Yes, it worked. So the problem seems to be somewhere in the handling for the page buffer.