olikraus/u8g2 #406: Constructor call for PCD8544 error!

Yedwa, 2017-10-27:
Hi There

I have been trying to get the nokia 5110 lcd to work without any success.  I have tried all the constructor call for this display and I get the error as shown:

**HelloWorld:25: error: 'u8g2' was not declared in this scope
   u8g2.firstPage();
   ^
exit status 1
'U8G2_pcd8544_84X48_F_4W_HW_SPI' does not name a type**

I would appreciate assistance.

Regards,Yedwa

olikraus, 2017-10-27:
```
U8G2_PCD8544_84X48_F_4W_HW_SPI
```

Yedwa, 2017-10-28:
Hi Olivier

I have corrected the letters, the error when compiling was not resolved and below is the constructor I used.

U8G2_PCD8544_84X48_F_4W_HW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

It comes up with this error:

exit status 1
'U8G2_PCD8544_84X48_F_4W_HW_SPI' does not name a type

Thank you for your assistance

neu-rah, 2017-10-28:
the following is working for me and for that screen:

```c++
  #define U8_DC 9
  #define U8_CS 8
  #define U8_RST 7
  U8G2_PCD8544_84X48_1_4W_HW_SPI u8g2(U8G2_R0, U8_CS, U8_DC , U8_RST);
```

olikraus, 2017-10-28:
Yes, either use
```
U8G2_PCD8544_84X48_F_4W_HW_SPI u8g2(U8G2_R0, / cs=/ 10, / dc=/ 9, / reset=*/ 8);
```
or
```
U8G2_PCD8544_84X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 13, / data=/ 11, / cs=/ 10, / dc=/ 9, / reset=*/ 8);
```

This is documented here:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#pcd8544-84x48

Yedwa, 2017-10-28:
Thank you very much Olivier and Neu-Rah.

closed 2017-10-28 by Yedwa