olikraus/u8g2 #407: Request/Question how can I add a downloaded font?

D13tr1ch, 2017-10-28:
Hi! I've been using u8g2 for a few hours, I downloaded a font and I don't know how to add it so I can compile the code. 
Also I wanted to know, which font can I use if I need the character "ñ"? I'm currently using u8g2_font_smart_patrol_nbp_tf (so you can imagine the shape/size) and I need it to look like that font
Thank you.

olikraus, 2017-10-28:
The glyph "ñ" (unicode 0xf1) is included in all fonts ending in "f" (_tf, _mf, _hf) provided that the font author has created this glyph. So the font is included in the following fonts:

u8g2_font_4x6_tf
u8g2_font_5x7_tf
u8g2_font_5x8_tf
u8g2_font_6x10_tf
u8g2_font_6x12_tf
u8g2_font_6x13_tf
u8g2_font_6x13B_tf
u8g2_font_6x13O_tf
u8g2_font_7x13_tf
u8g2_font_7x13B_tf
u8g2_font_7x13O_tf
u8g2_font_7x14_tf
u8g2_font_7x14B_tf
u8g2_font_8x13_tf
u8g2_font_8x13B_tf
u8g2_font_8x13O_tf
u8g2_font_9x15_tf
u8g2_font_9x15B_tf
u8g2_font_9x18_tf
u8g2_font_9x18B_tf
u8g2_font_10x20_tf
u8g2_font_t0_11_tf
u8g2_font_t0_11b_tf
u8g2_font_t0_12_tf
u8g2_font_t0_12b_tf
u8g2_font_t0_13_tf
u8g2_font_t0_13b_tf
u8g2_font_t0_14_tf
u8g2_font_t0_14b_tf
u8g2_font_t0_15_tf
u8g2_font_t0_15b_tf
u8g2_font_t0_16_tf
u8g2_font_t0_16b_tf
u8g2_font_t0_17_tf
u8g2_font_t0_17b_tf
u8g2_font_t0_18_tf
u8g2_font_t0_18b_tf
u8g2_font_t0_22_tf
u8g2_font_t0_22b_tf
u8g2_font_profont10_tf
u8g2_font_profont11_tf
u8g2_font_profont12_tf
u8g2_font_profont15_tf
u8g2_font_profont17_tf
u8g2_font_profont22_tf
u8g2_font_profont29_tf
u8g2_font_mozart_nbp_tf
u8g2_font_glasstown_nbp_tf
u8g2_font_shylock_nbp_tf
u8g2_font_roentgen_nbp_tf
u8g2_font_calibration_gothic_nbp_tf
u8g2_font_smart_patrol_nbp_tf
u8g2_font_prospero_bold_nbp_tf
u8g2_font_prospero_nbp_tf
u8g2_font_balthasar_regular_nbp_tf
u8g2_font_balthasar_titling_nbp_tf
u8g2_font_synchronizer_nbp_tf
u8g2_font_mercutio_basic_nbp_tf
u8g2_font_mercutio_sc_nbp_tf
u8g2_font_miranda_nbp_tf
u8g2_font_nine_by_five_nbp_tf
u8g2_font_rosencrantz_nbp_tf
u8g2_font_guildenstern_nbp_tf
u8g2_font_astragal_nbp_tf
u8g2_font_crox1cb_tf
u8g2_font_crox1c_tf
u8g2_font_crox1hb_tf
u8g2_font_crox1h_tf
u8g2_font_crox1tb_tf
u8g2_font_crox1t_tf
u8g2_font_crox2cb_tf
u8g2_font_crox2c_tf
u8g2_font_crox2hb_tf
u8g2_font_crox2h_tf
u8g2_font_crox2tb_tf
u8g2_font_crox2t_tf
u8g2_font_crox3cb_tf
u8g2_font_crox3c_tf
u8g2_font_crox3hb_tf
u8g2_font_crox3h_tf
u8g2_font_crox3tb_tf
u8g2_font_crox3t_tf
u8g2_font_crox4hb_tf
u8g2_font_crox4h_tf
u8g2_font_crox4tb_tf
u8g2_font_crox4t_tf
u8g2_font_crox5hb_tf
u8g2_font_crox5h_tf
u8g2_font_crox5tb_tf
u8g2_font_crox5t_tf
u8g2_font_cu12_tf
u8g2_font_courB08_tf
u8g2_font_courB10_tf
u8g2_font_courB12_tf
u8g2_font_courB14_tf
u8g2_font_courB18_tf
u8g2_font_courB24_tf
u8g2_font_courR08_tf
u8g2_font_courR10_tf
u8g2_font_courR12_tf
u8g2_font_courR14_tf
u8g2_font_courR18_tf
u8g2_font_courR24_tf
u8g2_font_helvB08_tf
u8g2_font_helvB10_tf
u8g2_font_helvB12_tf
u8g2_font_helvB14_tf
u8g2_font_helvB18_tf
u8g2_font_helvB24_tf
u8g2_font_helvR08_tf
u8g2_font_helvR10_tf
u8g2_font_helvR12_tf
u8g2_font_helvR14_tf
u8g2_font_helvR18_tf
u8g2_font_helvR24_tf
u8g2_font_ncenB08_tf
u8g2_font_ncenB10_tf
u8g2_font_ncenB12_tf
u8g2_font_ncenB14_tf
u8g2_font_ncenB18_tf
u8g2_font_ncenB24_tf
u8g2_font_ncenR08_tf
u8g2_font_ncenR10_tf
u8g2_font_ncenR12_tf
u8g2_font_ncenR14_tf
u8g2_font_ncenR18_tf
u8g2_font_ncenR24_tf
u8g2_font_timB08_tf
u8g2_font_timB10_tf
u8g2_font_timB12_tf
u8g2_font_timB14_tf
u8g2_font_timB18_tf
u8g2_font_timB24_tf
u8g2_font_timR08_tf
u8g2_font_timR10_tf
u8g2_font_timR12_tf
u8g2_font_timR14_tf
u8g2_font_timR18_tf
u8g2_font_timR24_tf
u8g2_font_baby_tf
u8g2_font_chikita_tf
u8g2_font_lucasfont_alternate_tf
u8g2_font_p01type_tf
u8g2_font_robot_de_niro_tf
u8g2_font_trixel_square_tf
u8g2_font_fub11_tf
u8g2_font_fub14_tf
u8g2_font_fub17_tf
u8g2_font_fub20_tf
u8g2_font_fub25_tf
u8g2_font_fub30_tf
u8g2_font_fub35_tf
u8g2_font_fub42_tf
u8g2_font_fur11_tf
u8g2_font_fur14_tf
u8g2_font_fur17_tf
u8g2_font_fur20_tf
u8g2_font_fur25_tf
u8g2_font_fur30_tf
u8g2_font_fur35_tf
u8g2_font_fur42_tf
u8g2_font_osb18_tf
u8g2_font_osb21_tf
u8g2_font_osb26_tf
u8g2_font_osb29_tf
u8g2_font_osb35_tf
u8g2_font_osb41_tf
u8g2_font_osr18_tf
u8g2_font_osr21_tf
u8g2_font_osr26_tf
u8g2_font_osr29_tf
u8g2_font_osr35_tf
u8g2_font_osr41_tf
u8g2_font_logisoso16_tf
u8g2_font_logisoso18_tf
u8g2_font_logisoso20_tf
u8g2_font_logisoso22_tf
u8g2_font_logisoso24_tf
u8g2_font_logisoso26_tf
u8g2_font_logisoso28_tf
u8g2_font_logisoso30_tf
u8g2_font_logisoso32_tf
u8g2_font_logisoso34_tf
u8g2_font_logisoso38_tf
u8g2_font_logisoso42_tf
u8g2_font_logisoso46_tf
u8g2_font_logisoso50_tf
u8g2_font_logisoso54_tf
u8g2_font_logisoso58_tf
u8g2_font_pxplusibmvga9_tf
u8g2_font_pxplusibmvga8_tf
u8g2_font_px437wyse700a_tf
u8g2_font_px437wyse700b_tf

D13tr1ch, 2017-10-28:
Thank you so much!

closed 2017-11-01 by olikraus