olikraus/u8g2 #409: drawVLine works incorrectly on SSD1322 256x64 display

adbi68, 2017-10-29:
In library v2.12.6 and above, drawVLine (and other procedures - probably - using them like drawRFrame, drawFrame etc.) works incorrectly, displaying series of vertical lines (see photo). 

[image]

This works fine on v2.11.4, so probably problem is here (quote from changelog):
"* Enabled U8G2_HVLINE_SPEED_OPTIMIZATION (not sure why this was disabled)"
Please, correct this in next version. **By the time, can I change this by myself on my files, and how?**
I am using constructor:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI(rotation, cs, dc [, reset])
On ST7920 128x64 display, all works OK, even on last library version.

olikraus, 2017-10-29:
Did you enable U8G2_16BIT in u8g2.h?

https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L72

You can disable U8G2_HVLINE_SPEED_OPTIMIZATION in the same file by commenting the corresponding #define statement.

adbi68, 2017-10-29:
Thx for quick help!
Enabling U8G2_16BIT solves the problem.

closed 2017-10-29 by adbi68

olikraus, 2017-10-29:
:)