olikraus/u8g2 #412: Request - please add SSD1322_NHD_128X64

lorol, 2017-10-31:
@olikraus & group,  could you please add support for SSD1322_NHD_128X64, specs are [here, NHD-2.7-12864WD module](http://www.newhavendisplay.com/nhd2712864wdw3m-p-9546.html)

olikraus, 2017-11-01:
Did you try the SSD1322 NHD_256X64 constructor? How will it look like?
I will work on this, but it may take some time.

lorol, 2017-11-01:
Thanks for your answer @olikraus . 

I'd like to use it as U8g2_Arduino lib, but didn't find on that repository where to discuss.
You are right, these newest SSD1322-based NHD OLED modules may be covered by your existing SSD1322 NHD_256X64. I have some uncertainties with the init, the columns offset and like to see grayscale feature. Let me play a bit and will report my findings & challenges.

Dash095, 2017-12-05:
Hello,

I currently have the same issue. 

The SSD1322 NHD_256X64 constructor appears to initialize the screen. However, there seem to be some rotation/mirroring and scaling problems when using it to drive a 128x64 display. The rotation/mirroring could be worked around using certain functions of the u8g2 library (though its not ideal), however, the scaling cannot. The scaling issue appears to be basically that in the horizontal direction, 256 pixels are used to describe the actual 128 pixels available. Meaning that drawing a 128x64 square only covers half the screen (appears as 64x64). This is especially annoying when using the library to print text. 

I was looking around in the code for the SSD1322 driver and found 2 functions that caught my eye:

`uint8_t u8x8_d_ssd1322_common(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
static const u8x8_display_info_t u8x8_ssd1322_256x64_display_info`

I think with some modifications to these functions, the driver might be usable (especially since the initialization seems to be working fine). However, I'm not sure where to start. 

Has anyone made progress with this issue thus far? or perhaps has some information which could help me getting started. Any help is much appreciated!

lorol, 2017-12-05:
Hi @Dash095 
I am glad you started working on this :) as well. I had no time to tap in. 
As a hint, you can compare the specs of some 256x64 model and the new one. I see that init:

`Set_Remap_Format(0x14); //was given as example on the 256 model`

```
writeCommand(0xA0); //set re-map  in sow on the new, 128 one
writeData(0x16);
writeData(0x11);
```

Also the horizontal addressing on new display has offset 28!
See this:

```
void setColumn(unsigned char xStart, unsigned char xEnd){ 
    writeCommand(0x15); //set column (x-axis) start/end address 
    writeData(xStart); //column start; 28 is left-most column 
    writeData(xEnd); //column end; 91 is right-most column
 }
```

Good luck!

olikraus, 2017-12-08:
I found some time to add support for this display. Is there someone who could test the new code?

Please download beta 2.20.8 from here: https://github.com/olikraus/U8g2_Arduino
Add the zip file to your Arduino IDE via the add zip library menu of the Arduino IDE.

What will be the output of u8x8/FlipMode example?

The new constructors are:
```
//U8X8_SSD1322_NHD_128X64_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8X8_SSD1322_NHD_128X64_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

U8G2_SSD1322_NHD_128X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_SSD1322_NHD_128X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

conectia, 2017-12-08:
I have tested the new code.

The driver still maps the 128x64 drawing into 64x64 pixels at the display.

The line goes from (0,0) to (128,64)
[image]

conectia, 2017-12-08:
u8g2 FlipMode output
[image]

olikraus, 2017-12-08:
thanks for the pictures, I created 2.20.9. Can you test again?

conectia, 2017-12-08:
No luck :(

It appears duplicated, both parts updated on each refresh
[image]

olikraus, 2017-12-09:
I have create a new release 2.20.10.
Looks like that NHD had selected a very strange architecture. To set one pixel, u8g2 needs to transfer 8 bit. Please have a look now.

Download: https://github.com/olikraus/U8g2_Arduino

conectia, 2017-12-09:
Much better !
Still some horizontal offset.

[image]

conectia, 2017-12-09:
The left hand side was created in the previous run ... Display persistence.

olikraus, 2017-12-09:
ok, another test 2.20.11
Can you send me the first page of the u8x8 (!) FlipMode example? Thanks

conectia, 2017-12-09:
Perfect!
Many, many thanks.

[image]

conectia, 2017-12-09:
This is u8x8 output. When flip mode 1 it appears correctly upside down.

[image]

olikraus, 2017-12-09:
Thanks, sounds good. Any other visible problems? Otherwise I think it is done...

Again: Due to a somehow strange internal GDRAM mapping, the display will be not so fast.

lorol, 2017-12-09:
Thank you guys! Perfect! 
Closing shortly.

conectia, 2017-12-09:
Everything OK.

I think they left unconnected odd columns pins in the board. That's why a full byte is required for a pixel.

olikraus, 2017-12-09:
closing this for now... feel free to reopen if there are any issues.

closed 2017-12-09 by olikraus

lorol, 2017-12-13:
All good, I tested, too!
If someone is interested, I also made a [font gray-scale hack](https://github.com/olikraus/u8g2/issues/421#issuecomment-351194809) for this dispay