olikraus/u8g2 #413: Arduino Uno, SSD1306 128x64, I2C - display showing artifacts & numbers

nerdiary, 2017-11-01:
Arduino Uno, SSD1306 128x64, I2C - display showing artifacts & numbers
Running u8x8 hello world program, see photo
Tried all the relevant constructors all have the same issue
[image]

olikraus, 2017-11-01:
Strange. Can you post the code which produces the picture above.

nerdiary, 2017-11-03:
I'm just using the example u8x8 hello world program. I have tried the following constructors and all have the same result:
- U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);
- U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE)

I verified that the OLED hardware was OK by uploading the same program to an ATTiny85. Works fine with no artifacts.

olikraus, 2017-11-04:
I have never seen something like this before :-(

closed 2017-12-07 by olikraus