olikraus/u8g2 #414: Reduce font size

metropt, 2017-11-02:
I'm running out of space so I need to reduce the fonts to match only what I'm using. Per example I use one of the fonts only for numbers. Editing the u8g2_fonts.c I resize the font I'm using for numbers until it can't show the number however it is not practical.

Reading your code I end up here:

```
void u8g2_read_font_info(u8g2_font_info_t *font_info, const uint8_t *font)
{
  /* offset 0 */
  font_info->glyph_cnt = u8g2_font_get_byte(font, 0);
  font_info->bbx_mode = u8g2_font_get_byte(font, 1);
  font_info->bits_per_0 = u8g2_font_get_byte(font, 2);
  font_info->bits_per_1 = u8g2_font_get_byte(font, 3);
  
  /* offset 4 */
  font_info->bits_per_char_width = u8g2_font_get_byte(font, 4);
  font_info->bits_per_char_height = u8g2_font_get_byte(font, 5);
  font_info->bits_per_char_x = u8g2_font_get_byte(font, 6);
  font_info->bits_per_char_y = u8g2_font_get_byte(font, 7);
  font_info->bits_per_delta_x = u8g2_font_get_byte(font, 8);
  
  /* offset 9 */
  font_info->max_char_width = u8g2_font_get_byte(font, 9);
  font_info->max_char_height = u8g2_font_get_byte(font, 10);
  font_info->x_offset = u8g2_font_get_byte(font, 11);
  font_info->y_offset = u8g2_font_get_byte(font, 12);
  
  /* offset 13 */
  font_info->ascent_A = u8g2_font_get_byte(font, 13);
  font_info->descent_g = u8g2_font_get_byte(font, 14);
  font_info->ascent_para = u8g2_font_get_byte(font, 15);
  font_info->descent_para = u8g2_font_get_byte(font, 16);
  
  /* offset 17 */
  font_info->start_pos_upper_A = u8g2_font_get_word(font, 17);
  font_info->start_pos_lower_a = u8g2_font_get_word(font, 19); 
  
  /* offset 21 */
#ifdef U8G2_WITH_UNICODE
  font_info->start_pos_unicode = u8g2_font_get_word(font, 21); 
#endif
}
```

what should help me reduce one font only to letters A to z.

How can I know from the information above where the glyphs start, at 23 byte?
For each glyph how do I know how many bytes it take?
I want to know that so I can delete all the glyphs before A to z and change these two values  ```
font_info->start_pos_upper_A = u8g2_font_get_word(font, 17);
font_info->start_pos_lower_a = u8g2_font_get_word(font, 19);
```
accordingly.

Thank you

olikraus, 2017-11-02:
> I'm running out of space so I need to reduce the fonts to match only what I'm using.

Which font do you use? Can you provide (part of) your code?
Do you use more than one font? You have separate fonts for numbers and text?

>  Editing the u8g2_fonts.c

really? 

> I resize the font I'm using for numbers

hmm? You can you do this. Even I would not be able to do this. All fonts are bitmaps.

> How can I know from the information above where the glyphs start, at 23 byte?

yes

> For each glyph how do I know how many bytes it take?

What exactly do you plan?  I think it is better to start from your code (not from u8g2). Please tell me which font you use in your code.

metropt, 2017-11-03:
I realized that there are 3 versions for each font, one with all the characters, one with letters+numbers and one only with numbers -.-' so my problem is resolved :)

closed 2017-11-03 by metropt

olikraus, 2017-11-04:
:-)