olikraus/u8g2 #417: If/Else statements with u8g2.print

Yedwa, 2017-11-08:
Hi Olivier

I am trouble printing on Nokia 5110 with u8g2.print function when used inside If/Else statement. 
May you be so kind to share an example of how this can be done successfully.
Thank you for your assistance.
Regards, Yedwa

olikraus, 2017-11-09:
Please attach you "none"-working code.

Yedwa, 2017-11-09:
The code was first written by Joop Brokking (http://www.brokking.net/imu.html) using liquid-crystal display.

void write_LCD(){
 
  if(lcd_loop_counter == 14)lcd_loop_counter = 0;                      
  lcd_loop_counter ++;                                             
  if(lcd_loop_counter == 1){
    angle_pitch_buffer = angle_pitch_output * 10;                       
    u8g2.setCursor(25, 28);                                            
  }
  if(lcd_loop_counter == 2){
    if(angle_pitch_buffer < 0) u8g2.print("-");                         
  }else u8g2.print("+");                                              

  if(lcd_loop_counter == 3)u8g2.print(abs(angle_pitch_buffer)/1000);    
  if(lcd_loop_counter == 4)u8g2.print((abs(angle_pitch_buffer)/100)%10);
  if(lcd_loop_counter == 5)u8g2.print((abs(angle_pitch_buffer)/10)%10); 
  if(lcd_loop_counter == 6)u8g2.print(".");                             
  if(lcd_loop_counter == 7)u8g2.print(abs(angle_pitch_buffer)%10);      

  if(lcd_loop_counter == 8){
    angle_roll_buffer = angle_roll_output * 10;
    u8g2.setCursor(25,38);
  }
  if(lcd_loop_counter == 9){
    if(angle_roll_buffer < 0)u8g2.print("-");                           
    else u8g2.print("+");                                               
  }
  if(lcd_loop_counter == 10)u8g2.print(abs(angle_roll_buffer)/1000);    
  if(lcd_loop_counter == 11)u8g2.print((abs(angle_roll_buffer)/100)%10)
  if(lcd_loop_counter == 12)u8g2.print((abs(angle_roll_buffer)/10)%10); 
  if(lcd_loop_counter == 13)u8g2.print(".");                            
  if(lcd_loop_counter == 14)u8g2.print(abs(angle_roll_buffer)%10); 
}

olikraus, 2017-11-09:
I do not see an issue here. Which constructor do you use?

Yedwa, 2017-11-09:
U8G2_PCD8544_84X48_F_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);     // Nokia 5110 Display

In the code when I did not include  u8g2.clearBuffer(); or u8g2.sendBuffer(); the lcd remain blank, however when I included u8g2.sendBuffer(); as shown below, the lcd updated partially.

  if(lcd_loop_counter == 14)lcd_loop_counter = 0;                     
  lcd_loop_counter ++;                                                 
  if(lcd_loop_counter == 1){
    angle_pitch_buffer = angle_pitch_output * 10;                      
    u8g2.setCursor(25, 28);
    u8g2.sendBuffer();

I am thinking my problem has to do with clearbuffer or sendbuffer which is missing in each loop counter.

closed 2017-11-09 by Yedwa