olikraus/u8g2 #418: Seeedstudio Grove - OLED Display 1.12" not working

joaodafonseca, 2017-11-11:
Hi,

I'm trying to run the examples of the library with the Seeedstudio OLED Display 1.12" but I only get glichy characters on the screen! 

I'm using the following constructor:
U8X8_SSD1327_SEEED_96X96_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);  // Seeedstudio Grove OLED 96x96

and using the pins 18 and 19 for SDA and SCL.

Any ideas? thanks in advance
João

joaodafonseca, 2017-11-11:
Looks like they use a new chip on the modules, SH1107G.
Any change that it's compatible with this library?

olikraus, 2017-11-11:
> Any change that it's compatible with this library?

Is there a link to the SH1107G controller datasheet?

joaodafonseca, 2017-11-11:
here: 
https://github.com/SeeedDocument/Grove_OLED_1.12/raw/master/resources/SSD1327_datasheet.pdf

thanks for looking into it!
J

olikraus, 2017-11-12:
Unfortunately this is NOT the sh1107 Datasheet.

joaodafonseca, 2017-11-12:
hmmm, found it in the oled wiki page:
http://wiki.seeed.cc/Grove-OLED_Display_1.12inch/

olikraus, 2017-11-12:
Yes, but for the any update I need to know how to program the SH1107. This is not mentioned in any documents of the seeed home page.

The SH1107 datasheet should be provided from "sino wealth". It will contain the command table of the SH1107 controller.

olikraus, 2017-11-12:
The SH1107 is not even mentioned on the vendor home page. Maybe it is too new.

http://www.sinowealth.com/zh/product.asp?cat_name=OLED&cat_id=35&class_name=OLED%20%C7%FD%B6%AF&class_id=100&from_page=top

joaodafonseca, 2017-11-13:
The library provided by seeed already works with the SH1107. Not sure if it helps!!!

https://github.com/Seeed-Studio/OLED_Display_96X96

olikraus, 2017-11-13:
Well, partly... For example this code is very strange:
https://github.com/Seeed-Studio/OLED_Display_96X96/blob/master/SeeedGrayOLED.cpp#L290

Without knowing the datasheet of the controller it is difficult to guess what the software does.

joaodafonseca, 2017-11-13:
yeah I see...what about this? 
https://www.crystalfontz.com/controllers/SolomonSystech/SSD1327/

olikraus, 2017-11-14:
What about what? I have the SSD1327 datasheet on my harddisk and the SSD1327 is well supported. But for support of the SH1107 I need a command description of the SH1107, which is in the datasheet of the SH1107.  That was your request, right? Support for the SH1107?

olikraus, 2017-12-07:
No response, closing this. I need a datasheet of the SH1107.

micooke, 2017-12-12:
@olikraus - try this, i received it today. I plan to host it on github at some stage, but till then:

https://app.box.com/s/45x28nhpp5bik19ci0q9e5r8abr7bdxo

olikraus, 2017-12-12:
Wow, yes, this is the information i was looking for. Is there still interest for supporting the SH1107?

micooke, 2017-12-12:
There certainty is from me 😊

olikraus, 2017-12-12:
well, then... would you also do some tests for such a device?

micooke, 2017-12-12:
I would be happy to, but mine is slightly different - I have a 64x128 oled that i can test

olikraus, 2017-12-12:
ok, so there is a 96x96 seeedstudio and a 64x128 oled.
For adding a display to u8g2, i need someone for testing, so i would start for the 64x128 oled...

micooke, 2017-12-14:
@joaodafonseca - if half your screen is garbled (like mine was) try adjusting the screen offset - 0xD3 command. A value of 0x60 (taken from my screen spec sheet) worked for me. 

@olikraus - the driver is not very different from the SSD1306. I needed to add the offset above, and change my display mode (x20 command)/routine to use the equivalent of what is 'paged' on the SSD1306. The SH1107 only supports Horizontal/Vertical with Column/Page but not Page/Column incrementing (respectively).

I also needed to set the SH1107 specific DC-DC variable setting. This chipset is external Vpp source only.

I modified a SFE library i had previously worked on and will push the changes in a day or so if you are interested.

olikraus, 2017-12-14:
u8g2 includes a semiautomatic process for the creation of new drivers, so pull requests are a little bit difficult: Pull requests need to update https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c

I hope i finde some time for this soon (maybe today evening)

micooke, 2017-12-14:
No problem, ill leave the PR then. Hopefully ive helped anyway :) . I need to port it to the nrf52832 for testing so ill look at that tonight/tomorrow.

olikraus, 2017-12-14:
I have created a new device for the SH1107 as a copy of the SSD1306 controller:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c

The u8x8 constructor is:

U8X8_SH1107_128X64_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

The u8g2 page mode constructor will look like this:
U8G2_SH1107_128X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

So now the question is, which changes are required for the SH1107?

olikraus, 2017-12-14:
I have added another device for the seeed 96x96 display: U8X8_SH1107_SEEED_96X96_4W_HW_SPI

I have only the init sequence for the SEEED 96x96, so i will start for this.

olikraus, 2017-12-14:
ok, i modified init sequence for the SEEED 96x96 constructor. I have not yet added the 0xd3 0x60 offset. In fact i have  not seen this in the datasheet.

How shell we continue? Shell i create a u8g2 beta zip lib for testing with Arduino IDE?

To continue I need this:
What is the init sequence for the SH1107 64x128 OLED? 
Is there someone who can test the SH1107 96x96?

Thanks, Oliver

micooke, 2017-12-14:
Hi Oliver, a zip would be excellent! I will eventually need to port it to the nrf52832 as that is the end application (my screen on a smart watch). Today though it's under that microscope to break it out for an nano.

Here's a link to the 64x128 datasheet
https://app.box.com/s/xvz5l18jm5mvqt9ifowasuk6brpd75t6

olikraus, 2017-12-14:
I have zip file 2.21.0 for download here:
https://github.com/olikraus/U8g2_Arduino

The init sequence for the SH1106_128x64 is still not yet modified, but I have added example constructors to all examples.

micooke, 2017-12-15:
No dice. Your initialisation script is quite different to mine - https://github.com/micooke/sparkfun_OLED/blob/master/src/SFE_MicroOLED.h#L512-L564

I'm sure a lot of the setting differences are neither here nor there, but there are a few commands i didnt recognise, and it looked like the dc-dc mode wasnt setup i.e. 
https://github.com/micooke/sparkfun_OLED/blob/master/src/SFE_MicroOLED.h#L563-L564

I tested it with a 3v3 Sparkfun Pro Micro. I hate those temperamental little things, but it worked with my setup code, and worked with your code and a generic i2c oled so the hookup is correct.

olikraus, 2017-12-16:
I have updated the init sequence for the U8G2_SH1107_128X64 device (e.g. U8G2_SH1107_128X64_1_4W_HW_SPI ).
Will it work for your display? Can you send me a picture how the HelloWorld example looks like?

U8g2 2.21.1 zip file is here: https://github.com/olikraus/U8g2_Arduino

micooke, 2017-12-19:
@olikraus - sorry for my silence on this. It didn't work, which isn't overly helpful, so I'm debugging it at the moment.
And the problem with that is that i broke my own library, which should be fixed by tomorrow.

Ill let you know what i find. One thing i did notice is that the screen is actually portrait, not landscape so it is 64x128. Cheers

micooke, 2017-12-22:
I hooked up the saleae - there is no traffic on the spi bus for this configuration e.g. ```U8G2_SH1107_128X64_1_4W_HW_SPI``` or ```U8G2_SH1107_128X64_2_4W_HW_SPI```

olikraus, 2017-12-22:
Hmm... SPI bus should be busy. Which board do you use?

micooke, 2017-12-22:
I normally use a nrf52832 based board, but just using a ATmega328 for testing. I tried using a SSD1306 HW SPI variant - no issues. My goto is a 128x64 i2c, no issues with that either. I'm just running the HelloWorld sketch, nothing else, programming over serial.
Ive tried on the Arduino Pro Micro and Nano, also a Arduino Leonardo and SFE Micro. I only used the logic analyser on the pro micro, and i saw traffic using the SSD1306 HW SPI example.

I don't understand enough about your code structure to debug it, so hopefully its a minor thing.

The setup code is a little incorrect because its 64x128 not 128x64, and either its missing display on or it is done outside initialisation? I didn't look to heavily for it - it was easier to just update the initialisation and check it.

micooke, 2017-12-22:
I wont be able to do much testing over the holiday break, so no rush on this. Great work on this library btw - thankyou @olikraus for your extensive display and font support, and for your responsiveness!

olikraus, 2017-12-22:
hmmm... regarding the low level interface, there is no difference to the SSD1306 display.
See here:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L649
or here:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L691

I have not seen a single picture of the sh1107 with u8g2. So it is difficult to fix the 64x128 issue.

micooke, 2017-12-22:
You should be able to replicate my issue, just try the HelloWorld sketch and look at MOSI, SCK pins without a screen attached. I didnt see any signalling at all for the SH1107, but did for the SSD1306 which is quite wierd as (as you pointed out) the instantiation is standardised across the different IC types.

I know it doesn't help, but here is a picture i took a while ago with the offset issue  https://app.box.com/s/po1eq5uj5u2isyg6dx2cidvutozr2rhm using Screentest example with my modified SFE library https://github.com/micooke/sparkfun_OLED/blob/master/examples/MicroOLED_Screentest/MicroOLED_Screentest.ino

The offset issue was/is wierd, as the value you change is for Y offset.

olikraus, 2017-12-23:
ok, I will check this in my lab.

olikraus, 2017-12-23:
ok, there was a mistake in the example itself.

U8g2 2.21.2 zip file is here: https://github.com/olikraus/U8g2_Arduino

micooke, 2018-01-17:
Sorry for the delay, and thanks for the update. With a little tweaking i get the top half of hello world on the screen, i need to investigate more and get back to you.

micooke, 2018-01-17:
Hi @olikraus - im a little stuck..
I modified the 128x64 so that the info is 64x128, the sync is 7f (instead of 3f), and it is using flipmode. I get something, but im stuck. Check out Ug8w
2_Library.jpg in the follow folder
https://app.box.com/s/po1eq5uj5u2isyg6dx2cidvutozr2rhm

The output is after running my modified SFE library on the host chip (NRF52). Part of this routine is creating a box at the edge of the display. Your library is writing to roughly half the display. Ill push my updates to my fork after a little more investigation. 

Cheers, Cookie

olikraus, 2018-01-18:
> Check out U8g2_Library.jpg in the follow folder
https://app.box.com/s/po1eq5uj5u2isyg6dx2cidvutozr2rhm

This is a really surprising output. Why is the lower part of the text missing? 
Can you create a picture of the u8g2 logo on your display?

Your SFE lib seem to write a correct border. Is it possible to see the code for this? I might learn how the display expects page and x column information.

micooke, 2018-01-19:
https://github.com/micooke/sparkfun_OLED

I just realised i forgot to switch the tile height, width - it now blanks the whole screen. 
I tried the logo - only the top is written,same as the font.

olikraus, 2018-01-19:
> https://github.com/micooke/sparkfun_OLED

Is this the link to the code for the OLED? Can you point out detailed file?

micooke, 2018-01-19:
https://github.com/micooke/sparkfun_OLED/blob/master/src/SFE_MicroOLED.h

And this is my example code
https://github.com/micooke/sparkfun_OLED/blob/master/examples/MicroOLED_Screentest/MicroOLED_Screentest.ino

A little confusing as it allows arbitrary screen sizes, and the SSD1306 as well as the SH1107. 

To explain the example defines; my sh1107 screen is attached to a T28 smartwatch. Ive broken out the SPI pins so that i can drive it with an external 3.3v pro mini ATmega328p. This is why there is #define switches for T28 and external.

The main two functions in the library header are the setup and display functions. Class declaration and elaboration are all in 1 file

olikraus, 2018-01-20:
Thanks for the code hints.
We talk about a 64x128 SH1107 OLED, right?

So in this loop,

https://github.com/micooke/sparkfun_OLED/blob/master/src/SFE_MicroOLED.h#L878

_lcdHeight (which is 128) loops through all pages, which are 128/8 = 16 pages, correct?

micooke, 2018-01-20:
Correct on all accounts, except my setup is spi so lhttps://github.com/micooke/sparkfun_OLED/blob/master/src/SFE_MicroOLED.h#L892

It looked like this is exactly what your library does. Unlike the Hoz mode on the SSD1306 that auto increments the line, the SH1107 only has page mode. Your display callback looks like it was designed for page mode anyway, just like mine.

olikraus, 2018-01-21:
hmmm i am somehow stuck... I guess I once need to buy such a display

micooke, 2018-01-21:
Hold off for a little bit. I've got a few more things to test first. The screens are about $12AUD from AliExpress including shipping - im certainly happy to pay your cost if you have a bpay.

micooke, 2018-01-22:
All sorted.
I was accidentally testing against the full_buffer examples, not the page_buffer examples. 

Results are here:
https://app.box.com/s/po1eq5uj5u2isyg6dx2cidvutozr2rhm

micooke, 2018-01-23:
@olikraus - please see cd2bc532dbdc238dea73ed929bba06afc28295f2 for the changes. Do you want a PR? I don't see many so didn't know your policy.

micooke, 2018-01-29:
@joaodafonseca - changes have been merged, time for this issue to be closed 😊

olikraus, 2018-01-29:
But i need to do some tests with the patch :-/

micooke, 2018-01-29:
Whoopsie, sorry - jumped the gun there

olikraus, 2018-01-29:
I applied codebuild script... Further testing needs to be done.

olikraus, 2018-02-02:
can you download U8g2 2.21.6 from here:
https://github.com/olikraus/U8g2_Arduino

Install via Arduino IDE (add zip lib menu). Will the SH1107 support work? Thanks...

micooke, 2018-02-08:
Hey @olikraus, sorry for the delay (i've been away).
I tested v2.21.7 and its all hunky dory 👍

olikraus, 2018-02-08:
Sounds good. I also did some general lib tests, so all looks good.

olikraus, 2018-02-13:
So... I hope all is done here and I did not miss anything... closing...

closed 2018-02-13 by olikraus

watershade, 2018-06-20:
I am using seeduino(like Arduino Uno) to test Grove-OLED 1.12''. The hardware version is 2.2. It uses SH1107G as driver but screen change to 128*128. And address is 0x3C. I try to use your lib to test but nothing show in my OLED.  I have test official library form [Here](https://github.com/Seeed-Studio/OLED_Display_96X96). It is too simple and didn't support V2.2 well.  But I can use this to test hardware and I find OLED works well.

olikraus, 2018-06-20:
Which constructor did you use? Which u8g2 version?

watershade, 2018-06-20:
I use platformio to test it. In platformio library, the newest version of u8g2 is 2.23.8.

[image]

[3 more replies]