olikraus/u8g2 #419: how to write in Russian???

timurufa86, 2017-11-13:
Good afternoon!
I cannot understand how to write on cyrillic... I use the u8g2_font_cu12_t_cyrillic font. This command I set the u8g2.setFont(u8g2_font_cu12_t_cyrillic) font; can somewhere else what that needs to be added?
Tell me please!

olikraus, 2017-11-13:
There are some additional requirements:
- Your Editor must support UTF-8 (I think this is true for the Arduino IDE)
- You must use drawUTF8 (https://github.com/olikraus/u8g2/wiki/u8g2reference#drawutf8) or enable UTF8 for the print command (https://github.com/olikraus/u8g2/wiki/u8g2reference#enableutf8print).

timurufa86, 2017-11-24:
hi!!!
  I tried an example where hi the world on Chinese - in it the Chinese eroglifa are not displayed too

timurufa86, 2017-11-24:
you can give me a code working for example with a text "Привет мир"

olikraus, 2017-11-24:
void setup(void) {
  u8g2.begin();
  u8g2.enableUTF8Print();		// enable UTF8 support for the Arduino print() function
}

void loop(void) {
  u8g2.setFont(u8g2_font_cu12_t_cyrillic);  // use chinese2 for all the glyphs of "你好世界"
  u8g2.firstPage();
  do {
    u8g2.setCursor(0, 40);
    u8g2.print("Привет мир");
  } while ( u8g2.nextPage() );
}

closed 2017-12-06 by olikraus

kewin2013, 2018-03-01:
Everything done as written here, but Cyrillic is not displayed. What can be?
```
#include <Arduino.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ A4, /* data=*/ A3, /* cs=*/ A0, /* dc=*/ A1, /* reset=*/ A2); 

void setup(void) {
  u8g2.begin();
  u8g2.enableUTF8Print();    // enable UTF8 support for the Arduino print() function
}

void loop(void) {
  u8g2.setFont(u8g2_font_cu12_t_cyrillic);  
  u8g2.firstPage();
  do {
    u8g2.setCursor(0, 40);
    u8g2.print("Привет мир");
  } while ( u8g2.nextPage() );
}
```

olikraus, 2018-03-01:
does the HelloWorld example work?
I mean, does the display work correctly?

timurufa86, 2018-03-02:
Hello! An example from the box does not work, but if you play with the
fonts, you can find a working ...

kewin2013, 2018-03-02:
Код заработал, теперь отображается кириллица

Dmitrykrz, 2019-06-30:
> Код заработал, теперь отображается кириллица

Что надо было поменять чтобы заработал?

> does the HelloWorld example work?
> I mean, does the display work correctly?
Screen works fine. I can see english text on the screen but only ??? for russian text. I have tried all cyrrilic fonts and examples from this thread.

olikraus, 2019-06-30:
Do you use the Arduino IDE?

Dmitrykrz, 2019-06-30:
> Do you use the Arduino IDE?

Microsoft Visual Studio with vMicro

olikraus, 2019-06-30:
Great. UTF8 editor support is required for this feature.