olikraus/u8g2 #420: Can you add support for SH1122 driver 256x64?

commandojay1, 2017-11-14:
Specifically I am using Wisechip UG-5664GSWHG01, and using the SPI interface.

Thanks

olikraus, 2017-11-14:
Is the Datasheet available for the sh1122?

commandojay1, 2017-11-14:
Should be this one.
http://www.displayfuture.com/Display/datasheet/controller/SH1122.pdf

The mechanical drawing of the display is here:
http://ultran.ru/sites/default/files/catalog/svetodiody/brend/datasheets/ug-5664gswhg01.pdf

In the full display datasheet I have it has the same mechanical drawing as above. The rest of the datasheet is electrical specs and says to refer to the SH1122 datasheet for the command set. I'm not comfortable posting the full display datasheet since it is impossible to find on the web and I received it directly from Wisechip and marked as confidential.
Two other pages of the datasheet are listed openly here but doesn't give you more than what the SH1122 datasheet gives.
https://www.alibaba.com/product-detail/sunlight-readable-2-inch-2-08_60516251929.html

olikraus, 2017-11-15:
It is impossible to implement support for this display, without knowing the command set.
Also note, that u8g2 only suppots monochrome mode.

olikraus, 2017-12-30:
looks like the datasheet is available here: http://www.displayfuture.com/Display/datasheet/controller/SH1122.pdf

shell I start the implementation for this display? I need a person who does the testing....

olikraus, 2018-02-13:
sooo.... how to continue here....?

commandojay1, 2018-02-13:
I hadn't responded because of "life". I didn't have a PCB to connect the display ribbon cable to yet. The board is made and I'll be ordering some next week. At that point I'd be able to test what you make.

olikraus, 2018-02-14:
hmmm... I just hope you do not rely on me to provide the display driver... this is just a open source lib...

commandojay1, 2018-02-14:
I'm just starting out, I'll take what I can get. I was looking for initialization of the display as my first step.

olikraus, 2018-02-14:
Ok...

commandojay1, 2018-03-10:
I have the bare PCB. Hope to have the parts this week, and assembly of the main board with the main controller (esp32).
[image]

olikraus, 2018-03-10:
Looks good

lcdmaker, 2018-03-12:
I have worked in LCD & LCMs over 10 years,welcome to contact me 
skype ivy5731
ivy.q@lcdmaker.com

olikraus, 2018-03-24:
Note for myself: SH1122 256x64 OLED ordered.

olikraus, 2018-04-12:
added some code, but still a lot of work required

olikraus, 2018-04-13:
implemented

[image]

closed 2018-04-13 by olikraus

olikraus, 2018-04-14:
I ordered just a display with the same chipset. I think the major part of the work is done for your display, Please reopen, if it doesn't fully work.

olikraus, 2018-04-14:
Beta Release: https://github.com/olikraus/U8g2_Arduino

commandojay1, 2019-04-29:
FYI, it works. Using your library along with Neil Kolban's mapping code for the ESP32 using ESP-IDF framework, I was able to get it working. It only took me 1 year to test it (well 10 months of doing nothing, 1 month of getting other peripherals working first, then 2 days on the display).
Thanks!

olikraus, 2019-04-29:
great, thanks for the feedback after such a long time :-)

iggymayer, 2019-06-19:
Is your SH1122-UG-5664GSWHG01 lit normally on the ESP32? My screen doesn't light up on ESP32-DevKitC, and there are no problems with the screen and the circuit.

commandojay1, 2019-06-19:
Well mine is an OLED, so there is no backlight. You have this specific screen? See the attached files for how I connected mine. It's in three images since the OLED sits on a daughter board. Maybe you'll see something that can help.

[image]
[image]
[image]

iggymayer, 2019-06-21:
Thank you, the hardware design should be no problem, I will try to change to IIC.

zeynepdicle, 2023-06-15:
Hİ @commandojay1,
I am experiencing issues working with SH1122 on ESP-IDF, and I'm unsure how to resolve them. Can you assist me?
Best regards,
Zeynep Dicle

commandojay1, 2024-12-06:
@zeynepdicle,

Obviously it's been a long time, but just in case, The link below goes to the project zip. I'm not an expert at Git and not sure what files to exclude, so I put all except the build folder to make sure you have everything. . I rebuilt and tested it using Espressif-IDE and framework 5.3.1-2. With that setup, opening Projects from File System and selecting the top folder should load the project and it should build successfully. Running it will depend on your SPI and other pin setup, and the exact screen (dimensions) you are using. The code gives the output of this image.
[image]

Dropbox of zip file: https://www.dropbox.com/scl/fi/l4ceykonlxmuc27vr2gb8/SH1122.zip?rlkey=j6cl2f545thv6hamx9r9310kg&st=4n6dvi2z&dl=0