olikraus/u8g2 #421: Grayscale

lorol, 2017-11-14:
Hi,
Is there an universal Gray-scale support for monochrome displays with such feature?
Example OLEDs based on SSD1325 SSD1322 can have a nice effects like this 
(photo of real 1325-based screen from another project - a string can be displayed with given brightness, say 4 levels ):

[image]

SimGas, 2017-11-15:
There is not. I think there was a grayscale support in the old u8glib, but due to lacking relevance he didn't transfer the function to u8g2

christophepersoz, 2017-11-16:
Indeed, U8G2 supports only monochrome displays while U8Glib supports grayscale on monochrome display. I used this option on a SSD1322 display.

olikraus, 2017-11-16:
I intentionally removed grayscale support from U8g2. Vison is to create an extra grayscale lib. But this project has not yet started.

lorol, 2017-11-16:
Thank you for your clarification. I guess, it is challenging to standardize across such exotic low-level implementations and keep a minimal overhead for the rest of devices with no gray scale.  I have SSD1322 to try something ... @christophepersoz - I 'll appreciate if you share details of your tests.

christophepersoz, 2017-11-16:
I'm running the SSD1322 in HW SPI on a Teensy board. It does work for the Teensy 3.2 to 3.5 - haven't tried yet on 3.6 but should run too.

If you are using a Teensy board, uou need a wrapper to use U8GLib, all is described in the link below :
[Teensy 3.0 and U8Glib](https://forum.pjrc.com/threads/23445-Teensy-3-0-and-u8glib?p=103616&viewfull=1#post103616)
If you are using an another board, you can be inpired by the word sone on Teensy to develop your own ;)

For Teensy you can also have a look here : [https://github.com/sumotoy/SSD_13XX](https://github.com/sumotoy/SSD_13XX)

Hope it helps.

lorol, 2017-11-17:
Thank you, FYI, I am using a ESP8266 module to drive the OLED, still, on HW SP.

closed 2017-12-07 by olikraus

lorol, 2017-12-12:
Hi, I managed to do a quick workaround of  "grayscaled font brightness" for given text portion trough redefining of u8x8.setContrast() usage on U8X8_SSD1322_NHD_128X64_4W_HW_SPI.
If someone is interested, I will post a patch I made against rev.190

[image]

[image]

dhebbeker, 2020-04-14:
> [U8g2-NHD-2.7-12864WD-Font-Br.patch.txt](https://github.com/olikraus/u8g2/files/1557455/U8g2-NHD-2.7-12864WD-Font-Br.patch.txt)

I applied [the patch for `u8x8_d_ssd1322.c` (converted to git)](https://github.com/olikraus/u8g2/files/4476199/0001-quick-workaround-hack-of-grayscaled-font-brightness.patch.txt). 

The modified [`setContrast`](https://github.com/olikraus/u8g2/wiki/u8g2reference#setcontrast) must be called outside the [picture loop](https://github.com/olikraus/u8glib/wiki/tpictureloop) (between [`firstPage`](https://github.com/olikraus/u8g2/wiki/u8g2reference#firstpage) and [`nextPage`](https://github.com/olikraus/u8g2/wiki/u8g2reference#nextpage)), correct?

dhebbeker, 2020-04-14:
I do not understand, how you achieved different gray levels on the same page...

dhebbeker, 2020-04-14:
Ok, I got it! The **U8x8** library has been used instead of the U8g2 library! Now I have it running.

DennisLeng, 2020-04-15:
U8g2-NHD-2.7-12864WD-Font-Br.patch.txt

I did not run according to this codes for SSD132_256_64，at stm32 F411，please help me to How to refresh the display?

dhebbeker, 2020-04-15:
@DennisLeng, were you able to apply [**this** patch ](https://github.com/olikraus/u8g2/files/4476199/0001-quick-workaround-hack-of-grayscaled-font-brightness.patch.txt) to the library and merge it with the master branch? You need to use the U8x8 library (see [its introduction](https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#introduction)). 

Then you may adjust your drawing procedure as @lorol [did](https://github.com/olikraus/u8g2/issues/421#issuecomment-351194809):

> Usage:
> 
> ```
> ...
> u8x8.setContrast(7); // Gray levels are [0..15], >=15: full
> u8x8.drawString( ... // Following text portions will be with level 7, about 1/2 of full brightness 
> ...
> u8x8.setContrast(15);// Until you change
> ...
> ```