olikraus/u8g2 #422: SPI Interference with SD1607 200x200 Epaper display

SimGas, 2017-11-15:
Hi!

I'm using a SD1607 200x200 b/w display from waveshare. I set it to 3-wire Software SPI-mode and everything works fine. However, if I try to add another SPI device to my system, the display only shows parts(approx 2/3 with bme library) or no of my information(with SD-library). This happens even when there is no SPI connection physically.

I initially thought this is an issue with the used sensor (bme280, adafruit library), but then I just tried to include the standard sd-library. As soon as i open the communication with another SPI device(SD.begin, bme.begin, whatever), the display stops working and freaks out.

I used the following constructor:
`U8G2_SSD1607_200X200_F_3W_SW_SPI u8g2(U8G2_R3, /* clock=*/ SCK, /* data=*/ MOSI, /* cs=*/ CS_EINK);  // eInk/ePaper Display, original LUT from embedded artists`

Is there already a solution for this problem(didn't find anything) or do you have any idea how to solve the problem?

Thanks,
SimGas

olikraus, 2017-11-15:
Mixing hardware and software accees of different libraries to SPI bus will not work.
On the other side, there is no 3W_HW_SPI driver (Du to restrictions of the ATMega). 
My suggestion is to use 4W_HW_SPI for u8g2 (of course some rewiring would be required for your display).

SimGas, 2017-11-15:
Thank you for opening my eyes. Unfortunately I can't use 4W SPI, because I already ran out of pins(I've been using an adafruit Trinket M0)
I'll change my board choice now.

Sorry for opening an issue which has nothing to do with the actual library

closed 2017-11-15 by SimGas

olikraus, 2017-11-15:
no problem, happy to support here...

glydrfreak, 2018-01-02:
Wait up! It's funny how similar our projects are. I'm developing a variometer (basically just an altimeter)...

* 2.9" ePaper from waveshare
* Micro SD
* Adafruit's Bluefruit Feather M0
* MS5611 Pressure/temperature sensor in SPI mode.

The purpose is for a vertical speed indicator for paragliding.

I can get some incredibly smooth averaged altitude data at about 91 samples per second with the MS5611 sensor. But as soon as I enable the ePaper display, the sample rate of course decreases slightly to about 87 altitude samples per second, and the altitude data is suddenly very very noisy! Even if I only refresh the ePaper one time, the altitude samples get very noisy and it stays noisy until I call epd.Reset() , which is:

`digitalWrite(rstPin, LOW);
delay(200);
digitalWrite(rstPin, HIGH);
delay(200);`

Once I call epd.Reset() then the altitude samples return to the same smooth data I began with, but I still can't refresh the ePaper display without the resulting altitude noise. 

My gut tells me that this must be some interference between the two devices, or maybe somewhere else in the hardware. I opened up my own issue on my repository if anyone has anything helpful for me.

Much thanks!