olikraus/u8g2 #423: Is U8g2 compatible with Adafruit Feather nRF52 ?

christophepersoz, 2017-11-15:
Hi Oliver,

I'm trying to get u8g2 working in I2C on the Adafruit Feather nF52, but it seems that there is a problem with it. I2C SDL and SDA remains high, and never change accros the time.
Does u8g2 need a bridge to use it on this board?

Thanks a in advance for your fabulous work and for your answer.
Cheers.

Christophe

christophepersoz, 2017-11-16:
Hello, 
I finally made it works, and it does work well. Thanks again for your nice work.
Christophe

closed 2017-11-16 by christophepersoz

olikraus, 2017-11-16:
Excause me for beeing late with my answer. Very good you found your own solution.

How did you make it work? Was the U8g2 FAQ helpful here? Any hints for others with the same problem?

christophepersoz, 2017-11-16:
No problems for your answer delay ;)

At the beginning I tried HW_I2C which wasn't works, maybe due to bad init. So I went back on the Adafruit SSD1306 to run and exemple and be sure that the display was working as expected. Once check I went back to U8G2 in my sketch replacing all the Adafruit SSD1306 occurrences by the U8G2 ones, still in SW I2C. It works, then I tried to move back to HW I2C which was finally working too.

I assume that maybe a wrong parameter on the first init putted the display into a "instable" state which do not allows me to make the lib works at the first glance.

So the hints is, when something is not working, go back to basic examples, and see if they are working, then adapt your sketch accordingly with basics functions, to be sure that it will works nicely ;)

Thanks Oliver.