olikraus/u8g2 #424: IL3820 2.9 inch e-paper only works to a width of 250 px

SimGas, 2017-11-18:
Hi,

I've just got a new 2.9 inch e-paper display which seems to work fine at the first glimpse. But I discovered that it only the pixels 250x128 work. As soon as I reach the x-coordinate 250, it skips a few pixels of the text and starts over at x-coordinate 0.
I attached a picture to visualize my problem.
The problem also appears at both page-buffer modes.  
[image]
 I just used the Full Buffer "Hello World" example and changed the position to 210,50.
I used the constructor: `U8G2_IL3820_V2_296X128_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 3, /* data=*/ 4, /* cs=*/ 0, /* dc=*/ 1);`
Might it be an error in the library or is something wrong with my display?

Thanks
SimGas

SimGas, 2017-11-18:
I read the instructions again. This time more careful.

I thought, to enable 16 bit mode I have to insert //#define U8G2_16BIT, but as you've written in the instruction, it has to be uncommented in u8g2.h.

Sorry again for my silliness

closed 2017-11-18 by SimGas

olikraus, 2017-11-18:
I am glad that U8g2 works for you.