olikraus/u8g2 #427: Support for waveshare 7.5in b/w epaper

evan-opener, 2017-11-22:
Hi, I saw the demo for a lot of display devices and it was impressive. I want to know if you can support the waveshare 7.5inch b/w epaper. The datasheet is blow: 
https://www.waveshare.com/w/upload/b/b6/7.5inch-e-paper-specification.pdf
It also has a raspberryPi hat version with the connector.
https://www.ebay.com/itm/7-5inch-E-Ink-display-HAT-for-Raspberry-Pi-three-color-640x384-resolution-SPI-/253175592970

Current I just use the library from ZinggJM/GxEPD and there is no Chinese font support.
Thanks in advance.

evan-opener, 2017-11-23:
I know it is a little bit rackless to ask for the support.
If I want to use the construct function, where I can modify the  parameter to change the resolution and pin definition?

olikraus, 2017-11-23:
It is complicated and probably requires a lot of investigation.

See for example here: #318

evan-opener, 2017-11-24:
@olikraus 
Thanks. I will study them first.

closed 2017-11-26 by evan-opener