olikraus/u8g2 #429: GetAscent and GetDescent do not take note of font position

MattMatic, 2017-11-22:
Contrary to the documentation, the u8g2_GetAscent and _GetDescent do not reference the vertical position.

Changed to:
u8g2_GetAscent(u8g2) ((int8_t)(u8g2)->font_ref_ascent - (u8g2)->font_calc_vref(u8g2)))
u8g2_GetDescent(u8g2) ((int8_t)(u8g2)->font_ref_descent - (u8g2)->font_calc_vref(u8g2)))

olikraus, 2017-11-22:
Ascent and Descent are defined with respect to the character baseline. So I consider these values to be independent from the current reference position.
I could also say, that Ascent and Descent are font properties and not drawing characteristics.

In other words: The current definition should be valid.

MattMatic, 2017-11-22:
Well, they might be correct - but not always ;)
The problem occurs when trying to draw a box underneath the text. You have no way to know the vertical alignment! I presume that's why a) the documentation says that GetDescent/GetAscent ARE affected by font alignment and b) there is also GetFontAscent/GetFontDescent in the u8g2.h file.

So, I suspect that what you _meant_ to do was have GetFontAscent/GetFontDescent to return the actual font information, but GetAscent/GetDescent to return the adjusted value depending on SetFontPosXXXX.... which is what I had to do so I could work out the boundary of the text. :)

olikraus, 2017-11-22:
What about introduction of
u8g2_GetCurrentFontPosAscent()
and
u8g2_GetCurrentFontPosDescent()

MattMatic, 2017-11-22:
No problem for me. However, the docs should be updated (of course). And the slight confusion over the current GetAscent vs GetFontAscent - which are both the same! Please don't add a third ;-)

I'm considering just side-stepping this and sticking to SetFontPosBaseline and doing my own maths on the original GetFontAscent/GetFontDescent, particularly because I also need horizontal alignment Left/Centre/Right.

olikraus, 2017-12-07:
closing this without further action. Could be reopend, if more user request a different Ascent()/Descent() calculation.

closed 2017-12-07 by olikraus