olikraus/u8g2 #431: "Degree" character doesn't print correct...

dl9sec, 2017-11-23:
Hi,

I spotted a Problem with the "°" (degree, ASCII 176) character.
In both configurations (u8g2 and u8x8) a "Â°" ist printed instead of a single "°" character.
I tried this with the .print() as well as with the .drwStr method, and with several font,  both with the same result.

The code for a 4-wire SPI 126x64 OLED was quite simple:

`
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//U8X8_SSD1306_128X64_NONAME_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

void setup(void)
{
u8g2.begin();
}

void loop(void)
{
  u8g2.clearBuffer();                 // clear the internal memory
  
 u8g2.setFont(u8g2_font_10x20_tf);
  u8g2.setCursor(0,41);
  u8g2.print("T1:  37.25°C");
 
  u8g2.sendBuffer();                  // transfer internal memory to the display
  delay(1500);
}
`
The Output will result in:

`
T1:  37.25Â°C
`

Any ideas or workarounds?

Thank you.

Regards, Thorsten

uncle-buzz, 2017-11-23:
did you try to enable UTF8 ?

dl9sec, 2017-11-23:
Aaaaah! That's it. Thank you.

For those who wants to save 150 Bytes of code memory:

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

**const char str_degree[2] = {176, 0};**

void setup(void)
{
u8g2.begin();
}

void loop(void)
{
  u8g2.clearBuffer();                 // clear the internal memory
  
 u8g2.setFont(u8g2_font_10x20_tf);
  u8g2.setCursor(0,41);
  u8g2.print("T1:  37.25"); u8g2.print(str_degree); u8g2.print("C");
 
  u8g2.sendBuffer();                  // transfer internal memory to the display
  delay(1500);
}
```
:-)

olikraus, 2017-11-23:
Or simpler:
u8g2.print("T1:  37.25\0xb0C");

Actually there are two solution: With and without UTF8 enabled:
With UTF8 enabled, you can just write:
  u8g2.print("T1:  37.25°C");
Without UTF8 enabled, you need to feed the encoding number into the string. 
  u8g2.print("T1:  37.25\0xb0C");

The second solution will only work with unicode values below 256.

dl9sec, 2017-11-23:
Great stuff.
Thanks a lot for the hints :-)

closed 2017-11-23 by dl9sec

olikraus, 2017-11-23:
Edit: Fixed the second print(). Correct hex sequence starts with \x

dl9sec, 2017-11-24:
Unfortunately the hex-escaping "...\xb0C" doesn't seem to work because of the immediately following "C". When i put a whitespace between like "...\xb0 C" the characters are shown corect. Also if i use e.g. "...\xb0K", it is shown as "°K". Strange...

olikraus, 2017-11-24:
Mabye because b0C is a valid hex number.